chipweinberger/flutter_blue_plus #583: [Question]: Is there a way to know more about a disconnection logged as "The connection has timed out unexpectedly."?

R-Emagina, 2023-09-23:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.3.0

### What OS?

iOS

### OS Version

16

### Bluetooth Module

Microchip WBZ451

### Ask your question

Hi,

I'm working on an app that needs connecting and maintaining that connection to a Microchip MCU using BLE.
Everything works fine except I'm frequently getting random disconnections. I can't pinpoint an error in the MCU causing the disconnection as it happens at any time.
Is there a way I can know more about this disconnection by FBP?
The only thing I get for now are in those logs below.
I know nobody can really debug in my stead since I'm connecting to a custom made BLE device, but I'm trying to find ways to debug it that I haven't thought of already.
Thanks!!

### Logs

```shell
[FBP-iOS] handleMethodCall: writeCharacteristic
flutter: [FBP] <writeCharacteristic> result: true
[FBP-iOS] didWriteValueForCharacteristic
flutter: [FBP] [[ OnCharacteristicWritten ]] result: {remote_id: AF961D97-E565-D133-FEE7-9032EF1B9926, secondary_service_uuid: null, characteristic_uuid: 49535343-8841-43f4-a8d4-ecbe34729bb3, success: 1, error_code: null, service_uuid: 49535343-fe7d-4ae5-8fa9-9fafd205e455, error_string: null}
[FBP-iOS] didDisconnectPeripheral: [Error] The connection has timed out unexpectedly.
flutter: [FBP] [[ OnConnectionStateChanged ]] result: {remote_id: AF961D97-E565-D133-FEE7-9032EF1B9926, connection_state: 0, disconnect_reason_code: 6, disconnect_reason_string: The connection has timed out unexpectedly.}
flutter: Listener: BluetoothConnectionState.disconnected
```

chipweinberger, 2023-09-23:
use ios packet logger

https://www.bluetooth.com/blog/a-new-way-to-debug-iosbluetooth-applications/

closed 2023-09-23 by chipweinberger

R-Emagina, 2023-09-23:
Thanks, I'll check that out!