chipweinberger/flutter_blue_plus #584: [Help]: Listening on Characteristic.

mqhamdam, 2023-09-25:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

-? how to check ?

### Ask your question

Good Morning,

I'm trying to listen to characteristic updates, and don't know how properly do so. 
I receive characteristic updates only when I send write command to write characteristic.

I have two device one uses BLE v4 other v5.
in case of BLE v4, I'm getting characteristic updates, but values are always same, in case of BLE v5. characterstic updates only works when I write on write characteristic.

Below the way I write and listen to characteristic.
 
```
extension BluetoothPlusX on BluetoothDevice {
  Future<void> sendData(ProstheticWriteProtocol writeProtocol) async {
    
    final discoveredServices = await discoverServices();
    final writeCharacteristic = discoveredServices[3].characteristics[0];
    log("App > Device Protocol Chars Values : ${writeProtocol.toBytes()}");
    log("App > Device Protocol CHARS Size : ${writeProtocol.toBytes().length}");
    log("App > Device Protocol data : $writeProtocol");

    await writeCharacteristic.write(writeProtocol.toBytes());
  }

  Stream<ProstheticReadProtocol?> get listenToReadChars async* {
   
    final discoveredServices = await discoverServices();
    final readCharacteristic = discoveredServices[3].characteristics[1];

    if (readCharacteristic.isNotifying == false) {
      await readCharacteristic.setNotifyValue(true);
    }
    yield* collectedPackets(
        readCharacteristic.lastValueStream, readCharacteristic);
  }
}
```

I could say it's device problem, but  we have android app that works fine and receive characteristic updates in real-time. (only in BLE v4)

### Logs

```shell
-- no logs
-- no updates
-- no errors
-- no warnings
```

chipweinberger, 2023-09-25:
from the readme 

---

### onValueReceived is never called

**1. you are not subscribed OR not calling read**

Your device will only send values after you call `await characteristic.setNotifyValue(true)`, or `await characteristic.read()`

**2. you are calling write**

`onValueReceived` is only called for reads & notifies.

You can do a single read with `await characteristic.read(...)`

**3. your device has nothing to send**

If you are using `setNotifyValue`, your device chooses when to send data.

Try interacting with your device to get it to send new data.

**4. your device has bugs**

Try rebooting your ble device. 

Some ble devices have buggy software and stop sending data

chipweinberger, 2023-09-25:
`discoveredServices[3].characteristics[1];`

This is bad code. Android & iOS do not guarentee the order. you should check UUID

chipweinberger, 2023-09-25:
also, try the example app

```
cd ./example
flutter run
```

mqhamdam, 2023-09-25:
@chipweinberger Thank you for fast response and suggestions. 
```
extension BluetoothPlusX on BluetoothDevice {
  Future<void> sendData(ProstheticWriteProtocol writeProtocol) async {
    final discoveredServices = await discoverServices();
    final writeCharacteristic = findCharacteristic(
      findService(discoveredServices, SERVICE_UUID).characteristics,
      WRITE_CHARACTERISTIC_UUID,
    );
    log("writeChars ID ${writeCharacteristic.characteristicUuid}");
    log("writechars UUID ${writeCharacteristic.uuid}");
    await writeCharacteristic.write(writeProtocol.toBytes());
  }

  Stream<ProstheticReadProtocol?> get listenToReadChars async* {
    //await connect();
    final discoveredServices = await discoverServices();
    final readCharacteristic = findCharacteristic(
      findService(discoveredServices, SERVICE_UUID).characteristics,
      NOTIFY_CHARACTERISTIC_UUID,
    );
    log("readCharacteristic ID ${readCharacteristic.characteristicUuid}");
    log("readCharacteristic UUID ${readCharacteristic.uuid}");
    if (readCharacteristic.isNotifying == false) {
      await readCharacteristic.setNotifyValue(true);
    }
    yield* collectedPackets(
        readCharacteristic.lastValueStream, readCharacteristic);
  }
}
```

I've updated my code. 
but nothing have changed,
I'm receiving `onValueReceieved` only when I write to characteristic N1, I cannot send Read command to characteristic N2 because it applies only Notify parameter.

>  onValueReceived is never called
> 1. you are not subscribed OR not calling read
> 
> Your device will only send values after you call await characteristic.setNotifyValue(true), or await characteristic.read()
> 
> 2. you are calling write instead of read
> 
> onValueReceived is only called for reads & notifies, not writes.
> 
> You can do a single read with await characteristic.read(...)
> 
> 3. your device has nothing to send
> 
> If you are using setNotifyValue, your device chooses when to send data.
> 
> Try interacting with your device to get it to send new data.
> 
> 4. your device has bugs
> 
> Try rebooting your ble device.
> 
> Some ble devices have buggy software and stop sending data

- [x] I am subscribed to characteristic N2, since I receive data when I write to characteristic N1.
- [x] Characteristic N2 doesn't apply WRITE,READ parameter. I write only to Characteristic N1
- [x] I cannot do read to Characteristic N2, since it doesn't apply READ parameter.
- [x] If device decides when to send, why I'm getting realtime updates without sending write command.
- [x] Maybe, but it works on android app.
- [x] I did reboot both devices and IPAD

Thanks you again. !

chipweinberger, 2023-10-05:
did you resolve it?

chipweinberger, 2023-10-06:
closing. it is likely a problem with your device.

closed 2023-10-06 by chipweinberger