chipweinberger/flutter_blue_plus #585: [Bug]: `connectionState` does not change to disconnected when bluetooth is turned off on iOS

trey-wallis, 2023-09-27:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.13.2

### What OS?

iOS

### OS Version

16.6

### Bluetooth Module

Not sure

### What happened?

On iOS if a user turns off the bluetooth service, all devices will disconnect but the `connectionState` stream on each device never changes.

## Example
I scan for devices and pass the device from a scan result into the `BTDevice` class. I subscribe to the `connectionState`. I have my own enum of `BTDeviceState` so that external libraries don't need to install FBP. When the `connectionState` changes to `disconnected`, I handle UI changes. The issue is that the device is disconnected but since the `connectionState` doesn't change to `disconnected`, my app doesn't update its UI.

```dart
enum BTDeviceState { disconnected, connected }

class BTDevice {
   BluetoothDevice _device;
  StreamSubscription?
        _deviceStateSub;
   ValueNotifier<BTDeviceState> btDeviceStateNotifier = ValueNotifier<BTDeviceState>(BTDeviceState.disconnected);
  
  BTDevice(this._device){
    _initListeners();
  }

  void setBTDeviceState(BluetoothConnectionState state) {
    if (state == BluetoothConnectionState.connected) {
      btDeviceStateNotifier.value = BTDeviceState.connected;
    } else if (state == BluetoothConnectionState.disconnected) {
      btDeviceStateNotifier.value = BTDeviceState.disconnected;
    }
  }

  void _initListeners() {
       Logger.t("called");
        btDeviceStateNotifier.addListener(() async {
          if (stateNotifier.value == BTDeviceState.disconnected) {
            Logger.d("handling disconnection state");
            ///clean up UI variables
          } else {
            Logger.d("handling connection state");
            ///setup UI variables
           }
        }

        _deviceStateSub = _device.connectionState.listen((state) {
          Logger.d("connectionState: $state");
          setBTDeviceState(state);
        });
      }
  }
```

### Logs

```shell
Performing hot restart...                                               
Restarted application in 1,245ms.
[FBP-iOS] handleMethodCall: flutterHotRestart
[FBP-iOS] disconnectAllDevices
[FBP-iOS] connectedPeripherals: 0
[FBP-iOS] HotRestart: complete
flutter: [app] BluetoothService.startScan: called
flutter: [app] BTDeviceScanner.scanBTDevicesByMacAddress: called
flutter: [app] BluetoothStackState.init.: isScanning: true
[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] handleMethodCall: setLogLevel
[FBP-iOS] handleMethodCall: getAdapterState
flutter: [app] BluetoothStackState.init.: adapterState: BluetoothAdapterState.on
[FBP-iOS] handleMethodCall: startScan
flutter: [app] BTDeviceScanner._scanResultToBTDeviceStreamHandler: found new device
flutter: [app] BTDevice.connect: called
flutter: [app] BTDevice._initListeners: called
flutter: [app] BTDevice.connect: connecting to device
flutter: [app] BTDevice._initListeners.: connectionState: BluetoothConnectionState.disconnected
[FBP-iOS] handleMethodCall: connect
[FBP-iOS] didConnectPeripheral
flutter: [app] BTDevice._initListeners.: connectionState: BluetoothConnectionState.connected
flutter: [app] BTDevice._initListeners.: handling connection state
[FBP-iOS] handleMethodCall: discoverServices
[FBP-iOS] didDiscoverServices
[FBP-iOS] Found service: 180A
[FBP-iOS] Found service: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] Found service: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] handleMethodCall: setNotification
[FBP-iOS] didUpdateNotificationStateForCharacteristic
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
flutter: [app] received data: [129, 255, 127, 0]
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
flutter: [app] received data: [129, 255, 127, 0]
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
flutter: [app] received data: [129, 255, 127, 0]
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
flutter: [app] received data: [129, 125, 95, 99]
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] didUpdateValueForCharacteristic XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOff
[FBP-iOS] disconnectAllDevices
[FBP-iOS] calling disconnect: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x2804cd4a0> can only accept this command while in the powered on state
flutter: [app] BluetoothStackState.init.: adapterState: BluetoothAdapterState.off
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
flutter: [app] BluetoothStackState.init.: adapterState: BluetoothAdapterState.on
```

chipweinberger, 2023-09-27:
I just pushed a commit, can you try master branch?

If master does not work, try adding this 'else' statement code around line 821

```
        // when the adapter is turned off, we are not supposed to call
        // cancelPeripheralConnection as it is implicit
        if (isAdapterTurnedOff == false) {
            [self.centralManager cancelPeripheralConnection:peripheral];
        } else {
            // See BmConnectionStateResponse
            NSDictionary *result = @{
                @"remote_id":                [[peripheral identifier] UUIDString],
                @"connection_state":         @([self bmConnectionStateEnum:peripheral.state]),
                @"disconnect_reason_code":   error ? @(error.code) : [NSNull null],
                @"disconnect_reason_string": error ? [error localizedDescription] : [NSNull null],
            };

            // Send connection state
            [_methodChannel invokeMethod:@"OnConnectionStateChanged" arguments:result];
        }
```

chipweinberger, 2023-09-27:
also, thanks for including all logs, very helpful.

trey-wallis, 2023-09-27:
The master branch did not work. But staying on the master branch and adding the code at line 821 did. When the bluetooth adapter turned off, the device that was disconnected changed its `connectionState` to `disconnected`. This seems like a fix.

### Logs
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOff
[FBP-iOS] disconnectAllDevices
[FBP-iOS] calling disconnect: XXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
flutter: [app] BTDevice._initListeners.: connectionState: BluetoothConnectionState.disconnected
flutter: [app] BTDevice._initListeners.: handling disconnection state

chipweinberger, 2023-09-27:
ok, I pushed a commit to master branch. try it please!

Once we fix https://github.com/boskokg/flutter_blue_plus/issues/586 as well, I will release to 1.15.9

please assist in fixing https://github.com/boskokg/flutter_blue_plus/issues/586

trey-wallis, 2023-09-27:
Testing out the master branch right now

trey-wallis, 2023-09-27:
Could not build the precompiled application for the device.
Parse Issue (Xcode): Expected ':'
/Users/trey/.pub-cache/git/flutter_blue_plus-ca2bb7c5ccedf0d1592b678bd887b14b9a0b155e/ios/Classes/FlutterBluePlusPlugin.m:829:76

Parse Issue (Xcode): Expected expression
/Users/trey/.pub-cache/git/flutter_blue_plus-ca2bb7c5ccedf0d1592b678bd887b14b9a0b155e/ios/Classes/FlutterBluePlusPlugin.m:829:76

trey-wallis, 2023-09-27:
I got this error previously when I tried the code snippet that you had posted. I had to add another condition to the ternary operator.

e.g.
```swift
@"disconnect_reason_string": error ? @"Bluetooth turned off" : [NSNull null]
```

chipweinberger, 2023-09-27:
ah ya, thats a typo. fixed. thanks.

trey-wallis, 2023-09-27:
I just tested master for both bugs. Works great! :)

chipweinberger, 2023-09-27:
fixed 1.15.9

closed 2023-09-27 by chipweinberger