chipweinberger/flutter_blue_plus #586: [Help]:

trey-wallis, 2023-09-27:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.13.2

### What OS?

iOS

### OS Version

16.6

### Bluetooth Module

Not sure

### Ask your question

What should be the expected behavior when the bluetooth adapter is turned off on iOS. Right now it seems if a user had a scan running before the adapter was turned off, the scan is still running.

Steps to reproduce:
1. Start a scan
2. Turn off the bluetooth adapter
3. Turn on the bluetooth adapter
4. Try starting a new scan
5. See exception

If I don't try restarting the scan, there's no exception. However, the bluetooth acts as if it's no longer scanning

Steps to reproduce:
1. Start a scan
2. Turn off the bluetooth adapter
3. Turn on the bluetooth adapter
4. There's no exception but no scan results are coming in now

The logs are for when I don't try starting a new scan. The logs for the first are the same, only there's an exception thrown from the extension.

I am using the scanning extension from the migration file

```dart
  Stream<ScanResult> _scan({
    List<Guid> withServices = const [],
    Duration? timeout,
    bool androidUsesFineLocation = false,
  }) {
    if (FlutterBluePlus.isScanningNow) {
      throw Exception("Another scan is already in progress");
    }

    final controller = StreamController<ScanResult>();

    var subscription = FlutterBluePlus.scanResults.listen(
      (r) => controller.add(r.first),
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );

    FlutterBluePlus.startScan(
      withServices: withServices,
      timeout: timeout,
      removeIfGone: null,
      oneByOne: true,
      androidUsesFineLocation: androidUsesFineLocation,
    );

    Future scanComplete =
        FlutterBluePlus.isScanning.where((e) => e == false).first;

    scanComplete.whenComplete(() {
      subscription.cancel();
      controller.close();
    });

    return controller.stream;
  }
```

### Logs

```shell
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOff
[FBP-iOS] disconnectAllDevices
[FBP-iOS] calling disconnect: XXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXXX
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x2827fdae0> can only accept this command while in the powered on state
flutter: [app] BluetoothStackState.init.: adapterState: BluetoothAdapterState.off
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
flutter: [oh_ble] BluetoothStackState.init.: adapterState: BluetoothAdapterState.on

Nothing happens even though it's still scanning
```

chipweinberger, 2023-09-27:
> If I try starting a scan, I get an error. 

What error do you get?

chipweinberger, 2023-09-27:
Also, feel free to try and fix this yourself, and open a PR.

For one, we should listen to `OnAdapterStateChanged` in `_methodCallHandler`

```
  static Future<dynamic> _methodCallHandler(MethodCall call) async {

    // keep track of adapter states
    if (call.method == "OnAdapterStateChanged") {
        if(isScanningNow) {
            _scanSubscription?.cancel();
            _scanTimeout?.cancel();
            _isScanning.add(false);
        }
    }
```

But I am not sure if there needs to be any changes to the iOS code. I'm curious what error you are getting.

trey-wallis, 2023-09-27:
Here is the error that I am getting. It is originating from the migration code.

[FBP-iOS] didUpdateValueForCharacteristic XXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXXX
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOff
[FBP-iOS] disconnectAllDevices
[FBP-iOS] calling disconnect: XXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXXX
[FBP-iOS] else
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn

══╡ EXCEPTION CAUGHT BY FOUNDATION LIBRARY ╞════════════════════════════════════════════════════════
The following _Exception was thrown while dispatching notifications for ValueNotifier<BTState>:
Exception: Another scan is already in progress

When the exception was thrown, this was the stack:
#0      BTDeviceScanner._scan (package:app/bt_device_scanner.dart:166:7)
#1      BTDeviceScanner.scanBTDevicesByMacAddress (package:app/bt_device_scanner.dart:133:14)
#2      BluetoothService.startScan (package:app/services/bluetooth_service.dart:137:49)
#3      BluetoothService._btStateListener (package:app/services/bluetooth_service.dart:70:9)
#4      ChangeNotifier.notifyListeners (package:flutter/src/foundation/change_notifier.dart:403:24)
#5      ValueNotifier.value= (package:flutter/src/foundation/change_notifier.dart:530:5)
#6      BluetoothStackState.init.<anonymous closure> (package:app/bluetooth_stack_state.dart:67:23)
(elided 17 frames from dart:async)

The ValueNotifier<BTState> sending notification was:
  ValueNotifier<BTState>#9ab2c(BTState.on)
════════════════════════════════════════════════════════════════════════════════════════════════════

[image]

chipweinberger, 2023-09-27:
> Another scan is already in progress

Oh ok, great. The code change I suggested above will fix this.

trey-wallis, 2023-09-27:
Just tested it. Yep! That fixes it.

chipweinberger, 2023-09-27:
ok, I pushed a fix to master branch. 

I will do a release tomorrow, if master branch works for you.

chipweinberger, 2023-09-27:
fixed 1.15.9

closed 2023-09-27 by chipweinberger