chipweinberger/flutter_blue_plus #587: [Help]:

Sumitdhiman-HHC, 2023-09-27:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.13.2

### What OS?

Android, iOS

### OS Version

16.2

### Bluetooth Module

BLE device

### Ask your question

Using Stream:
*** characteristic.onValueReceived.listen ***

I getting the value in splitting parts instead of a whole data, due to that not able to decode it properly in the formatted value. Help me or let me know the procedure through which I can solve this problem.

Thanks in advance.

### Logs

```shell
[   +7 ms] I/flutter (17724): secret_value: [48, 49, 44, 32, 34, 80, 114, 111, 118, 105, 100, 101, 100, 32, 116, 114]
[   +7 ms] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr}
[        ] D/[FBP-Android](17724): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[   +2 ms] I/flutter (17724): secret_value: [97, 110, 115, 97, 99, 116, 105, 111, 110, 32, 73, 68, 34, 58, 32, 49]
[   +2 ms] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr, ansaction ID": 1}
[   +3 ms] I/flutter (17724): secret_value: [44, 32, 34, 82, 101, 115, 112, 111, 110, 115, 101, 34, 58, 32, 34, 69]
[   +1 ms] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr, ansaction ID": 1, , "Response": "E}
[  +10 ms] D/[FBP-Android](17724): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[   +1 ms] D/[FBP-Android](17724): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[   +3 ms] I/flutter (17724): secret_value: [114, 114, 111, 114, 34, 44, 32, 34, 69, 114, 114, 111, 114, 32, 116, 101]
[  +10 ms] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr, ansaction ID": 1, , "Response": "E, rror", "Error te}
[   +2 ms] I/flutter (17724): secret_value: [120, 116, 34, 58, 32, 34, 78, 111, 110, 45, 115, 101, 113, 117, 101, 110]
[        ] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr, ansaction ID": 1, , "Response": "E, rror", "Error te, xt": "Non-sequen}
[  +32 ms] D/[FBP-Android](17724): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[        ] D/[FBP-Android](17724): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[  +12 ms] I/flutter (17724): secret_value: [116, 105, 97, 108, 32, 116, 114, 97, 110, 115, 97, 99, 116, 105, 111, 110]
[        ] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr, ansaction ID": 1, , "Response": "E, rror", "Error te, xt": "Non-sequen, tial transaction}
[   +5 ms] I/flutter (17724): secret_value: [32, 73, 68, 34, 125]
[        ] I/flutter (17724): response_data_2: {{"Expected trans, action ID": 2, ", Error number": 1, 01, "Provided tr, ansaction ID": 1, , "Response": "E, rror", "Error te, xt": "Non-sequen, tial transaction, ID"}}
```

chipweinberger, 2023-09-27:
There is nothing FlutterBluePlus can do about this problem.

You will need to change your device code, or find a way to combine the data yourself.

closed 2023-09-27 by chipweinberger

Sumitdhiman-HHC, 2023-09-28:
I already set the mtu(512) but it is giving the response as I mentioned above