chipweinberger/flutter_blue_plus #588: [Help]:

KatayR, 2023-09-27:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.13.5

### What OS?

macOS

### OS Version

13.6

### Bluetooth Module

Proteus 2 (AMB2623)

### Ask your question

I was using Bluetooth Classic for a different module and I was able to succesfuly implement all the features I wanted. 

I want to provide bluetooth address to the package and it will automatically bond + connect to it. It would be cool if it enters password too because I was able to do it with Bluetooth Classic but if it's not possible, then nevermind.

After that, I will read and write data by using predefined UUIDs but I think I can do it myself.

For now, all I care about is auto bond + connect feature. I couldn't do it.

I would appreciate any help.

Thanks 

### Logs

```shell
No logs
```

chipweinberger, 2023-09-27:
autoConnect is support on android. use `device.connect(autoConnect:true)`

but for iOS, autoConnect it is only supported on iOS 17+ and is untested.

closed 2023-09-27 by chipweinberger

KatayR, 2023-09-28:
Well, I think I should've been clearer.

I have many Proteus 2 (AMB2623) modules and so a bunch of bluetooth addresses. I was using qr codes to get bluetooth address for each module and provide that address to bluetooth classic so it would try to connect to that specific address. I want to achieve that in BLE.

chipweinberger, 2023-09-28:
in BLE, the addresses are randomly generated on iOS & android for privacy.

But you could use a custom device name per device or something instead for identification. 

It sounds doable, you just need to read up on how BLE works more.

KatayR, 2023-09-28:
Oh.. I thought that "remote_id" is address so I wrote this:

```
  Future<void> setupBluetooth() async {
    if (await FlutterBluePlus.isAvailable == false) {
      print("Bluetooth not supported by this device");
      return;
    }

    if (Platform.isAndroid) {
      await FlutterBluePlus.turnOn();
    }

    await FlutterBluePlus.adapterState
        .where((s) => s == BluetoothAdapterState.on)
        .first;

    device = await BluetoothDevice.fromId('00:18:DA:22:C1:7B');
    print("Device: $device");
  }
```
  
  but it didn't work. I guess that's why.

KatayR, 2023-09-29:
I DID IT! Tho, address changes depending on the os. On Android, I can use standard Bluetooth address('00:18:DA:22:C1:7B') but on iOS, I have to use some other address which gets generated for each device and is unique('0C5713D6-1173-46B7-B165-91D3F4CA6444', for example), so my QR code plan is dead.

Code is here(not optimized or beautiful at all but it works for now):

```
// this code works for Proteus 2, which uses Nordic Semiconductor nRF52832-CHAA SoC.

final String serviceUuid = "6e400001-C352-11E5-953D-0002A5D5C51B";
  final String writeCharUuid = "6e400002-C352-11E5-953D-0002A5D5C51B";
  final String readCharUuid = "6e400003-C352-11E5-953D-0002A5D5C51B";

  BluetoothDevice myDevice = BluetoothDevice(
    localName: 'name',
    type: BluetoothDeviceType.le,
    remoteId: const DeviceIdentifier('00:18:DA:22:C1:7B'),
    // if ios: 0C5713D6-1173-46B7-B165-91D3F4CA6444
    // if android: 00:18:DA:22:C1:7B
  );

  void connectToDevice(BluetoothDevice device) async {
    // Connect to the device
    await device.connect();

   // Discover all services, but this could be bypassed
    List<BluetoothService> services = await device.discoverServices();

    // Find the correct service
    BluetoothService service = services.firstWhere(
      (s) => s.uuid.toString().toLowerCase() == serviceUuid.toLowerCase(),
    );

    // Find write characteristic
    BluetoothCharacteristic writeCharacteristic =
        service.characteristics.firstWhere(
      (c) => c.uuid.toString().toLowerCase() == writeCharUuid.toLowerCase(),
    );

    // Find read characteristic
    BluetoothCharacteristic readCharacteristic =
        service.characteristics.firstWhere(
      (c) => c.uuid.toString().toLowerCase() == readCharUuid.toLowerCase(),
    );

    await readCharacteristic
        .setNotifyValue(writeCharacteristic.isNotifying == false);
    var snackBar = snackBarGood("Subscription: Success");
    snackBarKeyC.currentState?.removeCurrentSnackBar();
    snackBarKeyC.currentState?.showSnackBar(snackBar);

    // Message to write to the characteristic
    String text = "message sent\r\n";
    int header = 0x01;
    List<int> message = [header];

    // add text to message as bytes
    message.addAll(text.codeUnits);

    await writeCharacteristic.write(message);
    snackBar = snackBarGood("Write: Success");
    snackBarKeyC.currentState?.removeCurrentSnackBar();
    snackBarKeyC.currentState?.showSnackBar(snackBar);
    if (writeCharacteristic.properties.read) {
      await writeCharacteristic.read();
    }
  }
```

chipweinberger, 2023-09-29:
glad you got it!