chipweinberger/flutter_blue_plus #594: Example not work [Help]:

ThanyasitA, 2023-09-29:
### FlutterBluePlus Version

FlutterBlue 0.8.0 

### Flutter Version

3.13.6

### What OS?

All

### OS Version

Android

### Bluetooth Module

ESP32 bluetooth classic

### Ask your question

First and foremost, I'm new to Flutter, and I'd like to create an app that can retrieve data from an ESP32 using Bluetooth Classic. Unfortunately, I've encountered issues with both the FlutterBlue and FlutterBlueplus libraries, as they don't seem to be working as expected.

Could you please direct me to some basic code examples that have been proven to work for this scenario?

I'm also curious if it's possible to use FlutterBlueplus with an ESP32 that uses Bluetooth Classic, or if there are any alternative libraries or approaches I should consider?"

### Logs

```shell
.
```

chipweinberger, 2023-09-29:
> it's possible to use FlutterBlueplus with an ESP32 that uses Bluetooth Classic, 

not possible. I don't know of any libraries either.

> Could you please direct me to some basic code examples that have been proven to work for this scenario?

The only basic code example are in the readme. There are no other examples. Sorry, the example app is too complicated for most people.

> Example not work

What error or issue do you encounter?

ThanyasitA, 2023-09-29:
```
import 'package:flutter/material.dart';
import 'package:flutter_blue/flutter_blue.dart';

void main() {
  runApp(const MyApp());
}

class MyApp extends StatelessWidget {
  const MyApp({Key? key}) : super(key: key);

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      home: Scaffold(
        appBar: AppBar(
          title: const Text('My App'),
        ),
        body: const Center(
          child: Column(
            mainAxisAlignment: MainAxisAlignment.center,
            children: [
              Text('Test'),
              Text('Please wait...'),
              ElevatedButton(
                onPressed:
                    startBluetoothScan,
                child: Text('Start Bluetooth Scan'),
              ),
            ],
          ),
        ),
      ),
    );
  }
}

void startBluetoothScan() {
  FlutterBlue flutterBlue = FlutterBlue.instance;
  flutterBlue.startScan(timeout: const Duration(seconds: 5));
  print('123');
  flutterBlue.scanResults.listen((results) {
    // do something with scan results
    for (ScanResult r in results) {
      print('${r.device.name} found! rssi: ${r.rssi}');
    }
  });
}
```
I tried creating a basic code, but it couldn't detect the device.
Do you have any suggestions or advice?

**DEBUG CONSOLE**
_I/flutter (16027): 123
D/BluetoothAdapter(16027): isLeEnabled(): ON
D/BluetoothLeScanner(16027): onScannerRegistered() - status=0 scannerId=5 mScannerId=0_

chipweinberger, 2023-09-29:
sorry, you need to read the README. I cannot help you more. There is an entire section about not finding devices in the scan results.

closed 2023-09-29 by chipweinberger