chipweinberger/flutter_blue_plus #595: [Bug]: IOS getConnectedSystemDevices can not get device's name

SmallBlueWhale, 2023-09-29:
### FlutterBluePlus Version

1.15.10

### Flutter Version

3.10.0

### What OS?

iOS

### OS Version

ios 16.0.0

### Bluetooth Module

unknown

### What happened?

When i use getConnectedSystemDevices , i got empty local name in Dart as picture shown below. 
[image]

Also i debug the code in Xcode. i found that i can get the name in this code as picture shown below , but the cache is empty. The name has no usages. Can you fix this?
[image]
[image]
[image]

### Logs

```shell
as picture shown
```

chipweinberger, 2023-09-29:
thanks for filing. I'll consider a solution to this.

chipweinberger, 2023-09-29:
try master branch.

chipweinberger, 2023-09-30:
fixed in 1.16.0

closed 2023-09-30 by chipweinberger