chipweinberger/flutter_blue_plus #596: [Bug]: about bluetooth receiving data with connected device

Mani2049, 2023-09-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

flutter_blue_plus: ^1.5.2

### Flutter Version

flutter 3.13.1

### What OS?

Android, iOS

### OS Version

version 11.7.9

### Bluetooth Module

esp-32

### What happened?

when we connect the device with Bluetooth and send device command but its not receiving response. we have to collect a response from device.

### Logs

```shell
1.5.2 #
[fix] Android: setNotification was throwing exception (regression)
```

chipweinberger, 2023-09-30:
?

Mani2049, 2023-09-30:
iam asking about flutter flow (flutter_blue_plus package) its bluetooth ble
package.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
09/30/23,
12:29:33 PM

chipweinberger, 2023-09-30:
sorry i don't understand 

post code & logs

update to latest version

chipweinberger, 2023-09-30:
onValueReceived is never called

1. you are not subscribed OR not calling read

Your device will only send values after you call await characteristic.setNotifyValue(true), or await characteristic.read()

2. you are calling write instead of read

onValueReceived is only called for reads & notifies, not writes.

You can do a single read with await characteristic.read(...)

3. your device has nothing to send

If you are using setNotifyValue, your device chooses when to send data.

Try interacting with your device to get it to send new data.

4. your device has bugs

Try rebooting your ble device.

Some ble devices have buggy software and stop sending data

closed 2023-09-30 by chipweinberger

Mani2049, 2023-09-30:
https://blog.flutterflow.io/creating-an-app-for-interacting-with-any-iot-devices-using-ble/
i followed this link to solve my problem but i stuck under receive data function its not receiving a response. from my device. and  i used to read like this await characteristic.read()