chipweinberger/flutter_blue_plus #597: [Bug]: about bluetooth receiving data with connected device

Mani2049, 2023-09-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

flutter_blue_plus: ^1.5.2

### Flutter Version

flutter 3.13.1

### What OS?

Android, iOS

### OS Version

version 11.7.9

### Bluetooth Module

esp-32

### What happened?

https://blog.flutterflow.io/creating-an-app-for-interacting-with-any-iot-devices-using-ble/
i followed this link to solve my problem but i stuck under receive data function its not receiving a response. from my device. and i used to read like this await characteristic.read()

### Logs

```shell
1.5.2 
[fix] Android: setNotification was throwing exception (regression)
```

BachewalSuresh, 2023-09-30:
we also facing same bluetooth receiving data  issue please solve head of the department

BachewalSuresh, 2023-09-30:
import 'package:flutter_blue_plus/flutter_blue_plus.dart';

Future<String?> receiveData(BTDeviceStruct deviceInfo) async {
  final device = BluetoothDevice.fromId(deviceInfo.id, name: deviceInfo.name);
  final services = await device.discoverServices();
  for (BluetoothService service in services) {
    for (BluetoothCharacteristic characteristic in service.characteristics) {
      final isRead = characteristic.properties.read;
      final isNotify = characteristic.properties.notify;
      if (isRead && isNotify) {
        final value = await characteristic.read();
        return String.fromCharCodes(value);
      }
    }
  }
  return null;
}

this code we are using to receive a data from my esp-32 device

chipweinberger, 2023-09-30:
you need to include logs 

logs are required

closed 2023-09-30 by chipweinberger

chipweinberger, 2023-09-30:
upgrade to latest version

`flutter_blue_plus: 1.16.1`

then make some small changes to code. see `MIGRATION.md` file.

BachewalSuresh, 2023-10-19:
where do i find MIGRATIOn.md file

BachewalSuresh, 2023-10-19:
> where do i find MIGRATIOn.md file

MIGRATION.md

BachewalSuresh, 2023-10-19:
i am facing a problem with stripe payment getting me that [An unknown error occurred] in flutterflow project.