chipweinberger/flutter_blue_plus #598: [Help]: In version 1.15.0 in "Scanning API" the "scanMode" parameter disappeared from the "startScan" method

dkozenkov, 2023-10-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.15.0

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

all

### Bluetooth Module

all

### Ask your question

In version 1.15.0 in "Scanning API" the "scanMode" parameter disappeared from the "startScan" method.
How can I change the scanning mode in Android now?

### Logs

```shell
No logs
```

chipweinberger, 2023-10-01:
why do you need to change it?

you're right it was removed.

dkozenkov, 2023-10-02:
> why do you need to change it? which mode do you use?

In my application, scanning is performed continuously. And therefore, to save device battery power, I need to use the scanning mode "ScanMode.lowPower" on Android.

chipweinberger, 2023-10-02:
android will automatically switch to low power after awhile. That is my understanding.

Do you see some other behavior?

dkozenkov, 2023-10-05:
> android will automatically switch to low power after awhile. That is my understanding.
Yes, according to my observations, it looks like you are right.

> Is that good enough for you? Do you see some other behavior?
But there may be situations when it is not SCAN_MODE_LOW_POWER that is required, but SCAN_MODE_OPPORTUNISTIC.
I would like to be able to independently set the desired scanning mode on Android.

chipweinberger, 2023-10-05:
I don't think anyone uses `SCAN_MODE_OPPORTUNISTIC` =)

I will close this for now, but if there is greater demand perhaps I will add it back.

closed 2023-10-05 by chipweinberger