chipweinberger/flutter_blue_plus #599: [Bug]: issue with setNotifyValue and isNotifying on android 13 after latest security update

fitterfly-parth, 2023-10-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.2

### Flutter Version

3.13.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

We are using Accu-chek instant.

### What happened?

After updating the Os version setNotifyValue shows true in the log but it does not return any result and after some time it shows timeout.

### Logs

```shell
D/[FBP-Android](11758): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](11758): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(11758): connect() - device: B0:B1:13:59:FA:01, auto: false, eattSupport: false
D/BluetoothGatt(11758): registerApp()
D/BluetoothGatt(11758): registerApp() - UUID=0e0b67b8-15a3-4a69-b4b7-08f2af4bf772
D/BluetoothGatt(11758): onClientRegistered() - status=0 clientIf=12
D/BluetoothGatt(11758): onClientConnectionState() - status=0 clientIf=12 device=B0:B1:13:59:FA:01
D/[FBP-Android](11758): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](11758): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(11758): discoverServices() - device: B0:B1:13:59:FA:01
D/BluetoothGatt(11758): onConnectionUpdated() - Device=B0:B1:13:59:FA:01 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(11758): onSearchComplete() = Device=B0:B1:13:59:FA:01 Status=0
D/[FBP-Android](11758): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/BluetoothGatt(11758): onConnectionUpdated() - Device=B0:B1:13:59:FA:01 interval=38 latency=0 timeout=500 status=0
D/[FBP-Android](11758): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(11758): setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
```

fitterfly-parth, 2023-10-03:
E/flutter (15536): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: setNotifyValue: (code: 1) Timed out after 15s
E/flutter (15536): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:57:7)
E/flutter (15536): #1      _RootZone.run (dart:async/zone.dart:1655:54)
E/flutter (15536): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:938:34)
E/flutter (15536): #3      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
E/flutter (15536): #4      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
E/flutter (15536): #5      _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
E/flutter (15536): #6      _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:189:12)

chipweinberger, 2023-10-03:
After we call `setCharacteristicNotification()` (which just configures the android phone itself), we then update the CCCD descriptor on the remote device to enable notifications.

In other words, this might be a problem with your device. Is your device receiving the descriptor update?

fitterfly-parth, 2023-10-06:
[image]

chipweinberger, 2023-10-06:
your device is disconnecting

fitterfly-parth, 2023-10-06:
I disconnect the device if there is any error and timeout occurs. Please check my code.

<img width="1224" alt="Screenshot 2023-10-06 at 2 52 32 PM" src="https://github.com/boskokg/flutter_blue_plus/assets/104829044/c32d0b8c-0765-4655-ad67
[image]
-a3dcb7472f38">

fitterfly-parth, 2023-10-06:
I set timeout 30 seconds but it's Timed out after 15s only.

chipweinberger, 2023-10-06:
1. Is your device receiving the descriptor update?
2. you are hitting the default timeout. you can change it setNotifyValue(true, timeout:30);

chipweinberger, 2023-10-06:
try the example app

fitterfly-parth, 2023-10-06:
When 

> 1. Is your device receiving the descriptor update?
> 2. you are hitting the default timeout. you can change it setNotifyValue(true, timeout:30);

When it's timeout it shows ERR on the device.

I don’t know why this happen now everything working fine before update I have one device which is running on Andorid 13 also without new update on this device everything working fine. 

But only on latest updated version Android 13 it's not showing connection popup

chipweinberger, 2023-10-06:
Good luck. I have no special knowledge about this.

chipweinberger, 2023-10-07:
try 1.16.10

fitterfly-parth, 2023-10-09:
Still not working.

chipweinberger, 2023-10-09:
try the example app?

chipweinberger, 2023-10-11:
also try updating

fitterfly-parth, 2023-10-11:
I try 1.17.3 but still not working.

This issue occurs only for the Latest updated android device.

chipweinberger, 2023-10-11:
What is your OS build number?

Like you said, I cannot reproduce this issue on TQ3A.230805.001 (August 2023, Pixel 5a)

chipweinberger, 2023-10-11:
Also, you've also filed an AOSP bug report I assume? Can you share the link?

fitterfly-parth, 2023-10-12:
> What is your OS build number?
> 
> I cannot reproduce this issue on:
> 
> * TQ3A.230805.001 (August 2023, Pixel 5a)
> * TQ3A.230901.001 (September 2023, Pixel 5a)

I have Oppo CPH2371 and it's build number is CPH2371_13.1.0.591(EX01) And second device is OnePlus 9Pro and it's build number is LE2121_13.1.0.591(EX01).

fitterfly-parth, 2023-10-12:
Also, it's working fine with Android Native code but not working with your plugin.

chipweinberger, 2023-10-12:
We use native code as well. 

Can you spot any differences in your native code vs ours?

chipweinberger, 2023-10-30:
without more info, this cannot be fixed. closing.

closed 2023-10-30 by chipweinberger