chipweinberger/flutter_blue_plus #600: [Help]: iOS, odd  behaviors while filtering the searchScan results.

nissaba, 2023-10-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.15.9

### Flutter Version

3.13.5

### What OS?

iOS

### OS Version

iOS 17

### Bluetooth Module

Nordic NRF

### Ask your question

I get 2 odd and unexpected behaviors when scanning and filtering out the results. 

problem 1 is that the first time I try to scan the callback block does not get any calls to the scanResults.listen method.

the second one is really strange, the first line of code if it is a print  `debugPrint('${r.device.remoteId}: "${r.device.localName}" found! rssi: ${r.rssi}');`  directly after    `for (ScanResult r in results) {` and I put a breakpoint and the `r.device.localName` always points the first item in the result array, it as more things and each time the for loops r always point to the first item on the list. 

if the code is as below then I do not get problem #2 but still get problem #1.  And I don't see why this is happening.  The only thing is if I don't filter anything like in the documentation, it scans everything, and I am stuck with an array of result. Yes, I can filter out later when I output the device list for the UI in my state update. but for the sake of optimizing the memory usage, I would prefer filtering out results as they get in. 

```
import 'dart:async';

import 'package:flutter/cupertino.dart';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';
import 'package:notio/services/locator.dart';
import 'package:riverpod_annotation/riverpod_annotation.dart';

part 'scanned_device_model.g.dart';

@Riverpod(keepAlive: true)
class ScannedDevices extends _$ScannedDevices {
  final FlutterBluePlus _ble = locator.get<FlutterBluePlus>();
  final Set<BluetoothDevice> _devices = {};
  StreamSubscription? _subscription;

  @override
  BleScannerState build() {
    return BleScannerState(
        discoveredDevices: _devices.toList(),
        scanIsInProgress: _subscription != null);
  }

  void startScan() async {
    if (await FlutterBluePlus.isAvailable == false) {
      debugPrint("Bluetooth not supported by this device");
      return;
    }

    debugPrint('Start ble discovery');
    _devices.clear();
    _subscription?.cancel();
    _subscription = FlutterBluePlus.scanResults.listen(
      (results) {
        for (ScanResult r in results) {
          if (r.device.localName.startsWith('NOTIO')) {
            if (_devices.contains(r.device) == false) {
              debugPrint(
                  '${r.device.remoteId}: "${r.device.localName}" found! rssi: ${r.rssi}');
              _devices.add(r.device);
            } else {
              _devices.remove(r.device);
              _devices.add(r.device);
            }
          }

          _pushState();
        }
      },
      onError: (e) => debugPrint(e),
    );

    await FlutterBluePlus.startScan(
        timeout: const Duration(seconds: 15), androidUsesFineLocation: false);
    //_pushState();
  }

  void _pushState() {
    state = BleScannerState(
      discoveredDevices: _devices.toList(),
      scanIsInProgress: _subscription != null,
    );
  }

  Future<void> stopScan() async {
    await FlutterBluePlus.stopScan();
    _pushState();
  }
}

...
```

### Logs

```shell
no logs
```

chipweinberger, 2023-10-05:
sorry i don't really understand the question

nissaba, 2023-10-06:
I wanted to know why this is happening. how to stop it. 
I was lucky, the console logs gave me an error at some point, only saw it 1 time ever, but it said that the ble was not on. 
So I change 
```
if (await FlutterBluePlus.isAvailable == false) {
      debugPrint("Bluetooth not supported by this device");
      return;
    }
```

to just ```FlutterBluePlus.isAvailable ``` and moved it to the start of the build method of the first widget. this gave more time to the ble to turn on. 

I thought that ```await FlutterBluePlus.isAvailable``` have given the time to wake but in the end it came out as false and 
then the provider sits here doing nothing because we don't get updates from FlutterBluePlus.isAvailable as it is not a stream or a provider. 

So in short problem solved, took me a lot of guesswork to find it.

closed 2023-10-06 by nissaba

chipweinberger, 2023-10-06:
you should be listening to `FlutterBluePlus.adapterState` and waiting for the state to be ON