chipweinberger/flutter_blue_plus #601: [Feature]:

Sumitdhiman-HHC, 2023-10-05:
### FlutterBluePlus Version

1.15.10

### Flutter Version

3.13.6

### What OS?

Android, iOS

### OS Version

16

### Bluetooth Module

BLE device

### Feature Request

After sending the command the response is retrieving in splitting form, not able to get the complete response of a single command.

Currently I execute the command: Default Cycle Time

In the current response, as we are getting integer list ([123, 34, 69, 120, 112, 101, 99, 116, 101, 100, 32, 116, 114, 97, 110, 115]) of half response(here should be completed data) and after encoding it with: utf8.decode(listElement) we are getting: {"Expected trans

Same for the next responses, so why are not receiving the complete data at one time after execution of the command. So that we can proceed further event easily.

WE ARE GETTING THIS RESPONSE ON FUNCTION:

characteristic.onValueReceived.listen();

### Logs

```shell
command_response: [123, 34, 69, 120, 112, 101, 99, 116, 101, 100, 32, 116, 114, 97, 110, 115]
command_response_decoded: {"Expected trans
command_response: [97, 99, 116, 105, 111, 110, 32, 73, 68, 34, 58, 32, 49, 56, 50, 44]
command_response_decoded: action ID": 182,
command_response: [32, 34, 69, 114, 114, 111, 114, 32, 110, 117, 109, 98, 101, 114, 34, 58]
command_response_decoded: "Error number":
```

chipweinberger, 2023-10-05:
are you increasing the android mtu?

device.requestMtu(512);

closed 2023-10-05 by chipweinberger

Sumitdhiman-HHC, 2023-10-09:
I already added the above implementation of mtu. After that I'm getting the response in split form as I mentioned in above/previous command or Feature request.