chipweinberger/flutter_blue_plus #602: [Bug]:  But example app

lucasmrau, 2023-10-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

last

### Flutter Version

3.11

### What OS?

Android

### OS Version

13

### Bluetooth Module

ST

### What happened?

While using the example app

[image]

### Logs

```shell
Above
```

chipweinberger, 2023-10-06:
did you resolve it?

lucasmrau, 2023-10-06:
#688    Element.rebuild (package:flutter/src/widgets/framework.dart:5016:7)
framework.dart:5016
#689    BuildOwner.buildScope (package:flutter/src/widgets/framework.dart:2779:19)
framework.dart:2779
#690    WidgetsBinding.drawFrame (package:flutter/src/widgets/binding.dart:916:21)
binding.dart:916
#691    RendererBinding._handlePersistentFrameCallback (package:flutter/src/rendering/binding.dart:360:5)
binding.dart:360
#692    SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1297:15)
binding.dart:1297
#693    SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1227:9)
binding.dart:1227
#694    SchedulerBinding._handleDrawFrame (package:flutter/src/scheduler/binding.dart:1085:5)
binding.dart:1085
#695    _invoke (dart:ui/hooks.dart:170:13)
#696    PlatformDispatcher._drawFrame (dart:ui/platform_dispatcher.dart:401:5)
platform_dispatcher.dart:401
#697    _drawFrame (dart:ui/hooks.dart:140:31)
(elided 3 frames from dart:async)

chipweinberger, 2023-10-06:
fixed 1.16.6

closed 2023-10-06 by chipweinberger