chipweinberger/flutter_blue_plus #603: [Bug]: iOS platform methods return not bool but integer

BreadKey, 2023-10-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.4

### Flutter Version

3.13.6

### What OS?

iOS

### OS Version

16.6.1

### Bluetooth Module

Nordic

### What happened?

When connecting or disconnecting, invoking platform method returns not bool but integer.
```dart
  Future<void> disconnect({int timeout = 35}) async {
    // Only allow a single 'connectOrDisconnect' operation at the same time per device.
    String key = remoteId.str + ":connectOrDisconnect";
    _Mutex opMutex = await _MutexFactory.getMutexForKey(key);
    await opMutex.take();

    try {
      var responseStream = FlutterBluePlus._methodStream.stream
          .where((m) => m.method == "OnConnectionStateChanged")
          .map((m) => m.arguments)
          .map((args) => BmConnectionStateResponse.fromMap(args))
          .where((p) => p.remoteId == remoteId.str)
          .where((p) => p.connectionState == BmConnectionStateEnum.disconnected);

      // Start listening now, before invokeMethod, to ensure we don't miss the response
      Future<BmConnectionStateResponse> futureState = responseStream.first;

      // invoke
      bool changed = await FlutterBluePlus._invokeMethod('disconnect', remoteId.str);

      // only wait for disconnection if weren't already disconnected
      if (changed) {
        await futureState.fbpTimeout(timeout, "disconnect");
      }
    } finally {
      opMutex.give();
    }
  }
```

I think you should change ```@true``` to ```@YES```, ```@false``` to ```@NO```.

https://docs.flutter.dev/platform-integration/platform-channels?tab=type-mappings-obj-c-tab

### Logs

```shell
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: type 'int' is not a subtype of type 'bool'
#0      BluetoothDevice.disconnect (package:flutter_blue_plus/src/bluetooth_device.dart:100:12)
<asynchronous suspension>
```
```

chipweinberger, 2023-10-06:
same issue: https://github.com/boskokg/flutter_blue_plus/issues/605

chipweinberger, 2023-10-06:
should be fixed on master

dce7c26d93abfff1a8c6fd51e79a615115382e52

TheFe91, 2023-10-06:
When is this going to be released? I see that on releases on GH we have `1.16.4`, on pub it arrives only to `1.16.2` but no release has been done yet

agozie, 2023-10-06:
@chipweinberger, Same issue here. Thanks for the fix, though I haven't tried it yet.

chipweinberger, 2023-10-06:
1.16.3 & 1.16.4 were retracted due to this bug.

if some people test master branch ill push 1.16.5

TheFe91, 2023-10-06:
Just tested from master and works nicely

chipweinberger, 2023-10-06:
should be fixed, 1.16.5

closed 2023-10-06 by chipweinberger