chipweinberger/flutter_blue_plus #604: [Bug]:

rahul110897, 2023-10-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.4

### Flutter Version

3.13.6

### What OS?

Android

### OS Version

10

### Bluetooth Module

PSoC 6 Board

### What happened?

After setting the Notify value to true and reading characteristics, onValueReceived does not work

### Logs

```shell
logs are not coming
```

chipweinberger, 2023-10-06:
not enough details. cannot help you.

closed 2023-10-06 by chipweinberger

rahul110897, 2023-10-06:
In flutter_blue_plus  Version "1.16.2 " this functionality is working fine but in the "1.16.4" version not working.

chipweinberger, 2023-10-06:
i've retracted 1.16.4. use older version.

rahul110897, 2023-10-06:
In flutter_blue_plus Version "1.16.2 " this functionality is working fine but in the "1.16.3" version not working.

chipweinberger, 2023-10-06:
try master branch

rahul110897, 2023-10-06:
The issue "[Fix] Android: setNotify would time out if CCCD descriptor did not exist" has been resolved from your end. Have you implemented any additional code from our end?

chipweinberger, 2023-10-06:
> Means Which Plugin Version?

master branch. change pubspec.yaml to master branch. google for directions.

rahul110897, 2023-10-06:
I am not getting any platform

[image]

chipweinberger, 2023-10-06:
that is a problem with pub.dev.  There is a delay. wait 10 minutes.

rahul110897, 2023-10-06:
Ok thanks for the update and quick response we appreciate.