chipweinberger/flutter_blue_plus #607: [Help]:  After writing to a Bluetooth characteristic using characteristic.setNotifyValue(!characteristic.isNotifying), there are instances where data is not received in the onValueReceived stream.

faiyaz-shaikh, 2023-10-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.14.24

### Flutter Version

3.10.0

### What OS?

Android, iOS

### OS Version

android 12, iPadOS 16.6

### Bluetooth Module

Blu-Dex 

### Ask your question

## Issue Description

### Summary
After writing to a Bluetooth characteristic using `characteristic.setNotifyValue(!characteristic.isNotifying)`, there are instances where data is not received in the `onValueReceived` stream.

### Steps to Reproduce
1. Write to a Bluetooth characteristic using `characteristic.setNotifyValue(!characteristic.isNotifying)`.
2. Listen for characteristic updates using `characteristic.onValueReceived`.
3. Observe that sometimes data is not received after the write operation.

### Expected Behavior
The `onValueReceived` stream should consistently provide updates after a write operation on the Bluetooth characteristic.

### Code Sample
```dart
await characteristic.setNotifyValue(!characteristic.isNotifying);
streamSubscription = characteristic.onValueReceived.listen((value) async {
  // Switch cases for handling characteristic values
  switch (cases) {
    // Handle characteristic values
   // Characteristics Write operations
  }
});

### Logs

```shell
..
```

chipweinberger, 2023-10-06:
you did not read the README common problems!

chipweinberger, 2023-10-06:
## onValueReceived is never called

onValueReceived is only called for reads & notifies, not writes.

You can do a single read with await characteristic.read(...)

closed 2023-10-06 by chipweinberger