chipweinberger/flutter_blue_plus #608: [Help]:

shuizhuyu1015, 2023-10-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.10

### Flutter Version

3.3.2

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

nRF5

### Ask your question

I would like to ask the response logic of onValueReceived.
Is it necessary to call "read" after each "write" command, and then "onValueReceived" will receive the message returned by the device?

onValueReceived should have received a message from my device after each write command.
But now after sending the write command several times in a row, onValueReceived may lose the messages returned by the device.

### Logs

```shell
nothing
```

chipweinberger, 2023-10-08:
`write` does not trigger the callback.

it is only called when a value is received, not sent.

yes you could call `read` after each write to easily change this.

or you could create your own extension function `writeExtra` and your own stream `onValueReceivedOrWritten` and push to the stream yourself after each write.

chipweinberger, 2023-10-08:
I'm considering changing `lastValueStream` to do the behavior you want.

shuizhuyu1015, 2023-10-09:
> `write` does not trigger the callback.
> 
> it is only called when a value is received, not sent.
> 
> yes you could call `read` after each write to easily get the behavior you want.
> 
> or you could create your own extension function `writeExtra` and your own stream `onValueReceivedOrWritten` and push to the stream yourself after each write.

I did call `read` after `write` each time, but because I didn't wait for `onValueReceived` to respond, I `write`the next instruction and `read`, so I didn't receive the last write and read return messages.

chipweinberger, 2023-10-09:
Sorry, I don't understand you

but normally you should `await read()`, not `read()`

chipweinberger, 2023-10-09:
`lastValueStream` now emits values when write is called. please try it on master branch.

pubspec.yaml
```
dependencies:
  flutter_blue_plus:
    git:
      url: https://github.com/boskokg/flutter_blue_plus
      ref: master
```

shuizhuyu1015, 2023-10-09:
> `lastValueStream`现在，当调用 write 时会发出值。请在master分支上尝试一下！
> 
> pubspec.yaml
> 
> ```
> dependencies:
>   flutter_blue_plus:
>     git:
>       url: https://github.com/boskokg/flutter_blue_plus
>       ref: master
> ```

Ok, I'll try

shuizhuyu1015, 2023-10-09:
> master

I tried to introduce `master`, but `setNotifyValue` timed out.
`D/[FBP-Android](17473): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(17473): isLeEnabled(): ON
D/BluetoothLeScanner(17473): could not find callback wrapper
D/[FBP-Android](17473): [FBP] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](17473): [FBP] onMethodCall: startScan
D/BluetoothAdapter(17473): isLeEnabled(): ON
D/BluetoothLeScanner(17473): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/[FBP-Android](17473): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(17473): isLeEnabled(): ON
D/[FBP-Android](17473): [FBP] onMethodCall: connect
D/BluetoothGatt(17473): connect() - device: CF:5F:01:B9:D0:3F, auto: false
D/BluetoothGatt(17473): registerApp()
D/BluetoothGatt(17473): registerApp() - UUID=2790e058-a35c-4aab-9f53-2fcd1105f842
D/BluetoothGatt(17473): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt(17473): onClientConnectionState() - status=0 clientIf=8 device=CF:5F:01:B9:D0:3F
D/[FBP-Android](17473): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](17473): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(17473): discoverServices() - device: CF:5F:01:B9:D0:3F
D/BluetoothGatt(17473): onConnectionUpdated() - Device=CF:5F:01:B9:D0:3F interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(17473): onSearchComplete() = Device=CF:5F:01:B9:D0:3F Status=0
D/[FBP-Android](17473): [FBP] onServicesDiscovered: count: 4 status: 0
D/[FBP-Android](17473): [FBP] onMethodCall: setNotification
D/BluetoothGatt(17473): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
D/[FBP-Android](17473): [FBP] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/BluetoothGatt(17473): onConnectionUpdated() - Device=CF:5F:01:B9:D0:3F interval=39 latency=0 timeout=500 status=0
D/BluetoothGatt(17473): onConnectionUpdated() - Device=CF:5F:01:B9:D0:3F interval=60 latency=0 timeout=400 status=0
D/[FBP-Android](17473): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(17473): isLeEnabled(): ON
D/BluetoothLeScanner(17473): could not find callback wrapper
I/flutter (17473): catch error: FlutterBluePlusException: setNotifyValue: (code: 1) Timed out after 15s
`

chipweinberger, 2023-10-09:
hmm, maybe try again but stopScan before connect?

shuizhuyu1015, 2023-10-09:
Note that 1.16.12 is fine, but `setNotifyValue` still times out in master.

shuizhuyu1015, 2023-10-09:
There is another problem, when I click to turn off the Bluetooth adapter, turn it on again, and start scanning and connecting, the connection times out.

D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](25021): [FBP] disconnectAllDevices
D/[FBP-Android](25021): [FBP] calling disconnect: CF:5F:01:B9:D0:3F
D/BluetoothGatt(25021): cancelOpen() - device: CF:5F:01:B9:D0:3F
D/[FBP-Android](25021): [FBP] calling close: CF:5F:01:B9:D0:3F
D/BluetoothGatt(25021): close()
D/BluetoothGatt(25021): unregisterApp() - mClientIf=8
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](25021): [FBP] disconnectAllDevices
D/[FBP-Android](25021): [FBP] onMethodCall: disconnect
D/[FBP-Android](25021): [FBP] already disconnected
D/BluetoothAdapter(25021): onBluetoothServiceDown
D/[FBP-Android](25021): [FBP] onMethodCall: turnOn
D/[FBP-Android](25021): [FBP] onMethodCall: getAdapterState
I/BufferQueueProducer(25021): [SurfaceView[com.preova.yinyue/com.preova.yinyue.MainActivity]#7(BLAST Consumer)7](id:61bd00000007,api:1,p:25021,c:25021) disconnect: api 1
E/BLASTBufferQueue(25021): BLASTBufferItemConsumer::onDisconnect()
I/BufferQueueProducer(25021): [SurfaceView[com.preova.yinyue/com.preova.yinyue.MainActivity]#7(BLAST Consumer)7](id:61bd00000007,api:0,p:-1,c:25021) disconnect: api -1
I/BLASTBufferQueue(25021): [SurfaceView[com.preova.yinyue/com.preova.yinyue.MainActivity]#7] destructor()
I/BufferQueueConsumer(25021): [SurfaceView[com.preova.yinyue/com.preova.yinyue.MainActivity]#7(BLAST Consumer)7](id:61bd00000007,api:0,p:-1,c:25021) disconnect
I/BufferQueueProducer(25021): [ViewRootImpl[MainActivity]#6(BLAST Consumer)6](id:61bd00000006,api:1,p:25021,c:25021) disconnect: api 1
E/BLASTBufferQueue(25021): BLASTBufferItemConsumer::onDisconnect()
I/BLASTBufferQueue(25021): [ViewRootImpl[MainActivity]#6] destructor()
I/BufferQueueConsumer(25021): [ViewRootImpl[MainActivity]#6(BLAST Consumer)6](id:61bd00000006,api:0,p:-1,c:25021) disconnect
I/BLASTBufferQueue(25021): releaseBufferCallbackThunk bufferId:107464376713245 framenumber:1302 blastBufferQueue is dead
I/BLASTBufferQueue(25021): releaseBufferCallbackThunk bufferId:107464376713242 framenumber:1 blastBufferQueue is dead
D/BluetoothAdapter(25021): onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@60ef773
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: turningOn
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: on
I/BufferQueueConsumer(25021): [](id:61bd00000008,api:0,p:-1,c:25021) connect: controlledByApp=false
I/BLASTBufferQueue(25021): [ViewRootImpl[MainActivity]#8] constructor()
I/BufferQueueProducer(25021): [ViewRootImpl[MainActivity]#8(BLAST Consumer)8](id:61bd00000008,api:1,p:25021,c:25021) connect: api=1 producerControlledByApp=true
I/BufferQueueConsumer(25021): [](id:61bd00000009,api:0,p:-1,c:25021) connect: controlledByApp=false
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): could not find callback wrapper
D/[FBP-Android](25021): [FBP] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](25021): [FBP] onMethodCall: startScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/[FBP-Android](25021): [FBP] onMethodCall: connect
D/BluetoothGatt(25021): connect() - device: CF:5F:01:B9:D0:3F, auto: false
D/BluetoothGatt(25021): registerApp()
D/BluetoothGatt(25021): registerApp() - UUID=c34e5659-ca2e-4d98-9aed-32ef79b99257
D/BluetoothGatt(25021): onClientRegistered() - status=0 clientIf=8
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): could not find callback wrapper
I/flutter (25021): ！！！connect error！！！: FlutterBluePlusException: connect: (code: 1) Timed out after 12s
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): could not find callback wrapper
D/[FBP-Android](25021): [FBP] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](25021): [FBP] onMethodCall: discoverServices
E/flutter (25021): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discoverServices, device is disconnected, null, null)
D/BluetoothGatt(25021): onClientConnectionState() - status=133 clientIf=8 device=CF:5F:01:B9:D0:3F
D/[FBP-Android](25021): [FBP] onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected
D/BluetoothGatt(25021): close()
D/BluetoothGatt(25021): unregisterApp() - mClientIf=8

chipweinberger, 2023-10-09:
released 1.17.0

* **[Breaking Change/Fix]** should update `lastValue` & `lastValueStream` when `write()` is called

closed 2023-10-09 by chipweinberger