chipweinberger/flutter_blue_plus #609: [Bug]: onValueReceived not getting any values

skywa04885, 2023-10-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.10 icon indicating copy to clipboard operation

### Flutter Version

3.10.2

### What OS?

Android

### OS Version

13

### Bluetooth Module

ESP32

### What happened?

When using the onValueReceived stream of a characteristic, it never gets called, even when setNotifyValue(true) is used. The console _even_ logs the reception of these values. 

### Logs

```shell
Launching lib/main.dart on SM A137F in debug mode...
Running Gradle task 'assembleDebug'...
✓  Built build/app/outputs/flutter-apk/app-debug.apk.
Installing build/app/outputs/flutter-apk/app-debug.apk...
Debug service listening on ws://127.0.0.1:46201/U1-Nk9LUImw=/ws
Syncing files to device SM A137F...
D/[FBP-Android](29196): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](29196): [FBP] initializing BluetoothAdapter
I/BluetoothAdapter(29196): BluetoothAdapter() : com.github.skywa04885.powerfullittlefucker.powerfullittlefucker
I/BluetoothAdapter(29196): STATE_ON
I/BluetoothAdapter(29196): STATE_ON
D/BluetoothLeScanner(29196): could not find callback wrapper
D/[FBP-Android](29196): [FBP] disconnectAllDevices
D/[FBP-Android](29196): [FBP] connectedPeripherals: 0
D/[FBP-Android](29196): [FBP] onMethodCall: getAdapterState
D/hw-ProcessState(29196): Binder ioctl to enable oneway spam detection failed: Invalid argument
I/BLASTBufferQueue(29196): [SurfaceView[com.github.skywa04885.powerfullittlefucker.powerfullittlefucker/com.github.skywa04885.powerfullittlefucker.powerfullittlefucker.MainActivity]@0#1](f:0,a:0) onFrameAvailable the first frame is available
D/BLASTBufferQueue(29196): [SurfaceView[com.github.skywa04885.powerfullittlefucker.powerfullittlefucker/com.github.skywa04885.powerfullittlefucker.powerfullittlefucker.MainActivity]@0#1](f:0,a:1) acquireNextBufferLocked size=1080x2366 mFrameNumber=1 applyTransaction=true mTimestamp=129204380720546(auto) mPendingTransactions.size=0 graphicBufferId=125395865174016 transform=0
D/SurfaceView@270fc66(29196):  setAlpha: mUseAlpha = false alpha=1.0
D/SurfaceView@270fc66(29196):  updateSurfaceAlpha: setUseAlpha() is not called, ignored.
I/ViewRootImpl@8f2344a[MainActivity](29196): Setup new sync id=0
I/ViewRootImpl@8f2344a[MainActivity](29196): Setting syncFrameCallback
I/ViewRootImpl@8f2344a[MainActivity](29196): registerCallbacksForSync syncBuffer=false
I/ViewRootImpl@8f2344a[MainActivity](29196): mWNT: t=0xe954db60 mBlastBufferQueue=0xee1c6300 fn= 1 caller= android.view.ViewRootImpl.lambda$applyTransactionOnDraw$11$android-view-ViewRootImpl:12865 android.view.ViewRootImpl$$ExternalSyntheticLambda1.onFrameDraw:4 android.graphics.HardwareRenderer$FrameDrawingCallback.onFrameDraw:924 
I/ViewRootImpl@8f2344a[MainActivity](29196): Received frameDrawingCallback syncResult=0 frameNum=1.
D/SurfaceView@270fc66(29196): updateSurfacePosition RenderWorker, frameNr = 1, position = [0, 0, 1080, 2366] surfaceSize = 1080x2366
I/ViewRootImpl@8f2344a[MainActivity](29196): Setting up sync and frameCommitCallback
I/SurfaceView@270fc66(29196): uSP: rtp = Rect(0, 0 - 1080, 2366) rtsw = 1080 rtsh = 2366
I/SurfaceView@270fc66(29196): onSSPAndSRT: pl = 0 pt = 0 sx = 1.0 sy = 1.0
I/SurfaceView@270fc66(29196): aOrMT: ViewRootImpl@8f2344a[MainActivity] t = android.view.SurfaceControl$Transaction@2f27e68 fN = 1 android.view.SurfaceView.-$$Nest$mapplyOrMergeTransaction:0 android.view.SurfaceView$SurfaceViewPositionUpdateListener.positionChanged:1537 android.graphics.RenderNode$CompositePositionUpdateListener.positionChanged:373 
I/ViewRootImpl@8f2344a[MainActivity](29196): mWNT: t=0xe954dac0 mBlastBufferQueue=0xee1c6300 fn= 1 caller= android.view.SurfaceView.applyOrMergeTransaction:1455 android.view.SurfaceView.-$$Nest$mapplyOrMergeTransaction:0 android.view.SurfaceView$SurfaceViewPositionUpdateListener.positionChanged:1537 
I/BLASTBufferQueue(29196): [ViewRootImpl@8f2344a[MainActivity]#0](f:0,a:0) onFrameAvailable the first frame is available
D/BLASTBufferQueue(29196): [ViewRootImpl@8f2344a[MainActivity]#0](f:0,a:1) acquireNextBufferLocked size=1080x2408 mFrameNumber=1 applyTransaction=true mTimestamp=129204425682085(auto) mPendingTransactions.size=0 graphicBufferId=125395865174017 transform=0
I/ViewRootImpl@8f2344a[MainActivity](29196): Received frameCommittedCallback lastAttemptedDrawFrameNum=1 didProduceBuffer=true
D/OpenGLRenderer(29196): CFMS:: SetUp Pid : 29196    Tid : 29223
W/Parcel  (29196): Expecting binder but got null!
I/ViewRootImpl@8f2344a[MainActivity](29196): onSyncComplete
I/ViewRootImpl@8f2344a[MainActivity](29196): setupSync seqId=0 mSyncId=0 fn=1 caller=android.view.ViewRootImpl$$ExternalSyntheticLambda11.accept:6 android.window.SurfaceSyncer.lambda$setupSync$1$android-window-SurfaceSyncer:128 android.window.SurfaceSyncer$$ExternalSyntheticLambda1.accept:8 android.window.SurfaceSyncer$SyncSet.checkIfSyncIsComplete:382 android.window.SurfaceSyncer$SyncSet.markSyncReady:359 android.window.SurfaceSyncer.markSyncReady:151 android.view.ViewRootImpl.performTraversals:4503 
I/ViewRootImpl@8f2344a[MainActivity](29196): reportDrawFinished seqId=0 mSyncId=-1 fn=1 mSurfaceChangedTransaction=0xe9cf6ea0
[... 38 lines cut ...]
D/[FBP-Android](29196): [FBP] onServicesDiscovered: count: 3 status: 0
D/[FBP-Android](29196): [FBP] onMethodCall: setNotification
D/BluetoothGatt(29196): setCharacteristicNotification() - uuid: 222e975a-ce89-4081-9d91-c51aab4bec8d enable: true
W/[FBP-Android](29196): [FBP] CCCD descriptor for characteristic not found: 222e975a-ce89-4081-9d91-c51aab4bec8d
D/[FBP-Android](29196): [FBP] onMethodCall: setNotification
D/BluetoothGatt(29196): setCharacteristicNotification() - uuid: b20895d2-a647-46c7-b747-ea8b9ee354b2 enable: true
W/[FBP-Android](29196): [FBP] CCCD descriptor for characteristic not found: b20895d2-a647-46c7-b747-ea8b9ee354b2
D/[FBP-Android](29196): [FBP] onMethodCall: setNotification
D/BluetoothGatt(29196): setCharacteristicNotification() - uuid: 5a0ae25b-a6f0-444d-bb92-9b5bd74d78b7 enable: true
W/[FBP-Android](29196): [FBP] CCCD descriptor for characteristic not found: 5a0ae25b-a6f0-444d-bb92-9b5bd74d78b7
D/[FBP-Android](29196): [FBP] onMethodCall: setNotification
D/BluetoothGatt(29196): setCharacteristicNotification() - uuid: 25dc7e26-fceb-4a99-a8cb-bc811181888a enable: true
W/[FBP-Android](29196): [FBP] CCCD descriptor for characteristic not found: 25dc7e26-fceb-4a99-a8cb-bc811181888a
D/[FBP-Android](29196): [FBP] onMethodCall: setNotification
D/BluetoothGatt(29196): setCharacteristicNotification() - uuid: 3a77fd3d-1b97-4bf8-bd68-06a0c7a5b0f1 enable: true
W/[FBP-Android](29196): [FBP] CCCD descriptor for characteristic not found: 3a77fd3d-1b97-4bf8-bd68-06a0c7a5b0f1
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: 222e975a-ce89-4081-9d91-c51aab4bec8d
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: b20895d2-a647-46c7-b747-ea8b9ee354b2
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: 3a77fd3d-1b97-4bf8-bd68-06a0c7a5b0f1
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: 222e975a-ce89-4081-9d91-c51aab4bec8d
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: b20895d2-a647-46c7-b747-ea8b9ee354b2
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: 3a77fd3d-1b97-4bf8-bd68-06a0c7a5b0f1
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: 222e975a-ce89-4081-9d91-c51aab4bec8d
D/[FBP-Android](29196): [FBP] onCharacteristicChanged: uuid: b20895d2-a647-46c7-b747-ea8b9ee354b2
```

chipweinberger, 2023-10-08:
that's very odd!

perhaps you can try to debug it further!

add more print statements in various parts of the code? like in the FBP dart code that receives these messages.

skywa04885, 2023-10-08:
> that's very odd!
> 
> perhaps you can try to debug it further!
> 
> add more print statements in various parts of the code? like in the FBP dart code that receives these messages.

I tried printing, the result is just that none of the onData callbacks get called, only if i use the lastvaluestream it puts the last value from a read. However, all the streams get the data, if I call the appropriate reads. I tried looking in the FBP code, however it's a bit complex for me, I don't work with flutter too often.

chipweinberger, 2023-10-08:
just search the code base for OnValueReceived / onValueReceived and add some print statements around

chipweinberger, 2023-10-08:
also try the example app

phobosbag, 2023-10-08:
Same problem. 

My device is sending data, but my "listen" is not called.
I check c.isNotifying and it is True.
Then I add c.onValueReceived.listen and it is not called ever.
My device is sending data every minute and the logs show 
`D/[FBP-Android](26225): [FBP] onCharacteristicChanged: uuid: 6e400004-b5a3-f393-e0a9-e50e24dcca3e
D/[FBP-Android](26225): [FBP] onCharacteristicChanged: uuid: 6e400004-b5a3-f393-e0a9-e50e24dcca3e`

phobosbag, 2023-10-09:
I've been researching the problem.

This is where I get this error. Unfortunately it is not in the logs and it was hard to find it.
hosted\pub.dev\flutter_blue_plus-1.16.10\lib\src\bluetooth_msgs.dart 
line 434.
Unhandled Exception: type '_Uint8ArrayView' is not a subtype of type 'String'

**value: _hexDecode(json['value']),**
replaced by 
**value: json['value'] is String? _hexDecode(json['value']): json['value'],**
and everything worked like it used to.

It was called from here:
hosted\pub.dev\flutter_blue_plus-1.16.10\lib\src\flutter_blue_plus.dart 
line 345 (_methodCallHandler)

I'm using Android.
I'm not sure how it will work on iOS.

chipweinberger, 2023-10-09:
Ah yes, this regressed here:

https://github.com/boskokg/flutter_blue_plus/commit/cf78937a301c611c5aba4ebc1a8e576a94f4cfb9

due to a typo

chipweinberger, 2023-10-09:
fixed 1.16.11

closed 2023-10-09 by chipweinberger

chipweinberger, 2023-10-09:
1.16.12 actually

rickypid, 2023-10-17:
Thanks!! :ok_hand: