chipweinberger/flutter_blue_plus #610: [Help]: only on iOS can get onValueReceived return value

darkshut, 2023-10-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.10

### Flutter Version

3.13.6 

### What OS?

Android

### OS Version

android 12

### Bluetooth Module

arduino

### Ask your question

when run on iOS, can get onValueReceived return value (OnCharacteristicReceived), 
but on android can't get any reponse,
and if i use flutter_blue on android, it can get return value

### Logs

```shell
android log :
I/flutter (18625): sendMsg : {"Cmd":"Get_Login","Msg":"7acf2730-8b31-4b9c-b959-3fd8023b48f3","ATK":""}#
I/flutter (18625): [FBP] <writeCharacteristic> args: {remote_id: D4:D4:DA:59:5D:D6, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, secondary_service_uuid: null, characteristic_uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, write_type: 0, allow_long_write: 1, value: 7b22436d64223a224765745f4c6f67696e222c224d7367223a2237616366323733302d386233312d346239632d623935392d336664383032336234386633222c2241544b223a22227d23}
D/[FBP-Android](18625): [FBP] onMethodCall: writeCharacteristic
I/flutter (18625): [FBP] <writeCharacteristic> result: true
D/BluetoothGatt(18625): onConnectionUpdated() - Device=D4:D4:DA:59:5D:D6 interval=36 latency=0 timeout=500 status=0
D/[FBP-Android](18625): [FBP] onCharacteristicWrite: uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 0
I/flutter (18625): [FBP] [[ OnCharacteristicWritten ]] result: {secondary_service_uuid: null, error_string: GATT_SUCCESS, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, success: 1, remote_id: D4:D4:DA:59:5D:D6, error_code: 0, characteristic_uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e}
I/flutter (18625): send Msg success : {"Cmd":"Get_Login","Msg":"7acf2730-8b31-4b9c-b959-3fd8023b48f3","ATK":""}#
I/flutter (18625): send Msg complete : {"Cmd":"Get_Login","Msg":"7acf2730-8b31-4b9c-b959-3fd8023b48f3","ATK":""}#
I/flutter (18625): sendMsg success (<--- stop here)

iOS log :
flutter: sendMsg : {"Cmd":"Get_Login","Msg":"7acf2730-8b31-4b9c-b959-3fd8023b48f3","ATK":""}#
flutter: [FBP] <writeCharacteristic> args: {remote_id: 55E0FF94-7F54-5EEB-3E18-A85457E1089E, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, secondary_service_uuid: null, characteristic_uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, write_type: 0, allow_long_write: 1, value: 7b22436d64223a224765745f4c6f67696e222c224d7367223a2237616366323733302d386233312d346239632d623935392d336664383032336234386633222c2241544b223a22227d23}
[FBP-iOS] handleMethodCall: writeCharacteristic
flutter: [FBP] <writeCharacteristic> result: true
[FBP-iOS] didWriteValueForCharacteristic
[FBP-iOS] didUpdateValueForCharacteristic 55E0FF94-7F54-5EEB-3E18-A85457E1089E
flutter: [FBP] [[ OnCharacteristicWritten ]] result: {remote_id: 55E0FF94-7F54-5EEB-3E18-A85457E1089E, secondary_service_uuid: null, characteristic_uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, success: 1, error_code: null, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, error_string: null}
flutter: send Msg success : {"Cmd":"Get_Login","Msg":"7acf2730-8b31-4b9c-b959-3fd8023b48f3","ATK":""}#
flutter: send Msg complete : {"Cmd":"Get_Login","Msg":"7acf2730-8b31-4b9c-b959-3fd8023b48f3","ATK":""}#
flutter: sendMsg success
flutter: [FBP] [[ OnCharacteristicReceived ]] result: {remote_id: 55E0FF94-7F54-5EEB-3E18-A85457E1089E, secondary_service_uuid: null, characteristic_uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, success: 1, value: 7b22436d64223a224765745f4c6f67696e222c22526573223a2231222c224d73674a4f42223a7b2241757468546f6b656e223a2238303633313732353838227d7d23, error_code: null, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, error_string: null} (<---- can get return value)
```

chipweinberger, 2023-10-09:
fixed 1.16.11

closed 2023-10-09 by chipweinberger

darkshut, 2023-10-09:
sir,
i use 1.16.11 still can't solve the problem
i see you change log "[Fix] Android: onCharacteristicReceived not being called (regression 1.16.3)"
but i can't find 1.16.3 on https://pub.dev/packages/flutter_blue_plus/versions, 
so i use 1.16.2 to test
now android can get return value on 1.16.2
thank you very much

chipweinberger, 2023-10-09:
you're right. try 1.16.12

darkshut, 2023-10-09:
now i can use android get return value on 1.16.12
thank you sir