chipweinberger/flutter_blue_plus #613: [Bug]: turn off and turn on again, the device connection times out on Android.

shuizhuyu1015, 2023-10-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.17.0

### Flutter Version

3.3.2

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

Nordic nRF5

### What happened?

There is another problem, when I click to turn off the Bluetooth adapter, turn it on again, and start scanning and connecting, the connection times out.

### Logs

```shell
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](25021): [FBP] disconnectAllDevices
D/[FBP-Android](25021): [FBP] calling disconnect: CF:5F:01:B9:D0:3F
D/BluetoothGatt(25021): cancelOpen() - device: CF:5F:01:B9:D0:3F
D/[FBP-Android](25021): [FBP] calling close: CF:5F:01:B9:D0:3F
D/BluetoothGatt(25021): close()
D/BluetoothGatt(25021): unregisterApp() - mClientIf=8
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](25021): [FBP] disconnectAllDevices
D/[FBP-Android](25021): [FBP] onMethodCall: disconnect
D/[FBP-Android](25021): [FBP] already disconnected
D/BluetoothAdapter(25021): onBluetoothServiceDown
D/[FBP-Android](25021): [FBP] onMethodCall: turnOn
D/[FBP-Android](25021): [FBP] onMethodCall: getAdapterState
E/BLASTBufferQueue(25021): BLASTBufferItemConsumer::onDisconnect()
E/BLASTBufferQueue(25021): BLASTBufferItemConsumer::onDisconnect()
I/BLASTBufferQueue(25021): releaseBufferCallbackThunk bufferId:107464376713245 framenumber:1302 blastBufferQueue is dead
I/BLASTBufferQueue(25021): releaseBufferCallbackThunk bufferId:107464376713242 framenumber:1 blastBufferQueue is dead
D/BluetoothAdapter(25021): onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@60ef773
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: turningOn
D/[FBP-Android](25021): [FBP] OnAdapterStateChanged: on
I/BufferQueueConsumer(25021): [](id:61bd00000008,api:0,p:-1,c:25021) connect: controlledByApp=false
I/BLASTBufferQueue(25021): [ViewRootImpl[MainActivity]#8] constructor()
I/BufferQueueProducer(25021): [ViewRootImpl[MainActivity]#8(BLAST Consumer)8](id:61bd00000008,api:1,p:25021,c:25021) connect: api=1 producerControlledByApp=true
I/BufferQueueConsumer(25021): [](id:61bd00000009,api:0,p:-1,c:25021) connect: controlledByApp=false
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): could not find callback wrapper
D/[FBP-Android](25021): [FBP] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](25021): [FBP] onMethodCall: startScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/[FBP-Android](25021): [FBP] onMethodCall: connect
D/BluetoothGatt(25021): connect() - device: CF:5F:01:B9:D0:3F, auto: false
D/BluetoothGatt(25021): registerApp()
D/BluetoothGatt(25021): registerApp() - UUID=c34e5659-ca2e-4d98-9aed-32ef79b99257
D/BluetoothGatt(25021): onClientRegistered() - status=0 clientIf=8
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): could not find callback wrapper
I/flutter (25021): ！！！connect error！！！: FlutterBluePlusException: connect: (code: 1) Timed out after 12s
D/[FBP-Android](25021): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(25021): isLeEnabled(): ON
D/BluetoothLeScanner(25021): could not find callback wrapper
D/[FBP-Android](25021): [FBP] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](25021): [FBP] onMethodCall: discoverServices
E/flutter (25021): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discoverServices, device is disconnected, null, null)
D/BluetoothGatt(25021): onClientConnectionState() - status=133 clientIf=8 device=CF:5F:01:B9:D0:3F
D/[FBP-Android](25021): [FBP] onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected
D/BluetoothGatt(25021): close()
D/BluetoothGatt(25021): unregisterApp() - mClientIf=8
```

chipweinberger, 2023-10-10:
fixed 1.17.1

thanks for reporting

closed 2023-10-10 by chipweinberger

lorenzosfarra, 2023-10-28:
Hi, 

I'm sorry for commenting on a closed issue, but I think it is closely related to my concern.

I am using the v.1.20.7 version but it seems that I am suffering the same problem?

```
  flutter_blue_plus:
    dependency: "direct main"
    description:
      name: flutter_blue_plus
      sha256: d8eaa4af1a0283886e7b5b198a15cfc14a9977d12e2ef53bc887185431e01420
      url: "https://pub.dev"
    source: hosted
    version: "1.20.7"
```

Flutter 3.13.8
Android 12

### Turning off logs

```
D/BluetoothGatt(19598): onClientConnectionState() - status=0 clientIf=8 device=FA:C2:54:8D:2B:78
D/BluetoothGatt(19598): close()
D/BluetoothGatt(19598): close()
D/BluetoothGatt(19598): unregisterApp() - mClientIf=8
D/BluetoothGatt(19598): unregisterApp() - mClientIf=8
D/BluetoothGatt(19598): close()
D/BluetoothGatt(19598): unregisterApp() - mClientIf=0
D/[FBP-Android](19598): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](19598): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](19598): [FBP] calling disconnect: FA:C2:54:8D:2B:78
D/BluetoothGatt(19598): cancelOpen() - device: FA:C2:54:8D:2B:78
D/[FBP-Android](19598): [FBP] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/[FBP-Android](19598): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(19598): close()
D/BluetoothGatt(19598): unregisterApp() - mClientIf=9
D/[FBP-Android](19598): [FBP] calling close: FA:C2:54:8D:2B:78
D/BluetoothGatt(19598): close()
D/BluetoothGatt(19598): unregisterApp() - mClientIf=0
I/flutter (19598): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: FA:C2:54:8D:2B:78, connection_state: 0}
I/flutter (19598): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 5}
D/[FBP-Android](19598): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](19598): [FBP] disconnectAllDevices(adapterTurnOff)
I/flutter (19598): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 6}
D/BluetoothAdapter(19598): onBluetoothServiceDown
D/BluetoothAdapter(19598): onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@e6c3873
```

### Turning on logs

```
D/[FBP-Android](19598): [FBP] OnAdapterStateChanged: turningOn
I/flutter (19598): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 3}
D/[FBP-Android](19598): [FBP] OnAdapterStateChanged: on
I/flutter (19598): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 4}
I/flutter (19598): [FBP] <getSystemDevices> args: null
D/[FBP-Android](19598): [FBP] onMethodCall: getSystemDevices
I/flutter (19598): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (19598): [FBP] <getSystemDevices> args: null
D/[FBP-Android](19598): [FBP] onMethodCall: getSystemDevices
I/flutter (19598): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (19598): [FBP] <getSystemDevices> args: null
D/[FBP-Android](19598): [FBP] onMethodCall: getSystemDevices
I/flutter (19598): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (19598): [FBP] <getSystemDevices> args: null
D/[FBP-Android](19598): [FBP] onMethodCall: getSystemDevices
I/flutter (19598): [FBP] <getSystemDevices> result: {devices: []}

[... 50 lines cut ...]

////// If the device is still null, only on Android I try to do it from the remoteId

if (Platform.isAndroid) {
  newlyConnectedDevice = BluetoothDevice.fromId(deviceId!);
}

//// Listeners for device connection state
connectedDevice.connectionState.listen((bluetoothState) {
 if (bluetoothState == BluetoothConnectionState.connected)  onDeviceConnected(connectedDevice);
 else onDeviceDisconnected();

/// onDeviceConnected
deviceName = device!.platformName.isNotEmpty ? device!.platformName : '--';
deviceId = device!.remoteId.str;
service = await bluetoothDeviceGetMainService(device!); // Discover services
Map<Guid, BluetoothCharacteristic> characteristics = bluetoothDeviceGetCharacteristics(service!); // Discover characteristics
/// Set characteristic listeners
```

----

At one point I can see `I/flutter (19598): [FBP] <connect> result: true` that seems promising, so I wonder if I am doing something wrong in the process.

What do you think?

chipweinberger, 2023-10-30:
10s timeout is too short. use 30s