chipweinberger/flutter_blue_plus #614: [Help]: onValueReceived.listen() not triggering for some reason.

JEAPI-DEV, 2023-10-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### FlutterBluePlus Version

1.16.10

### Flutter Version

Framework • revision 796c8ef792 (4 months ago) • 2023-06-13 15:51:02 -0700 Engine • revision 45f6e00911 Tools • Dart 3.0.5 • DevTools 2.23.1

### What OS?

Android

### OS Version

Android 12 SKQ1.210908.001

### Bluetooth Module

TI CC2541

### Ask your question

Hi everyone and thanks for the help in advance.

So the app works fine and the app can connect, send messages, and scan for them. The problem is that for some reason the listener does not trigger I have tried debugging 5 hours and finding the issue however I came to no answer why it does not trigger.

To note is that the sendMessageWResponse (send Message with Response) does work and it does get a response, however that only works one time. When I call it a second time a new instance gets created and the Received message does get printed 2 times then 3 times and so on. I wouldn't need the sendMessageWResponse method if the startReceivingMessages method would work correctly.

I did test if the Arduino code from the device I'm connecting to sends the messages correctly and the messages do pop up correctly in the "Serial Bluetooth Terminal" App.

bluetoothmanager.dart:
```
import 'dart:convert';
import 'dart:io';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';

class BluetoothManager {
  late BluetoothDevice device;
  BluetoothCharacteristic? _selectedCharacteristic;

  String _messagebox = "";
  String _messagebuffer = "";
  String get messagebox => _messagebox;
  String get messagebuffer => _messagebuffer;
  BluetoothManager(this.device);

  Future<void> selectCharacteristic() async {
    await device.connect();
    List<BluetoothService> services = await device.discoverServices();
    for (BluetoothService service in services) {
      for (BluetoothCharacteristic characteristic in service.characteristics) {
        print('Characteristic: ${characteristic.uuid}, Properties: ${characteristic.properties}');
        if (characteristic.properties.write && characteristic.properties.notify) {
          _selectedCharacteristic = characteristic;
          break;
        } else {
          print('Characteristic ${characteristic.uuid} does not support write operation');
        }
      }
    }
    startReceivingMessages();
  }

  Future<void> sendMessage(String message) async {
    List<int> bytes = utf8.encode(message);
    try {
[... 351 lines cut ...]
I/flutter (30500): Received message: RX=22.10|64.00
I/flutter (30500): Read value: RX=22.10|64.00
D/InputMethodManager(30500): showSoftInput() view=io.flutter.embedding.android.FlutterView{2b2b7bd VFE...... .F....ID 0,0-1080,2270 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
I/AssistStructure(30500): Flattened final assist data: 412 bytes, containing 1 windows, 3 views
D/InputConnectionAdaptor(30500): The input method toggled cursor monitoring on
D/InsetsController(30500): show(ime(), fromIme=true)
I/OpenGLRenderer(30500): Davey! duration=9223113330873ms; Flags=0, FrameTimelineVsyncId=14198211, IntendedVsync=258705980901048, Vsync=258705980901048, InputEventId=0, HandleInputStart=258705982254744, AnimationStart=258705982269119, PerformTraversalsStart=258705982278442, DrawStart=258705993588754, FrameDeadline=258706004901047, FrameInterval=258705982164744, FrameStartTime=8333333, SyncQueued=258705994622348, SyncStart=258705994710161, IssueDrawCommandsStart=258705994827400, SwapBuffers=258705995196046, FrameCompleted=9223372036854775807, DequeueBufferDuration=19635, QueueBufferDuration=552136, GpuCompleted=9223372036854775807, SwapBuffersCompleted=258705996136671, DisplayPresentTime=0,
D/InputMethodManager(30500): showSoftInput() view=io.flutter.embedding.android.FlutterView{2b2b7bd VFE...... .F...... 0,0-1080,2270 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController(30500): show(ime(), fromIme=true)
D/InputMethodManager(30500): showSoftInput() view=io.flutter.embedding.android.FlutterView{2b2b7bd VFE...... .F...... 0,0-1080,2270 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController(30500): show(ime(), fromIme=true)
D/InputMethodManager(30500): showSoftInput() view=io.flutter.embedding.android.FlutterView{2b2b7bd VFE...... .F...... 0,0-1080,2270 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController(30500): show(ime(), fromIme=true)
D/InputMethodManager(30500): showSoftInput() view=io.flutter.embedding.android.FlutterView{2b2b7bd VFE...... .F...... 0,0-1080,2270 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController(30500): show(ime(), fromIme=true)
D/[FBP-Android](30500): [FBP] onMethodCall: connect
D/[FBP-Android](30500): [FBP] already connected
D/[FBP-Android](30500): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](30500): [FBP] onCharacteristicWrite: uuid: 0000ffe1-0000-1000-8000-00805f9b34fb status: 0
I/flutter (30500): Message sent: test
D/MIUIInput(30500): [KeyEvent] ViewRootImpl KeyEvent { action=ACTION_DOWN, keyCode=KEYCODE_BACK, scanCode=0, metaState=0, flags=0x48, repeatCount=0, eventTime=258709890, downTime=258709890, deviceId=-1, source=0x101, displayId=0 }
D/MIUIInput(30500): [KeyEvent] ViewRootImpl KeyEvent { action=ACTION_UP, keyCode=KEYCODE_BACK, scanCode=0, metaState=0, flags=0x48, repeatCount=0, eventTime=258709954, downTime=258709890, deviceId=-1, source=0x101, displayId=0 }
D/InputConnectionAdaptor(30500): The input method toggled cursor monitoring off
I/OpenGLRenderer(30500): Davey! duration=9223113326882ms; Flags=0, FrameTimelineVsyncId=14198571, IntendedVsync=258709972565070, Vsync=258709972565070, InputEventId=-110327805, HandleInputStart=258709974895940, AnimationStart=258709974906357, PerformTraversalsStart=258709974917295, DrawStart=258709994018961, FrameDeadline=258709988231736, FrameInterval=258709974794013, FrameStartTime=8333333, SyncQueued=258709994911930, SyncStart=258709995071617, IssueDrawCommandsStart=258709995198180, SwapBuffers=258709995823024, FrameCompleted=9223372036854775807, DequeueBufferDuration=43021, QueueBufferDuration=1515573, GpuCompleted=9223372036854775807, SwapBuffersCompleted=258709997813649, DisplayPresentTime=0,
```

JEAPI-DEV, 2023-10-10:
Nvm solved it just needed a cup of coffee. I simply changed a few and made a special method for reading. I'm gonna leave this open in case someone has the same problems I had. Here is the code for the bluetoothmanager.dart:  
```
import 'dart:async';
import 'dart:convert';
import 'dart:io';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';

class BluetoothManager {
  late BluetoothDevice device;
  BluetoothCharacteristic? _selectedCharacteristic;

  String _messagebox = "";
  String _messagebuffer = "";
  String get messagebox => _messagebox;
  String get messagebuffer => _messagebuffer;
  bool run = true;
  late StreamSubscription<List<int>> listener;

  BluetoothManager(this.device);

  Future<void> selectCharacteristic() async {
    await device.connect();
    List<BluetoothService> services = await device.discoverServices();
    for (BluetoothService service in services) {
      for (BluetoothCharacteristic characteristic in service.characteristics) {
        print('Characteristic: ${characteristic.uuid}, Properties: ${characteristic.properties}');
        if (characteristic.properties.write && characteristic.properties.notify) {
          _selectedCharacteristic = characteristic;
          break;
        } else {
          print('Characteristic ${characteristic.uuid} does not support write operation');
        }
      }
    }
    await startReceivingMessages();
    updateMesasges();
  }

  // dispose
  void dispose() {
    run = false;
    listener.cancel();
  }

  Future<void> sendMessage(String message) async {
    List<int> bytes = utf8.encode(message);
    try {
      if (device.connectionState != BluetoothDeviceState.connected) {
        await device.connect();
      }
      await _selectedCharacteristic!.write(bytes, withoutResponse: false);
      print("Message sent: $message");
    } catch (e) {
      print("Failed to send message: $e");
    }
  }

  // update value
  Future<void> updateMesasges() async {
    while(run){
      try {
        await Future.delayed(Duration(seconds: 1));
        List<int> readValue = await _selectedCharacteristic!.read();
        print("TTTXW : " + utf8.decode(readValue));
        //_messagebuffer = utf8.decode(readValue);
        print("Message buffer: $_messagebuffer");
      } catch (e) {
        print("Failed to read value: $e");
      }
    }
  }

  Future<void> startReceivingMessages() async {
    try {
      if (Platform.isAndroid) {
          await device.requestMtu(223);
      }

      if (_selectedCharacteristic != null) {
        listener = _selectedCharacteristic!.onValueReceived.listen((value) async {
            print("Raw value received: $value");
            if (value.isNotEmpty) {
              String message = utf8.decode(value);
              _messagebuffer = "$message\n";
               // Add a read operation
              print("Received message: $message");
            } else {
              print("Received empty value");
            }
        });

        await _selectedCharacteristic!.setNotifyValue(true);
        // remove 3 first characters from string and return the rest (TX=)
      } else {
        print("No characteristic selected for receiving messages");
      }
    } catch (e) {
      print("Failed to receive messages: $e");
    }
  }

  void stopReceivingMessages() {
   _selectedCharacteristic?.value.drain();
  }
}

```

chipweinberger, 2023-10-10:
glad you figured it out.

I will close this so we can find bugs that need to be fixed.

closed 2023-10-10 by chipweinberger

manish-tiwari7355, 2024-01-31:
Thanks, you save my lots of time