chipweinberger/flutter_blue_plus #615: [Help]: iOS supervision timeout set to 72 vs 500 on Android

sqcsabbey, 2023-10-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you tried the example app?

Yes

### FlutterBluePlus Version

1.14.16

### Flutter Version

3.13.4

### What OS?

iOS

### OS Version

16.7

### Bluetooth Module

Espressif Esp32-c3 using SDK 5.0 and the NimBLE stack

### What happened?

We get unexpected disconnects on iOS, and when this happens we see that the supervision timeout is set to 720ms instead of the 5000ms we see for Android. Is there a way to increase this value for iOS?

Here is the log line (from an iOS connection, also pasted into the log section) that we get with an unexpected disconnect: 

> I (12951) BTLE: conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

### Logs

```shell
E (12890) COP_task: Wrote: mode 0xc5, height 0, hAcc 0 , vAcc 0
W (12890) ublox_update_config: ======================================== Received Updated mode Configuration [0xc5] ========================================
E (12890) ublox_update_config: ================>>> Arg: [197 0xc5] Current mode: [69 0x45] New mode: [133 0x85]
E (12891) ublox_update_config: NOT Writing LLH
E (12891) ublox_update_config: MODE: BASE FIXED
E (12891) BTLE: Tried to stop a periodic sync while none was running
&&&&&&&&&&&&&&&&&&&&&&&&&&  td.lb_config.mode.bit.Valid_or_Not_Busy set to: [0, 0x05] &&&&&&&&&&&&&&&&&&&&&&&&&&
E (12892) ublox_update_config: Set_FIXED_BASE_Flag = true
I (12892) BTLE: Starting periodic advertising
I (12893) BTLE: Periodic Advertising started (returned 0)
E (12893) ublox_update_config:   BROADCAST: ENABLED [ bleprph_broadcast_start() ]
W (12894) lb_broadcast_task: Time: 230, RTCM ID: 1005, Length-6: 19, CRC24Q: 702413 [        0]
I (12948) BTLE: subscribe event; conn_handle=1 attr_handle=8 reason=2 prevn=0 curn=0 previ=1 curi=0
I (12948) BTLE: subscribe event; conn_handle=1 attr_handle=33 reason=2 prevn=1 curn=0 previ=0 curi=0
I (12949) BTLE: COP notifications disabled
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> BLE_GAP_EVENT_DISCONNECT [0x85] <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
I (12949) BTLE: disconnect; reason=531 0x0213
I (12949) BTLE: handle=1 our_ota_addr_type=0 our_ota_addr=34:b4:72:42:4f:92
I (12950) BTLE: our_id_addr_type=0 our_id_addr=34:b4:72:42:4f:92
I (12950) BTLE: peer_ota_addr_type=1 peer_ota_addr=69:9e:b7:92:cf:b3
I (12950) BTLE: peer_id_addr_type=1 peer_id_addr=69:9e:b7:92:cf:b3
I (12951) BTLE: conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0
&&&&&&&&&&&&&&&&&&&&&&&&&&  td.lb_config.mode.bit.Valid_or_Not_Busy set to: [0, 0x85] &&&&&&&&&&&&&&&&&&&&&&&&&&
W (12951) BTLE: >>>>>>>>>>>>>>>>>>>>>>>>>>>>>> (not td.lb_config.mode.bit.Broadcast_Enable) [0x85] <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> ble_gap_ext_adv_start <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
E (12982) UBX_TASK_TAG: < Set_FIXED_BASE_Flag > : <12400, 12400>
I (13182) UBX_TASK_TAG: [   516]---------------------------------------------------------------------------------------------------------
```

chipweinberger, 2023-10-10:
> Is there a way to increase this value for iOS?

No. But look here:

https://stackoverflow.com/questions/74693600/ios16-1-updated-ble-connection-interval-from-ios16-0#:~:text=Supervision%20Timeout%20from%202%20seconds,a%20multiple%20of%2015%20ms.

chipweinberger, 2023-10-10:
you need to request a larger value in the espressif code

```
static int handle_gap_event(struct ble_gap_event *event, void *arg)
{
    // always be advertising.
    start_gap_advertise();

    switch (event->type)
    {
        case BLE_GAP_EVENT_CONNECT: {

            if (event->connect.status == 0) {

                ESP_LOGI(TAG, "GAP connection: established. gapConnHandle: %" PRIu16 "", 
                    event->connect.conn_handle);

                struct ble_gap_conn_desc desc;
                int rc = ble_gap_conn_find(event->connect.conn_handle, &desc);   
                if (rc != 0) {
                    // raise issue
                }

                ble_gap_upd_params params = {
                    .itvl_min = 6, // Minimum connection interval (1.25ms units) 
                    .itvl_max = 12, // Maximum connection interval (1.25ms units)
                    .latency = 0, // Connection latency
                    .supervision_timeout = 500, // Supervision timeout (10ms units)
                    .min_ce_len = 0xFFFF, // Minimum connection event length (0.625ms units). Longer is faster.
                    .max_ce_len = 0xFFFF, // Maximum connection event length (0.625ms units). Longer is faster.
                };

                if(ble_gap_update_params(event->connect.conn_handle, &params) != 0) {
                    ESP_LOGE(TAG, "ble_gap_update_params() failed. 0x%04x %s, %s",
                        rc, ble_rc_category_str(rc), ble_rc_str(rc));
                }
```

chipweinberger, 2023-10-10:
> In our case, the app using this plugin is the master (central), so I think it would need to set this value?

The peripheral can suggest values, but iOS will refuse connection if it does not like them. See the esspressif code above.

sqcsabbey, 2023-10-10:
Thank you, we'll give that code a shot!

closed 2023-10-10 by chipweinberger