chipweinberger/flutter_blue_plus #616: [Bug]: Getting current MTU size hangs

jasaw, 2023-10-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.17.0

### Flutter Version

3.13.6

### What OS?

Android

### OS Version

7

### Bluetooth Module

Espressif ESP32

### What is your problem?

Getting the current MTU size via `await bleDevice.mtu.first` used to work, but hangs on version 1.17.0.
The example code method of subscribing to the MTU change stream `bleDevice.mtu.listen` before making a Bluetooth connection works.

### Logs

```shell
D/[FBP-Android]( 4727): [FBP] onMethodCall: setNotification
D/BluetoothGatt( 4727): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
D/[FBP-Android]( 4727): [FBP] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
I/flutter ( 4727): 0:00:29.383690	I	BleDevice:	 *** Getting current MTU size ...
D/ViewRootImpl@19335d4[MainActivity]( 4727): Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x1 surface={valid=true 505294397440} changed=false
```

chipweinberger, 2023-10-11:
I am not able to reproduce this issue.

Can you modify the example app to get the issue to reproduce?

chipweinberger, 2023-10-11:
in the code we always emit a value immediately, and fallback to 23 if we dont have a value yet.

It's possible `newStreamWithInitialValue` is broken, but we have not changed anything with that code in awhile.

```
  /// The current MTU size in bytes
  Stream<int> get mtu {
    // get initial value from our cache
    int initialValue = FlutterBluePlus._mtuValues[remoteId]?.mtu ?? 23;
    return FlutterBluePlus._methodStream.stream
        .where((m) => m.method == "OnMtuChanged")
        .map((m) => m.arguments)
        .map((args) => BmMtuChangedResponse.fromMap(args))
        .where((p) => p.remoteId == remoteId.str)
        .map((p) => p.mtu)
        .newStreamWithInitialValue(initialValue);
  }
```

jasaw, 2023-10-11:
I can confirm that the hanging behaviour started from version 1.16.6.
I haven't tried the example app.

chipweinberger, 2023-10-11:
the hanging behavior could be some issue in your app.

I do not plan to change any code at this point.

chipweinberger, 2023-10-11:
You're right, there is an issue. I can reproduce it with this:

```
  // Testing for broadcast streams
  for (var i = 0; i < 10000; i++) {
    var stream = Stream<int>.fromIterable([1, 2, 3, 4, 5]).asBroadcastStream();
    var transformedStream1 = stream.newStreamWithInitialValue(0);
    int firstValue = await transformedStream1.first;
    print(firstValue);
    assert(firstValue == 0);
  }
```

closed 2023-10-11 by chipweinberger