chipweinberger/flutter_blue_plus #617: [Help]: Requesting and Checking Permissions

kbessemer, 2023-10-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.17.0

### Flutter Version

3.13.5

### What OS?

iOS

### OS Version

iOS 17

### Bluetooth Module

STM 32

### What is your problem?

I am having trouble checking permissions. The request is succeeding but the check fails every time.

I am using the permission handler package, is there a better way?

### Logs

```shell
`if (!await Permission.bluetoothScan.isGranted && !await Permission.bluetoothConnect.isGranted) {
... send to new screen where permissions are requested 
}`

`if (!await Permission.bluetoothScan.isGranted) {
      PermissionStatus bleScan = await Permission.bluetoothScan.request();
      if (bleScan == false) {
        return false;
      }
    }
    if (!await Permission.bluetoothConnect.isGranted) {
      PermissionStatus bleConnect = await Permission.bluetoothConnect
          .request();
      if (bleConnect == false) {
        return false;
      }
    }`
```

chipweinberger, 2023-10-11:
You do not need to use the permission handler package.

Just use flutter blue plus, we ask for permissions *as needed*. i.e. when you call scan, connect, etc.

chipweinberger, 2023-10-11:
I've updated the example app to remove PermissionHandler =)

chipweinberger, 2023-10-11:
I've pushed 1.17.4 to remove PermissionHandler in the example app =)

closed 2023-10-11 by chipweinberger

Zeeshan0201, 2024-10-15:
I was facing the same issue,thanks