chipweinberger/flutter_blue_plus #618: [Help]: Can Flutter Blue Plus run in the killed state

Shashank02051997, 2023-10-12:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.17.5

### Flutter Version

3.13.1

### What OS?

Android, iOS

### OS Version

iOS: 16.6 and greater and Android: 10 and greater

### Bluetooth Module

Using Smart Band

### What is your problem?

The example provided in the repository works well on both Android and iOS platforms. However, I've noticed that I can only connect to a Bluetooth device when the app is in the foreground. When the app is killed, it disconnects from the device. Is there a solution to maintain a connection with a Bluetooth device even when the app is in a killed state?

### Logs

```shell
N/A
```

chipweinberger, 2023-10-12:
I think I've seen it done on android. Not sure about iOS.

The iOS/Android documentation will have more info.

Shashank02051997, 2023-10-12:
@chipweinberger Can you help me with Android, at least if you know someone who has done it before? Please tag that user here, or if you have a code snippet or anything that could help me connect to the device in the killed state, it would be greatly appreciated

Shashank02051997, 2023-10-12:
@chipweinberger Thank you so much for the quick response and for taking the time to find the closed pull request for me

closed 2023-10-12 by Shashank02051997