chipweinberger/flutter_blue_plus #620: [Help]: Why can scan the device after it is powered off?

shuizhuyu1015, 2023-10-12:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.17.3

### Flutter Version

3.3.2

### What OS?

Android, iOS

### OS Version

Android 12, iOS 16.6

### Bluetooth Module

Nordic nRF5

### What is your problem?

After the device is disconnected automatically, `startScan()`is called again, and the device that has been powered off is still returned in `scanResults`.

Only after `hot restart`,  `scanResults` show no devices that have been powered off.

### Logs

```shell
no logs
```

chipweinberger, 2023-10-12:
try example app

shuizhuyu1015, 2023-10-13:
The example app does not scan the device after it is powered off, but my own App can scan the device after it is powed off.
I don't know why, upset.😭

chipweinberger, 2023-10-13:
fixed 1.17.6

closed 2023-10-13 by chipweinberger