chipweinberger/flutter_blue_plus #622: [Bug]: Problem with decoding uuid of Microchip BM70

widzinskaj, 2023-10-12:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.17.5

### Flutter Version

3.13.6

### What OS?

Android

### OS Version

13

### Bluetooth Module

Microchip BM70

### What is your problem?

After updating the library I obtain different characteristic decoded through library than the Microchip BM70 characteristic.

Defined uuids:
`final Guid uartuuid =
Guid('0000ff00-0000-1000-8000-00805f9b34fb'); 
final Guid uartrx =
Guid('0000ff01-0000-1000-8000-00805f9b34fb');
final Guid uarttx =
Guid('0000ff02-0000-1000-8000-00805f9b34fb'); 
final Guid newbmsuuid =
Guid('49535343-fe7d-4ae5-8fa9-9fafd205e455'); 
final Guid newbmsrx = Guid('49535343-1e4d-4bd9-ba61-23c647249616');
final Guid newbmstx = Guid('49535343-8841-43f4-a8d4-ecbe34729bb3');
final Guid sticanuuid =
Guid('49535343-f09d-4ae5-8fa9-9fafd205e455');
final Guid sticanchar = Guid('49535343-109d-4bd9-ba61-23c647249616');`

Function discoverServices:
`Future<void> discoverServices() async {
    const maxRetries = 5;
    bool servicesDiscovered = false;

    for (int retries = 0; retries < maxRetries && !servicesDiscovered; retries++) {
      try {
        myList2 = await widget.device.discoverServices();
        servicesDiscovered = true;
      } catch (e) {
        if (kDebugMode) {
          print('Cannot discover services' + e.toString());
        }
        await Future.delayed(Duration(seconds: 1));
      }
    }

    await Future.delayed(const Duration(seconds: 1));

    if (myList2.isEmpty) {
      print('Cannot discover services - empty list');
    } else {
      print('Service discovered - connecting to characteristic');

      print(myList2[0].uuid);
      print(myList2[0].characteristics);
      //print(myList2[0].includedServices.toString());
      print(myList2[0].serviceUuid);

      for (BluetoothCharacteristic c in myList2[0].characteristics) {
        print('data check: ');
        print(c.uuid);
        print(c.characteristicUuid);
        //print(c.descriptors.toString());
      }

      for (BluetoothCharacteristic c in myList2[0].characteristics) {
        print('Processing characteristic started');
        processCharacteristic(c);
      }
    }
  }

  void processCharacteristic(BluetoothCharacteristic c) {
    print('Checking characteristic');

    if (c.uuid.toString() == uartrx.toString()) {
      print('UARTRX');
      processUartRxCharacteristic(c);
    } else if (c.uuid.toString() == uarttx.toString()) {
      print('UARTTX');
      processUartTxCharacteristic(c);
    } else if (c.uuid.toString() == newbmsrx.toString()) {
      print('NEWBMSRX');
      processNewBmsRxCharacteristic(c);
    } else if (c.uuid.toString() == newbmstx.toString()) {
      print('NEWBMSTX');
      processNewBmsTxCharacteristic(c);
    }
  }`
In logs bluetooth characteristics are different than the hardware ones.

### Logs

```shell
Launching lib\main.dart on moto e13 in debug mode...
Running Gradle task 'assembleDebug'...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
Debug service listening on ws://127.0.0.1:52068/EedFd1ESfjQ=/ws
Syncing files to device moto e13...
D/[FBP-Android](18339): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](18339): [FBP] initializing BluetoothAdapter
D/BluetoothAdapter(18339): isLeEnabled(): ON
D/BluetoothLeScanner(18339): could not find callback wrapper
D/[FBP-Android](18339): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](18339): [FBP] connectedPeripherals: 0
D/[FBP-Android](18339): [FBP] onMethodCall: getAdapterState
I/mali_gralloc(18339): register: id=0x26b00002b47, importpid=-1
I/mali_gralloc(18339): register: id=0x26b00002b48, importpid=-1
W/Parcel  (18339): Expecting binder but got null!
W/ziparchive(18339): Unable to open '/system_ext/framework/unipnp-framework.dm': No such file or directory
W/ziparchive(18339): Unable to open '/system_ext/framework/unipnp-framework.dm': No such file or directory
I/mali_gralloc(18339): register: id=0x26b00002b49, importpid=-1
D/[FBP-Android](18339): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](18339): [FBP] onMethodCall: startScan
D/BluetoothAdapter(18339): isLeEnabled(): ON
D/BluetoothLeScanner(18339): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
I/mali_gralloc(18339): register: id=0x26b00002b4a, importpid=-1
I/flutter (18339): Cannot discover servicesFlutterBluePlusException: discoverServices: (code: 6) device is not connected
D/[FBP-Android](18339): [FBP] onMethodCall: connect
D/BluetoothGatt(18339): connect() - device: 60:8A:10:50:FE:56, auto: false
D/BluetoothGatt(18339): registerApp()
D/BluetoothGatt(18339): registerApp() - UUID=8a275669-7fc8-40c6-8fa2-c2e61576b94c
D/BluetoothGatt(18339): onClientRegistered() - status=0 clientIf=5
D/[FBP-Android](18339): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(18339): onClientConnectionState() - status=0 clientIf=5 device=60:8A:10:50:FE:56
D/[FBP-Android](18339): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](18339): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(18339): discoverServices() - device: 60:8A:10:50:FE:56
D/[FBP-Android](18339): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(18339): onSearchComplete() = Device=60:8A:10:50:FE:56 Status=0
D/[FBP-Android](18339): [FBP] onServicesDiscovered: count: 5 status: 0
D/[FBP-Android](18339): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(18339): isLeEnabled(): ON
I/flutter (18339): Service discovered - connecting to characteristic
I/flutter (18339): 00001800-0000-1000-8000-00805f9b34fb
I/flutter (18339): [BluetoothCharacteristic{remoteId: 60:8A:10:50:FE:56, serviceUuid: 00001800-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, characteristicUuid: 00002a00-0000-1000-8000-00805f9b34fb, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}]
I/flutter (18339): 00001800-0000-1000-8000-00805f9b34fb
I/flutter (18339): data check: 
I/flutter (18339): 00002a00-0000-1000-8000-00805f9b34fb
I/flutter (18339): 00002a00-0000-1000-8000-00805f9b34fb
I/flutter (18339): Processing characteristic started
I/flutter (18339): Checking characteristic
I/mali_gralloc(18339): unregister: id=0x26b00002b4a, base=0x0, importpid=18339
I/mali_gralloc(18339): unregister: id=0x26b00002b48, base=0x0, importpid=18339
I/mali_gralloc(18339): unregister: id=0x26b00002b47, base=0x0, importpid=18339
I/mali_gralloc(18339): unregister: id=0x26b00002b49, base=0x0, importpid=18339
```

chipweinberger, 2023-10-12:
```
I/flutter (18339): 00001800-0000-1000-8000-00805f9b34fb
I/flutter (18339): 00001800-0000-1000-8000-00805f9b34fb
I/flutter (18339): 00002a00-0000-1000-8000-00805f9b34fb
I/flutter (18339): 00002a00-0000-1000-8000-00805f9b34fb
```

these are the standard ble characteristics. "Generic Access". "Device Name".

widzinskaj, 2023-10-13:
I still think there is a bug, because before update the characteristics were consisted with the hardware ones, and my flutter application were able to connect via ble with the device and exchange data. I want to mention, that the hardware characteristic didn't change and I know its uuids, because it's the device which was build by my colleague and its characteristic are well known.

widzinskaj, 2023-10-17:
[image]
[image]

The attached photos confirm, that the BLE characteristics of device are the same as I mentioned and expected:
```
final Guid newbmsuuid =
Guid('49535343-fe7d-4ae5-8fa9-9fafd205e455'); 
final Guid newbmsrx = Guid('49535343-1e4d-4bd9-ba61-23c647249616');
final Guid newbmstx = Guid('49535343-8841-43f4-a8d4-ecbe34729bb3');
``` 
I assume that the logic of discovering characteristic changed. Can you tell me what steps I should do to discover my services and connect to characteristic? Or what version of flutter_blue_plus I should downgrade to (before the changes related to discovering services)?

chipweinberger, 2023-10-17:
> Can you tell me what steps I should do to discover my services and connect to characteristic? 

you'll need to submit a PR to FBP to fix the issue, if you can determine the problem.

> Or what version of flutter_blue_plus I should downgrade to (before the changes related to discovering services)?

you can determine this by changing your FBP version a few times until it works. use a binary search method.

chipweinberger, 2023-10-30:
without more info, this cannot be fixed. closing.

closed 2023-10-30 by chipweinberger