chipweinberger/flutter_blue_plus #625: [Help]: Cannot stop connect action artificially

Yorkzero, 2023-10-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.15.8

### Flutter Version

3.13.1

### What OS?

Android

### OS Version

android7.1.1

### Bluetooth Module

unknown

### What is your problem?

Hi, I wanna stop connection if timeout, so I use the following code:

```dart
_device?.connect(timeout: const Duration(seconds: 5)).onError((error, stackTrace) {
      _device?.disconnect();
      FlutterBluePlus.stopScan();
    });
```

According to Logcat, the code get into `onError` successfully after 5 seconds, but my phone truly stop connection after 30 seconds! That confused me.

### Logs

```shell
2023-10-14 15:05:07.390 20759-20759 [FBP-Android]           com.makelocks.ble_padlock_app        D  [FBP-Android] onMethodCall: connect
2023-10-14 15:05:07.390 20759-20759 CheckPermission         com.makelocks.ble_padlock_app        E  _bluetooth code = 10 
2023-10-14 15:05:07.391 20759-20759 BluetoothGatt           com.makelocks.ble_padlock_app        D  connect() - device: BC:02:6E:C2:72:C2, auto: false
2023-10-14 15:05:07.391 20759-20759 BluetoothGatt           com.makelocks.ble_padlock_app        D  registerApp()
2023-10-14 15:05:07.391 20759-20759 BluetoothGatt           com.makelocks.ble_padlock_app        D  registerApp() - UUID=5168ddb8-2118-499d-a356-0ed30f3c3359
2023-10-14 15:05:07.394 20759-20771 BluetoothGatt           com.makelocks.ble_padlock_app        D  onClientRegistered() - status=0 clientIf=5
2023-10-14 15:05:12.518 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-10-14 15:05:12.518 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  │ #0   new PadlockEvent.failure (package:ble_padlock_app/business/padlock_ble/padlock_manager.dart:356:12)
2023-10-14 15:05:12.518 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  │ #1   PadlockManager._handleConnEvent (package:ble_padlock_app/business/padlock_ble/padlock_manager.dart:152:54)
2023-10-14 15:05:12.519 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
2023-10-14 15:05:12.519 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  │ 15:05:12.494 (+0:38:13.513922)
2023-10-14 15:05:12.519 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
2023-10-14 15:05:12.519 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  │ padlockEvent>>> event: PadlockCmd.conn, isOpen: false, ifSuccess: false, isFullBat: true, isLockError: false, isBatError: false, mac: null
2023-10-14 15:05:12.520 20759-20859 flutter                 com.makelocks.ble_padlock_app        I  └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
2023-10-14 15:05:12.550 20759-20759 [FBP-Android]           com.makelocks.ble_padlock_app        D  [FBP-Android] onMethodCall: stopScan
2023-10-14 15:05:12.553 20759-20759 BluetoothAdapter        com.makelocks.ble_padlock_app        D  STATE_ON
2023-10-14 15:05:12.553 20759-20759 BluetoothLeScanner      com.makelocks.ble_padlock_app        D  could not find callback wrapper
2023-10-14 15:05:12.655 20759-20759 [FBP-Android]           com.makelocks.ble_padlock_app        D  [FBP-Android] onMethodCall: disconnect
2023-10-14 15:05:12.655 20759-20759 [FBP-Android]           com.makelocks.ble_padlock_app        D  [FBP-Android] already disconnected
2023-10-14 15:05:37.401 20759-20896 BluetoothGatt           com.makelocks.ble_padlock_app        D  onClientConnectionState() - status=133 clientIf=5 device=BC:02:6E:C2:72:C2
2023-10-14 15:05:37.402 20759-20896 [FBP-Android]           com.makelocks.ble_padlock_app        D  [FBP-Android] onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected
2023-10-14 15:05:37.402 20759-20896 BluetoothGatt           com.makelocks.ble_padlock_app        D  close()
```

chipweinberger, 2023-10-14:
good observation. you are correct.

I'll fix this.

chipweinberger, 2023-10-16:
to support using `disconnect` to arbitrarily terminate a connection attempt early would be a bigger and riskier code change.

chipweinberger, 2023-10-16:
this code change is actually very annoying to make:
1. in iOS & android platform code, we need to store in progress *connecting* devices, not just connected once
2. on the dart side we need to make the `connectOrDisconnect` mutex two separate mutexes i.e. `connect` & `disconnect`
3. in the dart connect code we need to allow completing the connection future early when disconnect is called

I wrote this little extension to help with the 3rd point. 

```
extension FutureEarlyCompleteable<T> on Future<T> {
  Completer<T> withEarlyCompleter() {
    final completer = Completer<T>();
    this.then((value) {
      if (!completer.isCompleted) {
        completer.complete(value);
      }
    }).catchError((error) {
      if (!completer.isCompleted) {
        completer.completeError(error);
      }
    });
    return completer;
  }
}
```

Overall it is too big of a code change to do right now. Feel free to try and open a PR.

chipweinberger, 2023-11-06:
fixed in 1.26.1. 

timeouts will now work as expected, calling `_device.disconnect()` automatically to cancel the connection attempt.

closed 2023-11-06 by chipweinberger