chipweinberger/flutter_blue_plus #629: [Feature]: Add the option of not clear scan results after the stopScan

Jbz797, 2023-10-17:
### FlutterBluePlus Version

last

### Flutter Version

2.10.5

### What OS?

Android, iOS

### OS Version

...

### Bluetooth Module

...

### What is your feature request?

Add the option of not clear scan results after the stopScan, like this : 

```
Future<void> startScan({
  List<Guid> withServices = const [],
  Duration timeout,
  Duration removeIfGone,
  bool oneByOne = false,
  bool androidUsesFineLocation = false,
  bool clearIfStop = true,
})
```

### Logs

```shell
...
```

chipweinberger, 2023-10-17:
this option already exists.

you can store the results in your own permanent variable.

`var results;

FlutterBluePlus.scanResults.listen((data){
    results = data;
});`

is there a reason this is insufficient?

Jbz797, 2023-10-18:
Yes but I use `StreamBuilder` : 

```
StreamBuilder<List<ScanResult>>(
                    initialData: const [],
                    stream: FlutterBluePlus.scanResults,
                    builder: (context, snapshot) {
```

chipweinberger, 2023-10-18:
in general streamBuilder will still work

stopScan does not cause a new value to be emitted

it only clears the results so that next time a *new* stream builder widget would see empty results.

this is generally what you want.

what problem do you see in your app?

chipweinberger, 2023-10-18:
that said, i recommend using a stateful widget and `listen`

Jbz797, 2023-10-18:
With this code : 

```

FlutterBluePlus.startScan(
      withServices: [
        Guid('xxx')
      ],
      timeout: : const Duration(seconds: 5),
);

{ ... }

StreamBuilder<List<ScanResult>>(
                    initialData: const [],
                    stream: FlutterBluePlus.scanResults,
                    builder: (context, snapshot) {
                      print('snapshot.data.length : ${snapshot.data.length}');
```

Logs : 

```
I/flutter (15666): snapshot.data.length : 2
I/chatty  (15666): uid=10107(u0_a107) 1.ui identical 1 line
I/flutter (15666): snapshot.data.length : 2
D/[FBP-Android](15666): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(15666): getBluetoothLeScanner
D/BluetoothAdapter(15666): isLeEnabled(): ON
I/flutter (15666): snapshot.data.length : 2
I/flutter (15666): snapshot.data.length : 0
```

chipweinberger, 2023-10-18:
is your stream builder calling listen multiple times?

try modifying FlutterBluePlus _StreamController to print when scanResults.listen is called.

chipweinberger, 2023-10-19:
also, you should create a PR of the "fix" you want so that you can test it actually resolves your issue.

Jbz797, 2023-10-19:
"is your stream builder calling listen multiple times?" => no

This has not worked since update 1.17.6.
That's why I stayed with 1.17.5.

chipweinberger, 2023-10-19:
what happens on older versions?

chipweinberger, 2023-10-19:
okay. that doesn't make much sense to me.

I did spot 1 small issue. clearing the results should happen last in the method. 

```
  static Future<void> _stopScan({bool invokePlatform = true}) async {
    _scanResultsList.latestValue = []; // issue: this should be done last in this method
    _scanSubscription?.cancel();
    _scanTimeout?.cancel();
    _isScanning.add(false);
    if (invokePlatform) {
      await _invokeMethod('stopScan');
    }
  }
```

Also your logs do not make sense to me. Why are you still getting results after you call stopScan?

```
D/[FBP-Android](15666): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(15666): getBluetoothLeScanner
D/BluetoothAdapter(15666): isLeEnabled(): ON
I/flutter (15666): snapshot.data.length : 2
I/flutter (15666): snapshot.data.length : 0
```

Jbz797, 2023-10-19:
OK, you were right, I kept getting the results after the stopScan call because of a steam that encompassed the whole thing and was located higher up:

```
StreamBuilder<bool>(
     stream: FlutterBluePlus.isScanning,
     builder: (context, snapshot) {},
),
```

I don't know why the behavior changed in 1.17.6.

I just moved my FlutterBluePlus.isScanning stream somewhere else and it fixed the problem.

Thanks for your help and sorry for the inconvenience.

chipweinberger, 2023-10-19:
> OK, you were right, I kept getting the results after the stopScan call because of a steam that encompassed the whole thing and was located higher up:

Hmm. So it was causing the scanResults StreamBuilder to be re-built? causing another listen?

chipweinberger, 2023-10-19:
btw another option is to do this

```
StreamBuilder<List<ScanResult>>(
     initialData: const [],
     stream: FlutterBluePlus.scanResults.where((v) => v.isNotEmpty),
     builder: (context, snapshot) {
          print('snapshot.data.length : ${snapshot.data.length}');
```

closed 2023-10-19 by chipweinberger

chipweinberger, 2023-11-17:
I've added new controls for this in 1.29.0

`scanResults` will not behave like you want, without clearing.

and `onScanResults` will behave with clearing / it will not re-emit values on listen.

Jbz797, 2023-11-17:
Perfect, thanks