chipweinberger/flutter_blue_plus #630: [Help]: Android specific error 133

kbessemer, 2023-10-18:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.17.6

### Flutter Version

3.13.7

### What OS?

Android

### OS Version

12

### Bluetooth Module

STM32

### What is your problem?

Status 133 Android Specific Error

previously happening when disconnecting/reconnecting to STM32 regularly, now it is happening on first connect even

### Logs

```shell
D/[FBP-Android](  954): [FBP] onMethodCall: connect
I/BluetoothAdapter(  954): STATE_ON
D/BluetoothGatt(  954): connect() - device: C0:80:E1:26:D2:BB, auto: false
I/BluetoothAdapter(  954): isSecureModeEnabled
D/BluetoothGatt(  954): registerApp()
D/BluetoothGatt(  954): registerApp() - UUID=ad68a29c-066c-4c26-97a5-c8c1d026c3e6
D/BluetoothGatt(  954): onClientRegistered() - status=0 clientIf=8
D/[FBP-Android](  954): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(  954): onClientConnectionState() - status=133 clientIf=8 device=C0:80:E1:26:D2:BB
D/[FBP-Android](  954): [FBP] onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected
D/BluetoothGatt(  954): close()
D/BluetoothGatt(  954): unregisterApp() - mClientIf=8
I/flutter (  954): device connect
I/flutter (  954): listening to device, AddDevice
I/flutter (  954): BluetoothConnectionState.disconnected
D/[FBP-Android](  954): [FBP] onMethodCall: disconnect
D/[FBP-Android](  954): [FBP] already disconnected
```

kbessemer, 2023-10-18:
I want to add that I have my own fork of FBP... you had me change mutexOfKey to String key = "bluetoothMutex";

And I added a 100ms timeout in the take() function

chipweinberger, 2023-10-18:
there's really nothing more I can do to help.

this is a problem with android

closed 2023-10-18 by chipweinberger

sqcsabbey, 2023-11-10:
@chipweinberger Does this [help](https://stackoverflow.com/a/51095280/2199492)?

chipweinberger, 2023-11-10:
we do that already