chipweinberger/flutter_blue_plus #633: [Help]: Devices manually connected by the Android operating system, will they never be shown?

fabiototh, 2023-10-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

  flutter_blue_plus: ^1.17.5

### Flutter Version

Flutter 3.13.7

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Nordic nRF5

### What is your problem?

On Android 13, I connected to device using the Android  Operating System. 
but, i can't found this device, using: FlutterBluePlus.connectedSystemDevices, FlutterBluePlus.bondedDevices, or listem scanResult. Why?

I have two devices. Once appears in the list, however, the others that was manually connected does not.

### Logs

```shell
D/[FBP-Android](27344): [FBP] onMethodCall: startScan
4
I/BluetoothAdapter(27344): STATE_ON
D/BluetoothLeScanner(27344): Start Scan with callback
D/BluetoothLeScanner(27344): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/[FBP-Android](27344): [FBP] onMethodCall: getConnectedSystemDevices
I/BluetoothAdapter(27344): BluetoothAdapter() : ----
D/[FBP-Android](27344): [FBP] onMethodCall: stopScan
2
I/BluetoothAdapter(27344): STATE_ON
D/BluetoothLeScanner(27344): Stop Scan with callback
```

chipweinberger, 2023-10-19:
is it ble or bluetooth classic?

what is the device?

bluetooth  classic is not supported

fabiototh, 2023-10-19:
Device is Ble.

I have two devices of the same model. 
If I go to the bluetooth list of the Android operating system and the BLE device that is connected and disconnect. Appears in the device listing.

what is the device?
I only know that Nordic nRF.

chipweinberger, 2023-10-19:
> the others that was manually connected does not.

after you connect, your device probably stops advertising so it will not show up in scan results.

> i can't found this device, using: FlutterBluePlus.connectedSystemDevices

I do not know the reason for this. that is strange!

fabiototh, 2023-10-19:
Perfectly. If you need some log or tests, please, tell me. If not, I thing we can close this issue. 
@chipweinberger Thanks for you patience.

chipweinberger, 2023-10-20:
did you resolve it?

fabiototh, 2023-10-20:
> did you resolve it?

No, but i think that device probably stops advertising.

chipweinberger, 2023-10-20:
if you connect in settings, it should be in `FlutterBluePlus.systemDevices`

fabiototh, 2023-10-20:
> if you connect in settings, it should be in `FlutterBluePlus.systemDevices`

Did you mean FlutterBluePlus.connectedSystemDevices?
If so, the device should appear in the list of connected, however it does not.

chipweinberger, 2023-10-20:
hmm, i wonder why!

fabiototh, 2023-10-31:
No, but thanks to reply.

closed 2023-10-31 by fabiototh

samnbd, 2024-05-15:
This issue is only happening to me. FYI, the flutter_reactive_ble plugin was able to retrieve the same device even when it was paired.

chipweinberger, 2024-05-15:
you should compare the android code and see how they do it vs fbp