chipweinberger/flutter_blue_plus #634: [Help]: How to get the bluetooth sensor data stream

zubair999, 2023-10-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.3

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

9

### Bluetooth Module

C2541

### What is your problem?

I have bluetooth module which is sending some data string continously, I have install Serial Bluetooth Termial
https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en&gl=US
this app from play store and I am getting the data when device is connected, 

So how to stream data from the bluetooth in flutter_blue_plus.

I already running example app but not getting anywhere, device is connecting but how to get the data stream

### Logs

```shell
No log
```

chipweinberger, 2023-10-20:
Not supported. That app uses Bluetooth Classic!

chipweinberger, 2023-10-20:
1. scan
2. connect
3. discover services
4. loop through services & characteristics.
5. for every charachteristic, listen to characteristic.onValueReceived.listen((data) {print(${utf8.decode(data))});

closed 2023-10-20 by chipweinberger