chipweinberger/flutter_blue_plus #639: [Bug]: [VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: writeCharacteristic: (code: 14) Unlikely error. #0      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:228:9) <asynchronous suspension>

hau1610, 2023-10-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.4

### Flutter Version

3.10.4

### What OS?

iOS, macOS

### OS Version

xcode 15, mac os 14.0 (23A344), ios 17.0.3

### Bluetooth Module

arduino

### What is your problem?

 Unhandled Exception: FlutterBluePlusException: writeCharacteristic

### Logs

```shell
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: writeCharacteristic: (code: 14) Unlikely error.
#0      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:228:9)
<asynchronous suspension>
```

chipweinberger, 2023-10-23:
https://stackoverflow.com/questions/30267227/ios-corebluetooth-unlikely-error

https://developer.apple.com/forums/thread/78320

https://devzone.nordicsemi.com/f/nordic-q-a/1264/ios-corebluetooth-unlikely-error

https://github.com/nettlep/gobbledegook/issues/9

try googling. it is an Apple error.

chipweinberger, 2023-10-23:
imo probably due to a bug in your arduino bluetooth library.

kiranruth, 2023-10-27:
I'm getting the same error

`FlutterBluePlusException (FlutterBluePlusException: writeCharacteristic: (code: 14) UNKNOWN_GATT_ERROR (14))`

on an android device.

chipweinberger, 2023-10-27:
@kiranruth thats a different error. the code 14 is android code.

chipweinberger, 2023-11-03:
try 1.21.0

if you still have problems, reopen the issue.

closed 2023-11-03 by chipweinberger

chipweinberger, 2023-12-19:
This error code originates **from your BLE device**.

It is one of the BLE GATT codes. It means your device is malfunctioning, or you are sending an invalid request.

```
#define BLE_ATT_ERR_INVALID_HANDLE          0x01
#define BLE_ATT_ERR_READ_NOT_PERMITTED      0x02
#define BLE_ATT_ERR_WRITE_NOT_PERMITTED     0x03
#define BLE_ATT_ERR_INVALID_PDU             0x04
#define BLE_ATT_ERR_INSUFFICIENT_AUTHEN     0x05
#define BLE_ATT_ERR_REQ_NOT_SUPPORTED       0x06
#define BLE_ATT_ERR_INVALID_OFFSET          0x07
#define BLE_ATT_ERR_INSUFFICIENT_AUTHOR     0x08
#define BLE_ATT_ERR_PREPARE_QUEUE_FULL      0x09
#define BLE_ATT_ERR_ATTR_NOT_FOUND          0x0a
#define BLE_ATT_ERR_ATTR_NOT_LONG           0x0b
#define BLE_ATT_ERR_INSUFFICIENT_KEY_SZ     0x0c
#define BLE_ATT_ERR_INVALID_ATTR_VALUE_LEN  0x0d
#define BLE_ATT_ERR_UNLIKELY                0x0e
#define BLE_ATT_ERR_INSUFFICIENT_ENC        0x0f
#define BLE_ATT_ERR_UNSUPPORTED_GROUP       0x10
#define BLE_ATT_ERR_INSUFFICIENT_RES        0x11
```