chipweinberger/flutter_blue_plus #640: FlutterBluePlusException: setNotifyValue: (code: 10) The attribute could not be found.

Zqiang12345, 2023-10-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: ^1.20.4

### Flutter Version

Flutter 3.3.8

### What OS?

iOS

### OS Version

ios 16.6.1

### Bluetooth Module

ble

### What is your problem?

In Ios setNotifyValue(true). Unhandled Exception: FlutterBluePlusException: setNotifyValue: (code: 10) The attribute could not be found.    in Android is success

### Logs

```shell
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: setNotifyValue: (code: 10) The attribute could not be found.
```

chipweinberger, 2023-10-24:
did you discover services?

Zqiang12345, 2023-10-25:
Thank you, the problem is solved. After adding CCCD to the device, IOS can normally receive Bluetooth data notification, and Android is normal without CCCD

chipweinberger, 2023-10-25:
great. 

I've also added a check for CCCD in the iOS code, and will log a warning if it is missing.

https://github.com/boskokg/flutter_blue_plus/commit/3dbcaeec4067b981238bcefef6d9920bb9d12f19

closed 2023-10-25 by chipweinberger

nguyenhuuta, 2023-10-25:
@chipweinberger @Zqiang12345 i had the same problem, What is CCCD? 
Thanks you

Zqiang12345, 2023-10-26:
CCCD 全称是 Client Characteristic Configuration Descriptor。蓝牙设备端需要加CCCD descriptor ，不是手机端

Zqiang12345, 2023-10-26:
CCCD stands for Client Characteristic Configuration Descriptor. The Bluetooth device needs to add CCCD descriptor, not the mobile phone