chipweinberger/flutter_blue_plus #642: [Help]: How to get device Mac address on iOS

SAOKIM99, 2023-10-25:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.7 

### Flutter Version

3.13.6

### What OS?

iOS

### OS Version

iOS 16.7

### Bluetooth Module

esp32 PICO D4

### What is your problem?

I'm building a system that connects devices together via BLE. The phone will get the MAC address of device A and send it to device B. Device B uses that MAC address to scan and connect to device A.

I built it for Android and it works fine.

However, I was building for an iOS device when I noticed that: I can't get the device's MAC address (remodeId returns a strange UUID and it's not a MAC address).

How to get the device's MAC address??

Is there a parameter defined specifically for each device so that when Scanning, that parameter will be used to distinguish devices from each other (all devices are advertised with the same UUID)?
[image]

### Logs

```shell
image above
```

chipweinberger, 2023-10-25:
ios does not allow this.

mac address is hidden for privacy reasons.

closed 2023-10-25 by chipweinberger

EdwardRodr1guez, 2023-11-01:
> ios does not allow this.
> 
> mac address is hidden for privacy reasons by Apple.

@chipweinberger Greetings sir,
regarding this constraint, it means that if I store (in local device store for example) that strange uuid, I could not use that uuid for later reconnect to it (for example, in an app which list "favorite" or "recently" connected devices? ) in ios?

chipweinberger, 2023-11-01:
correct.

ios will occasionally change the randomly generated uuid.

EdwardRodr1guez, 2023-11-01:
Thanks sir, interesting, so, we could say this issue is related to the operative system (IOS) rather than flutter itself? I mean this "problem" is also expected to occur for example in swift?