chipweinberger/flutter_blue_plus #643: [Bug]: connectedDevices calls connectedSystemDevices

manuel75o, 2023-10-25:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.20.7

### Flutter Version

3.13.5

### What OS?

iOS

### OS Version

IOS 17.0.1

### Bluetooth Module

esp32

### What is your problem?

connectedDevices calls connectedSystem devices in background and shows them:

Provider:
`final connectedDevicesStreamProvider = StreamProvider((ref) async* {
  final connectedDevicesTemp = Stream.periodic(const Duration(seconds: 5))
      .asyncMap((_) => FlutterBluePlus.connectedDevices);
  await for (var devices in connectedDevicesTemp) {
    final connectedDevices = devices.toSet().toList();
    yield connectedDevices;
  }
});`

`final connectedDevices = ref.watch(connectedDevicesStreamProvider);`

### Logs

```shell
flutter: [FBP] <getConnectedSystemDevices> args: null
flutter: [FBP] <getConnectedSystemDevices> result: {devices: [{platform_name: null, remote_id: XXX}, {platform_name: Apple Watch von Manuel, remote_id: XXX}]}

flutter: [FBP] <getConnectedSystemDevices> args: null
flutter: [FBP] <getConnectedSystemDevices> result: {devices: [{platform_name: ESP1, remote_id: 2EC51AE9-89E0-F017-2C3E-97C558E5B4CE}, {platform_name: null, remote_id: XXX}, {platform_name: Apple Watch von Manuel, remote_id: XXX}]}
```

chipweinberger, 2023-10-25:
i don't think you've properly updated

manuel75o, 2023-10-25:
U are right, sorry for the inconvenience, had a dependencie problem.

closed 2023-10-25 by manuel75o