chipweinberger/flutter_blue_plus #644: [Bug]: Error connecting to device

fabiancrx, 2023-10-26:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.7

### Flutter Version

3.13.8

### What OS?

iOS

### OS Version

16.7.1

### Bluetooth Module

 RedPine  RS9113 , Nordic  nRF51822

### What is your problem?

No issue in the sample app, but when I try to connect to any device the below error occurs. This issue does not happen on android, and I'm not sure if it's related to other plugins like sqlite that I use too in that project.

If I'm missing any other logs, or I could provide more information let me know, I don't usually dabble on the native iOS code. 

### Logs

```shell
#### App logs

[VERBOSE-2:FlutterDarwinContextMetalImpeller.mm(37)] Using the Impeller rendering backend.
[FBP-iOS] initializing CBCentralManager
[FBP-iOS] initializing checkForMtuChangesTimer
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] handleMethodCall: getSystemDevices
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: connect
PlatformException(iosException, *** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[0], {stackTrace: 0   CoreFoundation                      0x00000001b32fe5ec CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 972268
1   libobjc.A.dylib                     0x00000001ab63bc00 objc_exception_throw + 60
2   CoreFoundation                      0x00000001b3229d24 CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 101668
3   CoreFoundation                      0x00000001b321929c CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 33436
4   flutter_blue_plus                   0x0000000104fda160 -[FlutterBluePlusPlugin handleMethodCall:result:] + 7196
5   Flutter                             0x000000010824c084 __45-[FlutterMethodChannel setMethodCallHandler:]_block_invoke + 172
6   Flutter                             0x0000000107cb0e08 ___ZN7flutter25PlatformMessageHandlerIos21HandlePlatformMessageENSt21_LIBCPP_ABI_NAMESPACE10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE_block_invoke + 116
7   libdispatch.dylib                   0x000000010557cf50 _dispatch_call_block_and_release + 32
8   libdispatch.dylib                   0x000000010557eb34 _dispatch_client_callout + 20
9   libdispatch.dylib                   0x000000010558efec _dispatch_main_queue_drain + 1084
10  libdispatch.dylib                   0x000000010558eba0 _dispatch_main_queue_callback_4CF + 44
11  CoreFoundation                      0x00000001b32489bc CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 227772
12  CoreFoundation                      0x00000001b32456c8 CEB74A0C-5EA4-3F5A-9DE8-73F0DAD2F4CB + 214728
13  CoreFoundation                      0x00000001b3244e18 CFRunLoopRunSpecific + 608
14  GraphicsServices                    0x00000001f5d095ec GSEventRunModal + 164
15  UIKitCore                           0x00000001b56532fc EC1206D7-B4C9-3257-BB63-08E26A802416 + 2290428
16  UIKitCore                           0x00000001b5652938 UIApplicationMain + 340
17  Runner                              0x00000001041d8918 main + 64
18  dyld                                0x00000001d5a2fd44 0D856E72-9A63-3B5B-879F-8D026115A4F0 + 23876}, n
```
----
#### Flutter doctor

```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.13.8, on macOS 14.0 23A344 darwin-arm64, locale en-US)
[✓] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 15.0.1)
[!] Android Studio (version unknown)
    ✗ Unable to determine Android Studio version.
[✓] VS Code (version 1.83.1)
[✓] Connected device (2 available)
[✓] Network resources

! Doctor found issues in 1 category.
```
```

chipweinberger, 2023-10-30:
Can you determine which specific line in the code is causing the crash?

perhaps you can add more logs to this part of the function

```
else if ([@"connect" isEqualToString:call.method])
        {
            // See BmConnectRequest
            NSDictionary* args = (NSDictionary*)call.arguments;
            NSString  *remoteId       = args[@"remote_id"];
            NSNumber  *autoConnect    = args[@"auto_connect"];

            // already connected?
            CBPeripheral *peripheral = [self getConnectedPeripheral:remoteId];
            if (peripheral != nil) {
                Log(LDEBUG, @"already connected");
                result(@NO); // no work to do
                return;
            }

            // check the devices iOS knowns about
            NSUUID *uuid = [[NSUUID alloc] initWithUUIDString:remoteId];
            NSArray<CBPeripheral *> *peripherals = [_centralManager retrievePeripheralsWithIdentifiers:@[uuid]];
            for (CBPeripheral *p in peripherals)
            {
                if ([[p.identifier UUIDString] isEqualToString:remoteId])
                {
                    peripheral = p;
                    break;
                }
            }
            if (peripheral == nil)
            {
                result([FlutterError errorWithCode:@"connect" message:@"Peripheral not found" details:remoteId]);
                return;
            }

            // we must keep a strong reference to any CBPeripheral before we connect to it.
            // Why? CoreBluetooth does not keep strong references and will warn about API MISUSE and weak ptrs.
            [self.knownPeripherals setObject:peripheral forKey:remoteId];

            // set ourself as delegate
            peripheral.delegate = self;

            // options
            NSMutableDictionary *options = [[NSMutableDictionary alloc] init];
            if (@available(iOS 17, *)) {
                // note: use CBConnectPeripheralOptionEnableAutoReconnect constant
                // when iOS 17 is more widely available
                [options setObject:autoConnect forKey:@"kCBConnectOptionEnableAutoReconnect"];
            } 

            [_centralManager connectPeripheral:peripheral options:options];
            
            result(@YES);
```

fabiancrx, 2023-11-01:
Added some logging and the line that throws is:
```
            NSArray<CBPeripheral *> *peripherals = [_centralManager retrievePeripheralsWithIdentifiers:@[uuid]];
```
I'll check if I'm not passing a wrong uuid, although it works on android.

chipweinberger, 2023-11-01:
is it possible `uuid` is null?

maybe log the value of `uuid`?

fabiancrx, 2023-11-01:
Yep, is coming as null.
##### LOG
```
[FBP-iOS] [CUSTOM_LOG] uuid (null)
```
##### Code
```objc
NSUUID *uuid = [[NSUUID alloc] initWithUUIDString:remoteId];
Log(LDEBUG, @"[CUSTOM_LOG] uuid %@",uuid);
NSArray<CBPeripheral *> *peripherals = [_centralManager retrievePeripheralsWithIdentifiers:@[uuid]];
```

chipweinberger, 2023-11-01:
what is the value of remoteId?

chipweinberger, 2023-11-01:
added an improvement for this issue in 1.20.8

fabiancrx, 2023-11-01:
> what is the value of `remoteId`?

`[FBP-iOS] [CUSTOM_LOG] remoteId AC:31:81:EA:-A:CB:A-:D3:2C:-B:9B:9-:23:F9:63:42:6F:D5`

that is with 1.20.7, I'll test 1.20.8 now.

chipweinberger, 2023-11-01:
on iOS, you cannot use mac address, 

you must use the uuid you get from scanning on ios.

closed 2023-11-01 by chipweinberger