chipweinberger/flutter_blue_plus #647: [Bug]: OnDone method in scanResults is not called ;

thapliyal-lifespark, 2023-10-28:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.7

### Flutter Version

3.13.5

### What OS?

Android

### OS Version

12

### Bluetooth Module

Esp32

### What is your problem?

The on done method in the scanResults stream is not called ;

`Future<void> startDiscovery(Function onConnect) async {
    try {
      // ignore: unused_local_variable
      StreamSubscription<List<ScanResult>>? scanSubscription;
      await askForPermission();
      _scannedDevices.clear();

      isConnecting =
          false; ////Reset the connecting flag so that when the scan results come up they can be connected to if needed

      isScanning = true;
      notifyListeners();
      BluetoothDevice scannedDevice;

      scanSubscription = FlutterBluePlus.scanResults.listen(
        (event) async {
          log("Scan Result is $event ");
          if (event.isNotEmpty) {
            scannedDevice = event.elementAt(0).device;
            FlutterBluePlus.stopScan();
            scanSubscription!.cancel();
            await connectToDevice(scannedDevice, onConnect);
          }
        },
        onDone: () {
          log("Scanning Done");
          isScanning = false;
          scanSubscription!.cancel();
          notifyListeners();
        },
      );

      await FlutterBluePlus.startScan(
        timeout: const Duration(seconds: 10),
        withServices: [Guid("0000acf0-0000-1000-8000-00805f9b34fb")],
      );
    } catch (e) {
      log("Error in startDiscovery $e");
    }
  }`

### Logs

```shell
Restarted application in 3,093ms.
[log] Running Main.dart
I/ViewRootImpl@a01c353[MainActivity]( 7682): Relayout returned: old=(0,0,1440,2960) new=(0,0,1440,2960) req=(1440,2960)0 dur=34 res=0x1 s={true 514749337600} ch=false
I/ViewRootImpl@a01c353[MainActivity]( 7682): MSG_RESIZED: frame=(0,0,1440,2960) ci=(0,84,0,168) vi=(0,84,0,0) or=1
[log] userToken is
D/[FBP-Android]( 7682): [FBP] onMethodCall: flutterHotRestart
2
D/BluetoothAdapter( 7682): STATE_ON
D/BluetoothLeScanner( 7682): could not find callback wrapper
D/[FBP-Android]( 7682): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 7682): [FBP] connectedPeripherals: 0
I/flutter ( 7682): [FBP] <setLogLevel> args: 5
D/[FBP-Android]( 7682): [FBP] onMethodCall: setLogLevel
I/flutter ( 7682): [FBP] <setLogLevel> result: true
I/ViewRootImpl@a01c353[MainActivity]( 7682): Relayout returned: old=(0,0,1440,2960) new=(0,0,1440,2960) req=(1440,2960)0 dur=18 res=0x1 s={true 514749337600} ch=false
I/ViewRootImpl@a01c353[MainActivity]( 7682): Relayout returned: old=(0,0,1440,2960) new=(0,0,1440,2960) req=(1440,2960)0 dur=23 res=0x1 s={true 514749337600} ch=false
I/ViewRootImpl@a01c353[MainActivity]( 7682): MSG_RESIZED: frame=(0,0,1440,2960) ci=(0,84,0,168) vi=(0,84,0,168) or=1
I/ViewRootImpl@a01c353[MainActivity]( 7682): ViewPostIme pointer 0
I/ViewRootImpl@a01c353[MainActivity]( 7682): ViewPostIme pointer 1
I/flutter ( 7682): [FBP] <getSystemDevices> args: null
D/[FBP-Android]( 7682): [FBP] onMethodCall: getSystemDevices
I/flutter ( 7682): [FBP] <getSystemDevices> result: {devices: []}
[log] connected devices []
[log] to ScanNConnect Page
I/ViewRootImpl@a01c353[MainActivity]( 7682): ViewPostIme pointer 0
I/ViewRootImpl@a01c353[MainActivity]( 7682): ViewPostIme pointer 1
[log] Checking the Adapter state
I/flutter ( 7682): [FBP] <getAdapterState> args: null
D/[FBP-Android]( 7682): [FBP] onMethodCall: getAdapterState
I/flutter ( 7682): [FBP] <getAdapterState> result: {adapter_state: 4}
[log] Scan Result is []
I/flutter ( 7682): [FBP] <startScan> args: {service_uuids: [0000acf0-0000-1000-8000-00805f9b34fb], mac_addresses: [], allow_duplicates: true, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android]( 7682): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 7682): STATE_ON
I/chatty  ( 7682): uid=11096(com.lifesparktech.walk) identical 1 line
D/BluetoothAdapter( 7682): STATE_ON
D/BluetoothLeScanner( 7682): Start Scan with callback
D/BluetoothLeScanner( 7682): onScannerRegistered() - status=0 scannerId=15 mScannerId=0
I/flutter ( 7682): [FBP] <startScan> result: true
I/flutter ( 7682): [FBP] <stopScan> args: null
D/[FBP-Android]( 7682): [FBP] onMethodCall: stopScan
2
D/BluetoothAdapter( 7682): STATE_ON
D/BluetoothLeScanner( 7682): Stop Scan with callback
I/flutter ( 7682): [FBP] <stopScan> result: true
```

chipweinberger, 2023-10-30:
correct

this is by design 

use isScanning if you want to be informed when it is done

chipweinberger, 2023-10-30:
or use your own timeout with stopScan

closed 2023-10-30 by chipweinberger