chipweinberger/flutter_blue_plus #650: [Help]: How to check whether a device has BLE or Bluetooth Classic

dante381, 2023-11-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.7

### Flutter Version

3.1.1

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

Arduino HC-08

### What is your problem?

Need help differentiating a BLE Device and a Bluetooth Classic device.  Is there an alternative like the type property in the previous version of Flutterblueplus.

### Logs

```shell
I/flutter (16373): []
D/[FBP-Android](16373): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](16373): [FBP] initializing BluetoothAdapter
I/BluetoothAdapter(16373): BluetoothAdapter() : com.example.abcd.abcd
I/BluetoothAdapter(16373): STATE_ON
I/BluetoothAdapter(16373): STATE_ON
D/BluetoothLeScanner(16373): could not find callback wrapper
D/[FBP-Android](16373): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](16373): [FBP] connectedPeripherals: 0
D/[FBP-Android](16373): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(16373): STATE_ON
I/BluetoothAdapter(16373): STATE_ON
D/BluetoothLeScanner(16373): could not find callback wrapper
D/[FBP-Android](16373): [FBP] onMethodCall: startScan
I/BluetoothAdapter(16373): STATE_ON
I/BluetoothAdapter(16373): STATE_ON
I/BluetoothAdapter(16373): STATE_ON
I/BluetoothAdapter(16373): STATE_ON
D/BluetoothLeScanner(16373): Start Scan with callback
D/BluetoothLeScanner(16373): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
I/BluetoothAdapter(16373): BluetoothAdapter() : com.example.abcd.abcd
I/flutter (16373): BluetoothDevice{remoteId: 29:89:1B:80:4A:D0, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 44:73:D6:D1:83:3C, platformName: MeetUp Soft Remote, services:
null}
I/flutter (16373): BluetoothDevice{remoteId: 28:C7:84:2F:E3:89, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: CA:23:00:00:00:45, platformName: NTEca2300000045, services: nu
ll}
I/flutter (16373): BluetoothDevice{remoteId: 0C:40:0B:D5:EE:29, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 2F:BD:DC:81:22:35, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: C0:22:00:00:05:94, platformName: NTEc02200000594, services: nu
ll}
I/flutter (16373): BluetoothDevice{remoteId: 26:C6:63:9C:5E:8C, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 52:A3:84:B6:62:B5, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 2B:A7:2C:F2:B2:E8, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 30:7E:59:EA:95:9D, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 02:27:F6:C3:EF:29, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 3F:EB:BF:6D:00:82, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: E4:CC:AB:8E:79:B9, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 04:97:6F:73:EF:D8, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 49:94:D6:77:DA:BC, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: E8:EE:CC:07:AB:71, platformName: , services: null}
I/flutter (16373): BluetoothDevice{remoteId: 03:1F:52:0B:EE:63, platformName: , services: null}
D/[FBP-Android](16373): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(16373): STATE_ON
I/BluetoothAdapter(16373): STATE_ON
D/BluetoothLeScanner(16373): Stop Scan with callback
```

chipweinberger, 2023-11-01:
every device in the scan results is ble

closed 2023-11-01 by chipweinberger

dante381, 2023-11-01:
It is displaying all the devices with bluetooth

chipweinberger, 2023-11-01:
are you sure? we use [getBluetoothLeScanner()](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBluetoothLeScanner())

dante381, 2023-11-02:
In the version 1.15 the property type 'LE' in BluetoothDevice used to show only 3 devices from the above log.
The rest of the devices are bluetooth earbuds.

So is there any alternative for this issue so that I can scan only LE devices.

chipweinberger, 2023-11-02:
If bluetooth classic is being returned, you will need to [file a bug with android AOSP](https://source.android.com/docs/setup/contribute/report-bugs).