chipweinberger/flutter_blue_plus #652: [Help]:

zubair999, 2023-11-02:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.20.7

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 5

### Bluetooth Module

C2541

### What is your problem?

I have received the data from my ble but data is only 20 byte and I am requesting for the increase mtu size after connecting to the ble but still I am getting same data.
I also installed the Serial Bluetooth Terminal(Simple App from Playstore on which I am getting the whole data perfectly without requesting anyting.

These are two library of the Android Native App which I clone android run android app on which I am getting whole data
https://github.com/kai-morich/SimpleBluetoothTerminal
https://github.com/kai-morich/SimpleBluetoothLeTerminal

So it means my device is ok,

Then why requestMtu is not working on Flutter Blue Plus.
Kindly help on, I am stuck for so many days.

### Logs

```shell
I am getting data
```

chipweinberger, 2023-11-02:
use `onValueReceived` and `setNotify`

closed 2023-11-02 by chipweinberger

zubair999, 2023-11-02:
I am already using onValueReceived

`
void printValue() async {
    _services.forEach((service) {
      if (service.uuid.toString() == SERVICE_UUID) {
        service.characteristics.forEach((characteristic) {
          if (characteristic.uuid.toString() == CHARACTERISTIC_UUID) {
            characteristic.setNotifyValue(!characteristic.isNotifying);
            characteristic.onValueReceived.listen((data) {
              print(">>>>>>>>>>>>>>>>");
              if (utf8.decode(data).contains("Orient")) {
                String stream1 = utf8.decode(data);
                print(stream1);
              }
            });
          }
        });
      }
    });
  }
`

This is the function I which I using to print the values in the debug console