chipweinberger/flutter_blue_plus #654: [Help]: Why does the same code get an error the second time it is called?

CallMeBossC, 2023-11-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.19.2

### Flutter Version

3.7.12

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

Not sure

### What is your problem?

When I first call gatt to connect my peripheral to transfer data, there is no problem at all. When I perform the same operation a second time, various issues arise. For example, I will call the method to disconnect the Bluetooth device when the transmission is completed, but when I access the device through the serial port to view the device status, the Bluetooth connection status is still connected and there is no disconnection. Another error is that the service has timed out. After restarting the app, there was no problem with the first call, but 100% found that the service timed out the second time.

### Logs

```shell
BluetoothLeScanner(21438): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
I/BlePlugin(21438): head部分：50046900  data部分：00a903000000000000000000000000000000seq4
D/BluetoothAdapter(21438): isLeEnabled(): ON
D/BluetoothAdapter(21438): isLeEnabled(): ON
I/flutter (21438): hasPairDevice.toString(){}
I/flutter (21438): Find device BluetoothDevice{remoteId: 3C:5D:3A:74:25:36, platformName: lamp, services: null} Start connecting
I/flutter (21438): hasPairDevice.toString(){}
I/chatty  (21438): uid=10959(com.mesh.easysmart) 2.ui identical 1 line
I/flutter (21438): hasPairDevice.toString(){}
D/[FBP-Android](21438): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(21438): isLeEnabled(): ON
D/[FBP-Android](21438): [FBP] onMethodCall: disconnect
D/[FBP-Android](21438): [FBP] already disconnected
I/BlePlugin(21438): head部分：50046900  data部分：00a903000000000000000000000000000000seq4
D/BluetoothAdapter(21438): isLeEnabled(): ON
D/BluetoothAdapter(21438): isLeEnabled(): ON
I/BlePlugin(21438): head部分：50046900  data部分：00a903000000000000000000000000000000seq4
D/BluetoothAdapter(21438): isLeEnabled(): ON
D/BluetoothAdapter(21438): isLeEnabled(): ON
I/BlePlugin(21438): head部分：50046900  data部分：00a903000000000000000000000000000000seq4
D/[FBP-Android](21438): [FBP] onMethodCall: connect
D/BluetoothGatt(21438): connect() - device: 3C:5D:3A:74:25:36, auto: false
D/BluetoothGatt(21438): registerApp()
D/BluetoothGatt(21438): registerApp() - UUID=4adfca33-f02e-4c3c-8319-5741e4fa492d
D/BluetoothGatt(21438): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt(21438): onClientConnectionState() - status=0 clientIf=6 device=3C:5D:3A:74:25:36
D/[FBP-Android](21438): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
I/flutter (21438): connected
D/[FBP-Android](21438): [FBP] onMethodCall: requestMtu
D/BluetoothGatt(21438): configureMTU() - device: 3C:5D:3A:74:25:36 mtu: 110
D/[FBP-Android](21438): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(21438): discoverServices() - device: 3C:5D:3A:74:25:36
D/BluetoothGatt(21438): onConnectionUpdated() - Device=3C:5D:3A:74:25:36 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(21438): onConfigureMTU() - Device=3C:5D:3A:74:25:36 mtu=110 status=0
D/[FBP-Android](21438): [FBP] onMtuChanged: mtu: 110 status: 0
D/BluetoothGatt(21438): onConnectionUpdated() - Device=3C:5D:3A:74:25:36 interval=36 latency=0 timeout=500 status=0
E/flutter (21438): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: discoverServices: (code: 1) Timed out after 15s
E/flutter (21438): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:58:7)
E/flutter (21438): #1      _RootZone.run (dart:async/zone.dart:1654:54)
E/flutter (21438): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:865:34)
E/flutter (21438): #3      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
E/flutter (21438): #4      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
E/flutter (21438): #5      _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
E/flutter (21438): #6      _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:192:26)
E/flutter (21438): 
D/[FBP-Android](21438): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(21438): discoverServices() - device: 3C:5D:3A:74:25:36
D/BluetoothGatt(21438): onSearchComplete() = Device=3C:5D:3A:74:25:36 Status=0
D/[FBP-Android](21438): [FBP] onServicesDiscovered: count: 6 status: 0
D/[FBP-Android](21438): [FBP] onMethodCall: connect
D/[FBP-Android](21438): [FBP] already connected
D/[FBP-Android](21438): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(21438): setCharacteristicNotification() - uuid: 0000ffe3-d8a9-e658-87ef-a96719815d7b enable: true
D/[FBP-Android](21438): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(21438): discoverServices() - device: 3C:5D:3A:74:25:36
D/BluetoothGatt(21438): onSearchComplete() = Device=3C:5D:3A:74:25:36 Status=0
D/[FBP-Android](21438): [FBP] onServicesDiscovered: count: 6 status: 0
D/[FBP-Android](21438): [FBP] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
```

chipweinberger, 2023-11-03:
make sure you `await` every call

CallMeBossC, 2023-11-03:
Yes, I have to wait for the discoverServices method, otherwise it will not be executed smoothly the first time. What surprises me most is that the error is reported not for the first connection but for the second time.

chipweinberger, 2023-11-03:
try 1.21.0

chipweinberger, 2023-11-04:
These logs are fishy.

It seems like you are not awaiting `requestMtu` because you start discovering services before the mtu finishes updating.

```
D/[FBP-Android](21438): [FBP] onMethodCall: requestMtu
D/BluetoothGatt(21438): configureMTU() - device: 3C:5D:3A:74:25:36 mtu: 110
D/[FBP-Android](21438): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(21438): discoverServices() - device: 3C:5D:3A:74:25:36
D/BluetoothGatt(21438): onConnectionUpdated() - Device=3C:5D:3A:74:25:36 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(21438): onConfigureMTU() - Device=3C:5D:3A:74:25:36 mtu=110 status=0
D/[FBP-Android](21438): [FBP] onMtuChanged: mtu: 110 status: 0
D/BluetoothGatt(21438): onConnectionUpdated() - Device=3C:5D:3A:74:25:36 interval=36 latency=0 timeout=500 status=0
```

chipweinberger, 2023-11-04:
try 1.21.0

I've made the awaiting internal to the FBP library to avoid these types of user errors.

CallMeBossC, 2023-11-06:
Thanks, I used version 1.26.0 and there was no problem

chipweinberger, 2023-11-06:
great

closed 2023-11-06 by chipweinberger