chipweinberger/flutter_blue_plus #656: [Bug]: During initState activation of notification for characteristic fails

Romcik077, 2023-11-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.20.7

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

Android 11/13

### Bluetooth Module

bgm210pa22jia2

### What is your problem?

I have 4 services with different uuids which have. Each service have 2 characteristics the uuid for these are same form service to service:
[image]
It was done to create in flutter one widget and create for each service a instance of widget.
The State characteristic I want to use as feedback with notification from BLE device.
The problem is that is I try to call setNotifyValue for State I receive next Exception:
`Exception has occurred.
PlatformException (PlatformException(setNotifyValue, gatt.writeDescriptor() returned 201 : ERROR_GATT_WRITE_REQUEST_BUSY, null, null))`
I tried in may different ways to do it in initState, but every time has same Exception.
If I move setNotifyValue to button somewhere in widget everything works without problem.

### Logs

```shell
Log with Exception:
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(31575): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt(31575): onClientConnectionState() - status=0 clientIf=9 device=28:76:81:44:4E:0D
D/[FBP-Android](31575): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/BluetoothGatt(31575): onClientConnectionState() - status=0 clientIf=9 device=28:76:81:44:4E:0D
D/[FBP-Android](31575): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](31575): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(31575): discoverServices() - device: 28:76:81:44:4E:0D
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(31575): onConnectionUpdated() - Device=28:76:81:44:4E:0D interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(31575): onSearchComplete() = Device=28:76:81:44:4E:0D Status=0
D/[FBP-Android](31575): [FBP] onServicesDiscovered: count: 9 status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(31575): discoverServices() - device: 28:76:81:44:4E:0D
D/BluetoothGatt(31575): onSearchComplete() = Device=28:76:81:44:4E:0D Status=0
D/[FBP-Android](31575): [FBP] onServicesDiscovered: count: 9 status: 0
D/BluetoothGatt(31575): onConnectionUpdated() - Device=28:76:81:44:4E:0D interval=36 latency=0 timeout=500 status=0
D/[FBP-Android](31575): [FBP] onMethodCall: readCharacteristic
2
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](31575): [FBP] onCharacteristicRead: uuid: 00003010-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](31575): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(31575): setCharacteristicNotification() - uuid: 00003011-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android](31575): [FBP] onCharacteristicRead: uuid: 00003010-0000-1000-8000-00805f9b34fb status: 0

Logs with setNotifyValue commented in initState:
D/BluetoothGatt(31575): discoverServices() - device: 28:76:81:44:4E:0D
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(31575): onConnectionUpdated() - Device=28:76:81:44:4E:0D interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(31575): onSearchComplete() = Device=28:76:81:44:4E:0D Status=0
D/[FBP-Android](31575): [FBP] onServicesDiscovered: count: 9 status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: readRssi
D/BluetoothGatt(31575): readRssi() - device: 28:76:81:44:4E:0D
D/[FBP-Android](31575): [FBP] onReadRemoteRssi: rssi: -52 status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt(31575): onConnectionUpdated() - Device=28:76:81:44:4E:0D interval=36 latency=0 timeout=500 status=0
D/[FBP-Android](31575): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](31575): [FBP] onCharacteristicRead: uuid: 00003010-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](31575): [FBP] onCharacteristicRead: uuid: 00003010-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](31575): [FBP] onCharacteristicRead: uuid: 00003010-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](31575): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](31575): [FBP] onCharacteristicRead: uuid: 00003010-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](31575): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(31575): isLeEnabled(): ON
```

Romcik077, 2023-11-03:
Important thing: If I try to process only one service instead of 4 or 2, the setNotifyValue call in initState works without problems.

chipweinberger, 2023-11-03:
you should always use `await` when calling any FBP function.

Romcik077, 2023-11-03:
I add await.
This is my code from initState, in different ways:
```    
   _controlCharacteristic.read().then((value) async {
      for (int i = 0; i < _selectedFunctionality.length; i++) {
        _selectedFunctionality[i] = i == value[0];
      }
      // await _stateCharacteristic.setNotifyValue(_stateCharacteristic.isNotifying == false);
    });
    WidgetsBinding.instance.addPostFrameCallback((timeStamp) async {
      await _stateCharacteristic.setNotifyValue(_stateCharacteristic.isNotifying == false);
    });
    // _stateCharacteristic.read().then((value) async {
    // await _stateCharacteristic.setNotifyValue(_stateCharacteristic.isNotifying == false);
    // setState(() {});
    // });
    if (_stateCharacteristic.properties.notify || _stateCharacteristic.properties.indicate) {
      try {
        // _stateCharacteristic
        //     .setNotifyValue(_stateCharacteristic.isNotifying == false)
        //     .catchError((error) {
        //   print("help $error");
        //   return false;
        // });
        // _testSubscription = _stateCharacteristic.lastValueStream.listen((value) {
        //   print(value);
        //   // for (int i = 0; i < _selectedFunctionality.length; i++) {
        //   //   _selectedFunctionality[i] = i == value[0];
        //   // }
        //   setState(() {});
        // });
      } catch (e) {
        Snackbar.show(ABC.c, prettyException("Subscribe Error:", e), success: false);
      }
    }
```

Every time same result.

Romcik077, 2023-11-03:
> you should _not_ be calling `setNotifyValue` or any other FBP call in `initState`.

OK, but how I can automatically activate notification for characteristic. Should I create a delay?

chipweinberger, 2023-11-03:
try 1.21.0

Romcik077, 2023-11-04:
Hello!
Now it works!
Thanks! Issue can be closed.

closed 2023-11-04 by chipweinberger