chipweinberger/flutter_blue_plus #657: [Bug]: Perpetual scan on engine detach

Ibetz1, 2023-11-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.22.1

### Flutter Version

3.13.6

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

esp32-s1

### What is your problem?

On abruptly closing the app during a scan, it never stops the scan causing this warning "W/[FBP-Android](10659): [FBP] invokeMethodUIThread: tried to call method on closed channel: OnScanResponse".

Upon reopening the app, the thread throwing the warning appears to be stuck open. I fixed the error myself in your "FlutterBluePlusPlugin.java" in the "onDetachedFromEngine" method by adding:

```java
// ON APP CLOSE
@Override
public void onDetachedFromEngine(@NonNull FlutterPluginBinding binding)
{
    log(LogLevel.DEBUG, "onDetachedFromEngine");

    pluginBinding = null;

    disconnectAllDevices("onDetachedFromEngine");

    // perpetual scan fix //
    BluetoothLeScanner scanner = mBluetoothAdapter.getBluetoothLeScanner();
    scanner.stopScan(getScanCallback());
    // end perpetual scan fix //
    
    context.unregisterReceiver(mBluetoothBondStateReceiver);
    context.unregisterReceiver(mBluetoothAdapterStateReceiver);
    context = null;

    methodChannel.setMethodCallHandler(null);
    methodChannel = null;

    mBluetoothAdapter = null;
    mBluetoothManager = null;
}
```

I am not sure if this does this for IOS but it might be something to look into, I also don't know if this is a proper fix or not.

### Logs

```shell
W/[FBP-Android](10659): [FBP] invokeMethodUIThread: tried to call method on closed channel: OnScanResponse (this gets looped infinitely)
```

chipweinberger, 2023-11-04:
seems like a reasonable fix

chipweinberger, 2023-11-04:
fixed 1.24.1

closed 2023-11-04 by chipweinberger

Ibetz1, 2023-11-06:
sweet thanks