chipweinberger/flutter_blue_plus #660: [Bug]: connect without scan option not working after 1.21.0

keskink, 2023-11-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.21.0

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

iOS 16.6.1

### Bluetooth Module

ti 2460

### What is your problem?

connect w/o scan option not working after 1.21.0 version.
1.20.8 working.

### Logs

```shell
n/a
```

chipweinberger, 2023-11-07:
Can you debug this further and give more details?

Logs?

keskink, 2023-11-07:
**flutter: ********** fxWS connecting  / 8F074E79-2598-D869-BC37-6095C35DCAFC**
[FBP-iOS] handleMethodCall: connect
[FBP-iOS] handleMethodCall: disconnect
[FBP-iOS] disconnect: canceling connection in progress
[FBP-iOS] didDisconnectPeripheral
**flutter: -----------Device State---------WinkelStick 8F074E79-2598-D869-BC37-6095C35DCAFC** BluetoothConnectionState.disconnected

bold lines are my remarks.
trying to connect then going to disconnect after wait aprox. 20sec.

sometimes connecting after long wait.

chipweinberger, 2023-11-07:
why is `[FBP-iOS] handleMethodCall: disconnect` being called? are you calling it?

chipweinberger, 2023-11-07:
btw, those logs mean you are using 1.26.1 or later, not 1.21.0.

keskink, 2023-11-07:
log from 1.26.4,
I don't sent timeout parameter, it is default. 
All ok at #1.20.8

chipweinberger, 2023-11-07:
if you figure it out lmk!

I have not hit this issue.

chipweinberger, 2023-11-10:
any progress?

closed 2023-11-11 by chipweinberger

chipweinberger, 2023-11-11:
im going to assume you fixed this. feel free to reopen.