chipweinberger/flutter_blue_plus #661: [Help]: App becomes very slow after calling `startScan` a few times later

wxxedu, 2023-11-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.26.4

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

iOS 17.0

### Bluetooth Module

N/A

### What is your problem?

The app becomes very slow after startScan is called a few times. I have cancelled the subscription to startScan in my code. 

### Logs

```shell
flutter: [FBP] <setLogLevel> args: 5
flutter: [FBP] <setLogLevel> result: true
flutter: [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 4}
flutter: [FBP] <startScan> args: {service_uuids: [], mac_addresses: [], allow_duplicates: true, android_scan_mode: 2, android_uses_fine_location: false}
flutter: [FBP] <startScan> result: true
flutter: 2023-11-07T16:35:34.663890 [INFO][BleDataSourceImpl] Scan started
flutter: 2023-11-07T16:35:34.664310 [INFO][UserBluetoothRepository] Adding subscription: SCANNING
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: {6: 0109202292164520fdcfca147b5444b70f9c61b6851fe316f98849}, service_uuids: null, connectable: 0, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: 5074BAE8-31D0-B64F-3115-8EAB33C3F74E}, rssi: -79}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 0, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: B62FD6B0-4F98-042B-C493-529A37A859A7}, rssi: -77}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: [5B36D774-8A3E-4401-AFAC-B78A2499ABCA], connectable: 1, service_data: null, local_name: Ket9, tx_power_level: 10}, device: {platform_name: Ket9, remote_id: 45D73EBA-DAEA-6402-925D-D2E9190018A7}, rssi: -64}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: {13: 0100000000}, service_uuids: [5B36D774-8A3E-4401-AFAC-B78A2499ABCA], connectable: 1, service_data: null, local_name: Ket9, tx_power_level: 10}, device: {platform_name: Ket9, remote_id: 45D73EBA-DAEA-6402-925D-D2E9190018A7}, rssi: -64}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 1, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: 12567BDD-4E77-8E39-10A3-8CA5F887D57E}, rssi: -46}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 1, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: 12567BDD-4E77-8E39-10A3-8CA5F887D57E}, rssi: -45}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 0, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: B037ABD1-9710-3574-5C52-596061E3A67C}, rssi: -45}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: {6: 010f210a769363f6cf344c4150544f502d444b4f5249373431}, service_uuids: null, connectable: 0, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: D3363F29-3C45-A429-CD9B-1A57A7C41836}, rssi: -65}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: {6: 01092002ad1a7ac9c22470741b48a75efed0b6be9a6fdb01b75715}, service_uuids: null, connectable: 0, service_data: null, local_name: null, tx_power_level: null}, device: {platform_name: null, remote_id: 1D9E1233-9C4E-5CA5-F585-47726150405A}, rssi: -63}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 1, service_data: null, local_name: null, tx_power_level: 7}, device: {platform_name: null, remote_id: F8796A96-7256-FC32-8EE5-D854593ADBAF}, rssi: -86}}
flutter: [FBP] [[ OnScanResponse ]] result: {result: {advertisement_data: {manufacturer_data: null, service_uuids: null, connectable: 1, service_data: null, local_name: null, tx_power_level: 7}, device: {platform_name: null, remote_id: F8796A96-7256-FC32-8EE5-D854593ADBAF}, rssi: -86}}
```

wxxedu, 2023-11-07:
Also it seems to me that after calling the `stopScan`, the `scanResults` is still being updated.

chipweinberger, 2023-11-07:
also, make sure you cancel the scanResults subscription!

so that you don't have multiple listeners

wxxedu, 2023-11-07:
Thanks so much! I now have some ideas as to how to work around this problem in my code. 

The part that's relatively confusing to me is that in the beginning when I call the `startScan`, it works fine. Then after a few minutes, once I tap on the function that calls the `startScan` again, it becomes very laggy. I have double checked that I have indeed cancelled the subscriptions, but I guess I'll need to clear up the logic a bit more to see what actually is going on as well as give the directions that you mentioned a try.

Thanks :)

chipweinberger, 2023-11-07:
in 1.26.5 I've added some more filters & a `continuousUpdates` toggle. You can use these to try and reduce how often your callback it called to increase perf.

* **[Add]** scanning: `withRemoteIds`, `withNames`, `withKeywords` filters
* **[Add]** scanning: `continuousUpdates` option replaces former `allowDuplicates` option

chipweinberger, 2023-11-09:
also in 1.27.2, I added a small perf improvement 

[[Perf] scanning: only add keys to internal advertisment hashmap when they exist ](https://github.com/boskokg/flutter_blue_plus/commit/9ce5cc451ac197039e59bf1a6af856a8bae61731)

chipweinberger, 2023-11-09:
in 1.27.3, I've also added a `continuousDivisor` option with you can use to divide the total number of advertisements processed to 1/2, 1/3, 1/4, etc, as needed. It is per-device, so you still get the 1st advertisement from each device.

jiangchengze, 2023-11-10:
I encountered the same situation. Occasionally the UI stuck when entering the scanning interface. It is easier to reproduce on low-configuration mobile phones.

chipweinberger, 2023-11-10:
afaict, FBP is doing the minimum amount of work posssible. If you find a solution, lmk.

chipweinberger, 2023-11-11:
@jiangchengze , are you setting any scan filters? you should set as many as you can.

joyhope, 2023-11-15:
Is there any explanation for why slow happen?

I also met app slow problem,  the complete app UI is unable to refresh,  but is not related to the scan,  we are still tracing where the problems come from.   (Our slow app happened after 3 days when BLE data was received, the UI screen is very hard to refresh.)

joyhope, 2023-11-16:
We are still testing whether the slow problem happen again.  I though it was fixed due to https://github.com/boskokg/flutter_blue_plus/commit/3328ac5ad9c6dd54baf8668532a46d8e6273c154.

The improper isScanning state may cause the app to make a quick response, and startScan and app level stop/start process may cause conflicting.

joyhope, 2023-11-16:
It is very strange, The App slow problem is happening again after running 3 days.  The difficulty is that there is no exceptional log appeared.  Our regression test shows: that the problem happened since 1.20.x version.  Our last stable version is 1.17.6 (I guess) It is still in testing. 

I read the changed log, but I did not find a related part.  Is there any suggestion?

chipweinberger, 2023-11-17:
no suggestions. I don't know the reason either!

joyhope, 2023-11-17:
1.17.6 seems slowly become slow now.  The problem may be between 1.17.1 to 1.17.6.

joyhope, 2023-11-17:
Our code uses characteristic.lastValueStream for notification.  There is different  implement between 1.15.5 ~ 1.17.1.

The FBP makes a right changings.  Our code should chanage to  characteristic.onValueReceived.

I make test again.

chipweinberger, 2023-11-17:
@joyhope , changing random things is not going to be helpful.

You should take a `profile`. https://medium.com/dartlang/dart-devtools-analyzing-application-performance-with-the-cpu-profiler-3e94a0ec06ae

AxelBlaz3, 2023-11-17:
In my case, it's happening when I start writing to the BLE device. Usually, the write time takes about 10-15 min for me. The first few minutes are blazing fast, then the thermal throttling starts. It's so slow now.

joyhope, 2023-11-19:
My slow problem may be solved(Still in testing). I found the problem.  The subscription is not canceled because of a new stream listen.   With the time on, there are more and more subscriptions in the system. 

The onValueReceived keeps using one stream, but lastValueStream will always use a new stream.  It is my code problem, not related to FBP. 

I hope my test will finally confirm the problem.

joyhope, 2023-11-19:
`      readFastCharacteristic?.cancel();
      readFastCharacteristic = characteristic.onValueReceived.listen(null) <<===new
      //readFastCharacteristic = characteristic.lastValueStream.listen(null); <<==old`

The cancel did not use await, I am not clear whether it caused the slow performance.

chipweinberger, 2023-11-19:
@joyhope every time you call "listen" it creates a new stream subscription and takes more resources to feed that subscription.

"creating more streams" does not matter. it's the subscriptions that matter.

yes you need to cancel subscriptions. the readme has examples.

joyhope, 2023-11-19:
As the code, the cancel is did called, but does not with await.

`      readFastCharacteristic?.cancel();
      readFastCharacteristic = characteristic.onValueReceived.listen(null);`

Using lastValueStream meets the slow problem, but using onValueRecived seems no problem until now.  But I could not explain why this happened.   With await xxx.cancel() is the correct way, but without the await, I could not explain why caused slow.

joyhope, 2023-11-19:
My test failed.  The app BLE notification becomes slower 

The latest no slower verison is 1.15. 5 (mayb 1.15.x) , 1.17.x and 1.28.x all meet the slow problem. I checked my code again and again, but nothing found except the cancel.

joyhope, 2023-11-19:
I said the notification seemed slow, but the data was not lost.  The UI will be updated if the notification happens, but if the slow happens, the UI refresh becomes very slow. In the normal, the UI will be updated at least 10HZ. With time on, the UI update will become 1s and will be even more long.  I could explain why this happened.

chipweinberger, 2023-11-19:
@joyhope do you know how to profile code and determine why it is slow?

joyhope, 2023-11-20:
I know use profile mode. but if the profile is not stable in my test environment. Anyway, I will construct the mode.  The first goal is to make the current app become stable as quickly as possible. 

If 1.15.x is confirmed stable, I will profile with the latest FBP and give feedback.

joyhope, 2023-11-20:
1.15.5 is confirmed no slow problem.    This time the app is back to 1.15.10 version, and tests again.

joyhope, 2023-11-21:
Help confirm the code: 
`Future<T> fbpEnsureDeviceIsConnected(BluetoothDevice device, String function) {
    // Create a completer to represent the result of this extended Future.
    var completer = Completer<T>();

    // disconnection listener.
    var subscription = device.connectionState.listen((event) {
      if (event == BluetoothConnectionState.disconnected) {
        if (!completer.isCompleted) {
          completer.completeError(FlutterBluePlusException(
              ErrorPlatform.fbp, function, FbpErrorCode.deviceIsDisconnected.index, "Device is disconnected"));
        }
      }
    });

    // When the original future completes
    // complete our completer and cancel the subscription.
    this.then((value) {
      if (!completer.isCompleted) {
        subscription.cancel();
        completer.complete(value);
      }
    }).catchError((error) {
      if (!completer.isCompleted) {
        subscription.cancel();
        completer.completeError(error);
      }
    });

    return completer.future;
  }`

The first completer.completeError(FlutterBluePlusException) does not call subscription.cancel() , whether this will cause the problem?

joyhope, 2023-11-22:
1.15.10 library has been testing for two days,  up till now it is working perfectly.  It is very strange that we just changed the FBP version, and the problem seems gone.  

I compared 1.15.x with 1.17.1 again and again,  I could not find any clue related such slow problem.  After the 1.15.10 passes 3 days test,  I will debug the latest FBP library.

joyhope, 2023-11-23:
The app confirmed the 1.15.10 version worked without fault.  I test a little longer and then try to debug the problem.

joyhope, 2023-11-29:
Just found many changes happened.  I will use the latest version to make a profile build.

joyhope, 2023-11-30:
I'd rather find the problem by the code review. 

I suspect the code is different between 1.15.X and 1.17.1

`  Stream<T> get stream {
    return latestValue != null
      ? _controller.stream.newStreamWithInitialValue(latestValue!)
      : _controller.stream;
  }`

In the code, the _StreamController object gets a stream, which always causes a new Stream. The new Stream may not be GC by the engine. 

I have not made a test on the code.

chipweinberger, 2023-11-30:
that's an interesting thought

joyhope, 2023-12-02:
I changed the code and made a test. It needs 3 days to get the result. (It is in the normal usage state.). I will update the result. Hope it works.

joyhope, 2023-12-04:
I confirmed using 1.15.10 was a good version. It has been working over 10 days.

joyhope, 2023-12-04:
1. what happen when it slows?
App real-time draw data from Bluetooth. The data is correctly received, but the app draws data causing a frame drop.  It drawed data frame becomes much longer than the beginning time.  In the beginning, 10hz refresh is no problem, but slow will cause 1s, 2s, even10s refresh.

2. is the app open for 10 days?
The app works for the device.  If start it will be ready for display data.  It could be running for months, without restart. 

3. does it ever recover?
It never recovered. it becomes worse and worse until the app touch screen is unable to work.

chipweinberger, 2023-12-04:
@joyhope I would add logs all over your program with timestamps so you can determine which part of the application is getting slow 

it might not be a FBP issue

my guess is that some data is growing and growing.

joyhope, 2023-12-05:
Maybe,  but it is true that  I backed to 1.15.10,  and then the problem is gone.  The library has been used for a very long time.  I first use flutter_blue, and then, you take over and go on.

The problem suddenly happened in the Oct, so I checked my app first.  The last step is thought the FBP caused.

joyhope, 2023-12-05:
I use 1.16.6 for the smaller problem versions.  I hope to get the exact problem version and then make analysis.

joyhope, 2023-12-06:
The version from 1.16.6 ~1.16.10 does not work for onValueReceived.  The App is unable to test. 1.16.12 said fixed the problem (regression 1.16.3)    This may be a clue for what happened.  The slow is mainly about onValueReceived.

joyhope, 2023-12-09:
The newer adapterState is create new stream, the old lib(1.16.12) return the same stream.  My app is using StreamBuild in the stateless widget. 

The FBP demo is using state widget, and using listen the state. 

Whether the StreamBuild use case with the new implement cause the problem?  Just a guess. Because I checked the code again and again,  nothing was found.

joyhope, 2023-12-09:
I used 1.17.1 version with adapteState 1.16.12 implement, so the adapterState is always the same stream object, otherwise, the new adapterState will not be closed.  

I made the test again. I hope it will be confirmed.

joyhope, 2023-12-09:
I am still not confident about the "problem".   Good is the problem is start begin 1.17.0, or 1.17.1.  Currently, I am just closer.

joyhope, 2023-12-09:
The problem does not happen in my case.  I made a test on 1.17.0， go on finding the first problem version.

joyhope, 2023-12-09:
I still have questions about this code

`  static Stream<BluetoothAdapterState> get adapterState async* {
    // start listening now so we do not miss any changes
    var buffer = _BufferStream.listen(FlutterBluePlus._methodStream.stream
        .where((m) => m.method == "OnAdapterStateChanged")
        .map((m) => m.arguments)
        .map((args) => BmBluetoothAdapterState.fromMap(args))
        .map((s) => _bmToBluetoothAdapterState(s.adapterState)));

    // initial state
    BluetoothAdapterState initialValue = await _invokeMethod('getAdapterState')
        .then((args) => BmBluetoothAdapterState.fromMap(args))
        .then((s) => _bmToBluetoothAdapterState(s.adapterState));

    // make sure the initial value has not become out of date
    // while we were awaiting for the initial state
    if (buffer.hasReceivedValue == false) {
      yield initialValue;
    }

    // stream
    yield* buffer.stream;
  }`

buffer is always get new object, so the buffer.hasReceivedValue == false is true, and then he buffer is nerver close?

chipweinberger, 2023-12-09:
it should close when you cancel the stream subscription i think

good luck

joyhope, 2023-12-09:
The code 
`
if (buffer.hasReceivedValue == false) {
  yield initialValue;
}
`

In 1.17.1 , I found is alway == false.  It means the buffer stream is always new, and never closed.  The stream is unable to be closed in the app.

chipweinberger, 2023-12-09:
hmm yes maybe BufferStream is not being closed.

I think buffer stream is not necessary

we can something like this 

if (_adapterStateNow != initialValue) {
return xxx. newStreamWithInitialValue(_adapterStateNow)
}

chipweinberger, 2023-12-09:
try master branch! thanks for finding this issue! you did a lot of work!

joyhope, 2023-12-10:
I will still take time to understand the problem, and I will test the latest code, and give feedback.  

Thank you for continuously improving the project.

joyhope, 2023-12-13:
I tried the master branch version, and after running 2 days, the app displayed slowly again.  It is a tough problem. I had to back 1.17.0, and check whether this is the latest no-problem version. And try to find the problem between "good" and "bad". 

Otherwise, I had to use profile mode. In fact, I am afraid that even with the profile version, the problem is still had to detect.

chipweinberger, 2023-12-13:
that's very surprising.

perhaps you can commenting out parts of your code to try and limit the causes. 

also, you should try and make the problem happen faster.

chipweinberger, 2023-12-15:
I've release 1.30.5 with the previous changes. It's still an improvement even though it does not fix the problem for you.

ycnc1314, 2023-12-15:
`class _NewStreamWithInitialValueTransformer<T> extends StreamTransformerBase<T, T> {
  final T initialValue;

  _NewStreamWithInitialValueTransformer(this.initialValue);

  @override
  Stream<T> bind(Stream<T> stream) {
/*    if (stream.isBroadcast) {
      return _bind(stream).asBroadcastStream();
    } else {
      return _bind(stream);
    }*/
    return _bind(stream);
  }
...
}`

joyhope, 2023-12-16:
I read this part many many times.  Why this is the problem?

joyhope, 2023-12-16:
I see this: [[Improve] mergeStreams: cancel all subscriptions & close on done](https://github.com/boskokg/flutter_blue_plus/commit/fa7d15f32ee611cfefc4d1770287210f0cd5a91a).   This is also a potential problem.  Thanks for this. 

I will finish the 1.17.0 test on Sunday, and then use the latest version.   As my test, 1.17.0 is the last version that could work for a long time.  

The problem should be between 1.17.0 and 1.17.1.   I read different more than 5 times.  It is a puzzle.

joyhope, 2023-12-18:
Confirmed 1.17.0 is last stable for long time version.  I use 1.30.5 for test.

ycnc1314, 2023-12-18:
> I read this part many many times. Why this is the problem?

‘’’
utils.dart
class _NewStreamWithInitialValueTransformer<T> extends StreamTransformerBase<T, T> {
  final T initialValue;

  _NewStreamWithInitialValueTransformer(this.initialValue);

  @override
  Stream<T> bind(Stream<T> stream) {
    //------------------------------------change here
/*    if (stream.isBroadcast) {
      return _bind(stream).asBroadcastStream();
    } else {
      return _bind(stream);
    }*/
    return _bind(stream);
    //----------------------------------------------
  }

  Stream<T> _bind(Stream<T> stream) {
    StreamController<T>? controller;
    StreamSubscription<T>? subscription;

    controller = StreamController<T>(
      onListen: () {
        // Emit the initial value
        controller?.add(initialValue);

        subscription = stream.listen(
          controller?.add,
          onError: (Object error) {
            controller?.addError(error);
            controller?.close();
          },
          onDone: controller?.close,
        );
      },
      onPause: ([Future<dynamic>? resumeSignal]) {
        subscription?.pause(resumeSignal);
      },
      onResume: () {
        subscription?.resume();
      },
      onCancel: () {
        return subscription?.cancel(); 
        //------------------'asBroadcastStream' has different controller,can't cancel this subscription
        //You can use  stream.asBroadcastStream(onListen,onCancel)
        //But it's not necessary
      },
      sync: true,
    );

    return controller.stream;
  }
}
‘’’

chipweinberger, 2023-12-18:
> You can use  stream.asBroadcastStream(onListen,onCancel)

why do you think it is not necessary?

We want to return a broadcast stream, if the original stream was broadcast. This would otherwise be a breaking change.

ycnc1314, 2023-12-18:
> why do you think it is not necessary?
> 
> We want to return a broadcast stream, if the original stream was broadcast. This would otherwise be a breaking change.

```

extension _StreamNewStreamWithInitialValue<T> on Stream<T> {
  Stream<T> newStreamWithInitialValue(T initialValue) {
    return transform(_NewStreamWithInitialValueTransformer(initialValue));
  }
}

  static Stream<BluetoothAdapterState> get adapterState async* {
    ...
    yield* FlutterBluePlus._methodStream.stream
        .where((m) => m.method == "OnAdapterStateChanged")
        .map((m) => m.arguments)
        .map((args) => BmBluetoothAdapterState.fromMap(args))
        .map((s) => _bmToAdapterState(s.adapterState))
        .newStreamWithInitialValue(_bmToAdapterState(_adapterStateNow!));
    }
```
Each subscription is a new stream.Each new stream is a copy of the original stream.

chipweinberger, 2023-12-18:
Correct. If users open a lot of streams it will cause perf problems because each one is a new copy. This is part of the API we should not change.

As long as users so not open too many streams, and close the old ones, it should not be a perf problem.

[12 more replies]

closed 2023-12-21 by chipweinberger