chipweinberger/flutter_blue_plus #662: [Help]: Cannot get BLE Indication to work but Notifications ok

IntellugoDev, 2023-11-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.25.1

### Flutter Version

 3.16.0-23.0.pre.26 

### What OS?

Android

### OS Version

13

### Bluetooth Module

ST BlueNRG-2

### What is your problem?

Using ` await characteristic.setNotifyValue(true);` to enable Indications as suggested in the guide does not enable indications on the BLE device rather sets a Notification enable flag on an Indication Characterisitcs. Additionally, I can't receive any update (Indications) from the device using the suggested use of  `characteristic.onValueReceived.listen((value) {});`

Note: That if I change the device characteristics to Notify instead of Indicate, all works well but needs Indication as some notifications are not received/acknowledged.

I also can see indication working if using a Bluetooth GATT profile app like nRFConnect

### Logs

```shell
N/A
```

chipweinberger, 2023-11-08:
// If a characteristic supports both notifications and indications,
                        // we'll use notifications. This matches how CoreBluetooth works on iOS.
                        if(canIndicate) {descriptorValue = BluetoothGattDescriptor.ENABLE_INDICATION_VALUE;}
                        if(canNotify)   {descriptorValue = BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE;}

IntellugoDev, 2023-11-08:
As I am using the Flutter Blue pluggin, I am not sure if the corresponding wrapper function for enabling Indication is implemented with the package. It appears to support enabling Notifications as Indication requires a different variable to be sent via the BLE i.e 0x02 as oppose to 0x01 for Notifications. Can I combine core Bluetooth with the Package usage?

chipweinberger, 2023-11-08:
> I am not sure if the corresponding wrapper function for enabling Indication is implemented with the package

No, we do not provide that because it is not possible on iOS.

If you want indications to be enabled, you need to delete support for `notifications` on your BLE peripheral device.

Because notifications always take precedence over indications. This is a limitation of iOS, and we've copied to android.

chipweinberger, 2023-11-08:
in 1.27.1, i've added a `forceIndications` option. It is android-only.

IntellugoDev, 2023-11-09:
Thanks for the addition. I can confirm that I can get Indications to work with the new version on Android. Although the problem I am trying to solve still remains that some Indications/Notifications don't reach the app side (I thought switching to Indications will solve this but seems to behave the same which makes sense from this explanation of the Indication - `The fact that it is the ATT layer that sends back the confirmation directly when it has received the indication, and not the app when it has successfully handled the indication.`

closed 2023-11-09 by IntellugoDev