chipweinberger/flutter_blue_plus #663: [Help]: How to programmaticly know a ANDROID_SPECIFIC_ERROR has occured causing a disconnection

ringonotts, 2023-11-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.1

### What OS?

Android

### OS Version

13

### Bluetooth Module

Custom made sensor

### What is your problem?

Hi,

My android app connects to up to seven BLE sensors at a time. This sometimes results in the ANDROID_SPECIFIC_ERROR being logged and a device will disconnect with the reason 133.

My question is, how do i write code to know that its a 133 error ? I use a listen method on the connectionState (See below) and when the state is BluetoothConnectionState.disconnected i would like to find out in my code (not from logs in the debug console) that the 133 ANDROID_SPECIFIC_ERROR occured so i can indicate to my user to toggle the phones bluetooth off and on (this fixes the issue when it rarely happens).

How do i get the log message please ?

My listen method is:

`_deviceStateListen(BmDevice bmdevice) async {
    deviceStateStream = bmdevice.bledevice.connectionState.listen((s) async {
      // Only process the device state if we are not doing a OTA update
      if (_isOtaInProgress == false) {
        _deviceState = s;
        if (s == BluetoothConnectionState.connected) {
          if (kDebugMode) {
            print("Device Connected");
          }
          setState(() {
            _initDevice();
          });

          bmdevice.bledevice.discoverServices().then((s) {
            _buildServices(bmdevice);
          });
        } else if (s == BluetoothConnectionState.disconnected) {
          // Change device download state to idle as that will have cancelled
          // Delete device and connect again ?
          });
          if (kDebugMode) {
            print("Batmon disconnected");
          }
        }
      }
    });
  `

Thanks!

### Logs

```shell
None available at time of asking for help sorry
```

chipweinberger, 2023-11-08:
use device.disconnectReason

closed 2023-11-08 by chipweinberger

ringonotts, 2023-11-08:
Awesome - thanks loads !!

chipweinberger, 2023-11-08:
I would recommend updating to a newer version. It will probably fix a lot of the errors.

ringonotts, 2023-11-09:
I need to but it breaks my scan code - i need the ".onDone" so i can display all the devices (in a known order rather than when the scan found each device) that have been found in the scan after the scan has finished.

Currently i use scan:
```
FlutterBluePlus.scan(
              withServices: tmp1, 
              timeout: const Duration(seconds: 5),
              androidUsesFineLocation: false)
          .listen((scanResult) {
        // do something with scan result
        _scanListenHandler(scanResult, id);
      }).onDone(() async {
        _isScanning = false;
        setState(() {});
        _displayDevices();                          // This line displays my devices in a known order if they are found. E.g. Device "A" will always appear first if it is found in the scan.
      });
```

The example code doesn't tell me when the scan has completed...
```
// Setup Listener for scan results.
var subscription = FlutterBluePlus.scanResults.listen((results) {
        if (results.isNotEmpty) {
            ScanResult r = results.last; // the most recently found device
            print('${r.device.remoteId}: "${r.advertisementData.advName}" found!');
        }
    },
    onError(e) => print(e);
);

// Start scanning
await FlutterBluePlus.startScan();
```

chipweinberger, 2023-11-09:
see MIGRATION.md. 

there is an extension you can use to being back the old `scan` method.

chipweinberger, 2023-11-11:
updated this comment ^

ringonotts, 2023-11-11:
Sorry Chip for the delay - I was OOO for a couple of days.

I am struggling with the new version (or the extension method) tbh. It works fine on a Pixel7a but not on a Pixel6a that has Android Enterprise on it.

The app crashes when scanning and doesn't seem to give me much of a stack trace. 

type 'Null' is not a subtype of type 'int'
#0      new BmScanAdvertisement.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:124:17)
#1      new BmScanResponse.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:145:46)
#2      FlutterBluePlus.startScan.<anonymous closure> (package:flutter_blue_plus/src/flutter_blue_plus.dart:210:39)
#3      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
#4      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#5      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#6      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#7      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#8      _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#9      _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#10     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#11     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#14     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#15     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#16     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#17     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#18     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#19     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
#20     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
#21     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
#22     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#23     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)

[image]

I'll try the timer method and see if that makes a difference...

ringonotts, 2023-11-11:
OK, so it's not the extension, its the scan method.

Sometimes it finds a device and gives the below log before crashing as above - other times it just crashes without finding any devices:

D/[FBP-Android](24330): [FBP] onMethodCall: startScan
D/BluetoothAdapter(24330): isLeEnabled(): ON
D/BluetoothLeScanner(24330): onScannerRegistered() - status=0 scannerId=3 mScannerId=0
I/flutter (24330): [FBP] <startScan> result: true
I/flutter (24330): BM:
I/flutter (24330): []
2
D/[FBP-Android](24330): [FBP] onScanResult
I/flutter (24330): [FBP] [[ OnScanResponse ]] result: {advertisements: [{connectable: 1, manufacturer_data: {}, remote_id: 00:80:E1:27:AB:60, service_data: {}}]}
4
D/[FBP-Android](24330): [FBP] onScanResult
I/orthings.batmon(24330): Background concurrent copying GC freed 216134(11MB) AllocSpace objects, 19(536KB) LOS objects, 88% free, 3243KB/27MB, paused 421us,60us total 145.202ms

ringonotts, 2023-11-11:
Pixel 7A working:

D/[FBP-Android](32569): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](32569): [FBP] initializing BluetoothAdapter
D/CompatibilityChangeReporter(32569): Compat change id reported: 263076149; UID 10291; state: DISABLED
D/[FBP-Android](32569): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](32569): [FBP] connectedPeripherals: 0
D/[FBP-Android](32569): [FBP] onMethodCall: getAdapterState
I/flutter (32569): BM:
I/flutter (32569): []
I/flutter (32569): [FBP] <setLogLevel> args: 5
D/[FBP-Android](32569): [FBP] onMethodCall: setLogLevel
I/flutter (32569): [FBP] <setLogLevel> result: true
I/flutter (32569): [FBP] <startScan> args: {with_services: [00000000-cc7a-482a-984a-7f2ed5b3e58f, d1bebb85-2019-6f14-fb6e-3b36abb03cd1], with_remote_ids: [], with_names: [], with_keywords: [], continuous_updates: false, continuous_divisor: 4, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](32569): [FBP] onMethodCall: startScan
D/BluetoothAdapter(32569): isLeEnabled(): ON
D/BluetoothLeScanner(32569): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
I/flutter (32569): [FBP] <startScan> result: true
I/flutter (32569): BM:
I/flutter (32569): []
D/[FBP-Android](32569): [FBP] onScanResult
I/flutter (32569): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -62, connectable: 1, manufacturer_data: {}, remote_id: 00:80:E1:27:AB:60, platform_name: BK-Solar1, tx_power_level: 6, service_data: {}, adv_name: BK-Solar1, service_uuids: [00000000-cc7a-482a-984a-7f2ed5b3e58f]}]}
I/flutter (32569): BM:
I/flutter (32569): [ScanResult{device: BluetoothDevice{remoteId: 00:80:E1:27:AB:60, platformName: BK-Solar1, services: null}, advertisementData: AdvertisementData{advName: BK-Solar1, txPowerLevel: 6, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [00000000-cc7a-482a-984a-7f2ed5b3e58f]}, rssi: -62, timeStamp: 2023-11-11 11:13:15.915928}]
D/[FBP-Android](32569): [FBP] onScanResult
I/flutter (32569): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -85, connectable: 1, manufacturer_data: {}, remote_id: 00:80:E1:27:B6:74, platform_name: BK-Inverter, tx_power_level: 6, service_data: {}, service_uuids: [00000000-cc7a-482a-984a-7f2ed5b3e58f]}]}
I/flutter (32569): BM:
I/flutter (32569): [ScanResult{device: BluetoothDevice{remoteId: 00:80:E1:27:AB:60, platformName: BK-Solar1, services: null}, advertisementData: AdvertisementData{advName: BK-Solar1, txPowerLevel: 6, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [00000000-cc7a-482a-984a-7f2ed5b3e58f]}, rssi: -62, timeStamp: 2023-11-11 11:13:15.915928}, ScanResult{device: BluetoothDevice{remoteId: 00:80:E1:27:B6:74, platformName: BK-Inverter, services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: 6, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [00000000-cc7a-482a-984a-7f2ed5b3e58f]}, rssi: -85, timeStamp: 2023-11-11 11:13:15.950944}]

ringonotts, 2023-11-11:
Pixel 6A crashing:

D/[FBP-Android](24978): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](24978): [FBP] initializing BluetoothAdapter
D/CompatibilityChangeReporter(24978): Compat change id reported: 263076149; UID 10288; state: DISABLED
D/[FBP-Android](24978): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](24978): [FBP] connectedPeripherals: 0
D/[FBP-Android](24978): [FBP] onMethodCall: getAdapterState
I/flutter (24978): BM:
I/flutter (24978): []
I/flutter (24978): [FBP] <setLogLevel> args: 5
D/[FBP-Android](24978): [FBP] onMethodCall: setLogLevel
I/flutter (24978): [FBP] <setLogLevel> result: true
I/flutter (24978): [FBP] <startScan> args: {with_services: [00000000-cc7a-482a-984a-7f2ed5b3e58f, d1bebb85-2019-6f14-fb6e-3b36abb03cd1], with_remote_ids: [], with_names: [], with_keywords: [], continuous_updates: false, continuous_divisor: 4, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](24978): [FBP] onMethodCall: startScan
D/BluetoothAdapter(24978): isLeEnabled(): ON
D/BluetoothLeScanner(24978): onScannerRegistered() - status=0 scannerId=3 mScannerId=0
I/flutter (24978): [FBP] <startScan> result: true
I/flutter (24978): BM:
I/flutter (24978): []
3
D/[FBP-Android](24978): [FBP] onScanResult
I/flutter (24978): [FBP] [[ OnScanResponse ]] result: {advertisements: [{connectable: 1, manufacturer_data: {}, remote_id: 00:80:E1:27:B1:BA, service_data: {}}]}
3
D/[FBP-Android](24978): [FBP] onScanResult

ringonotts, 2023-11-11:
Both pixels using the same version of Android (14):
[image]

chipweinberger, 2023-11-11:
type 'Null' is not a subtype of type 'int'
#0 new BmScanAdvertisement.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:124:17

this is p informative. this is a bug in fbp.

chipweinberger, 2023-11-11:
hmm those line numbers don't line up.

have you updated? make sure you are on the most recent version.

ringonotts, 2023-11-11:
I am on 1.27.6 which i thought was the latest - just updating to 1.28.10 and will report back

ringonotts, 2023-11-11:
Same issue:

D/[FBP-Android]( 5996): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 5996): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 5996): [FBP] onMethodCall: getAdapterState
I/flutter ( 5996): BM:
I/flutter ( 5996): []
I/flutter ( 5996): [FBP] <setLogLevel> args: 5
D/[FBP-Android]( 5996): [FBP] onMethodCall: setLogLevel
I/flutter ( 5996): [FBP] <setLogLevel> result: true
I/flutter ( 5996): [FBP] <startScan> args: {with_services: [00000000-cc7a-482a-984a-7f2ed5b3e58f, d1bebb85-2019-6f14-fb6e-3b36abb03cd1], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 4, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android]( 5996): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 5996): isLeEnabled(): ON
D/BluetoothLeScanner( 5996): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
I/flutter ( 5996): [FBP] <startScan> result: true
I/flutter ( 5996): BM:
I/flutter ( 5996): []
3
D/[FBP-Android]( 5996): [FBP] onScanResult
I/flutter ( 5996): [FBP] [[ OnScanResponse ]] result: {advertisements: [{connectable: 1, manufacturer_data: {}, remote_id: 00:80:E1:27:B6:2E, service_data: {}}]}
3
D/[FBP-Android]( 5996): [FBP] onScanResult

#0      new BmScanAdvertisement.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:156:17)
#1      new BmScanResponse.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:177:46)
#2      FlutterBluePlus.startScan.<anonymous closure> (package:flutter_blue_plus/src/flutter_blue_plus.dart:217:39)
#3      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
#4      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#5      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#6      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#7      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#8      _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#9      _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#10     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#11     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#14     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#15     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#16     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#17     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#18     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#19     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
#20     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
#21     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
#22     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#23     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)

ringonotts, 2023-11-14:
@chipweinberger Do you want me to raise a bug for this ?

Thanks, Ringo

chipweinberger, 2023-11-14:
for what bug?

chipweinberger, 2023-11-14:
p sure it's fixed. make sure you've properly updated.

otherwise sure create a big report

ringonotts, 2023-11-14:
Hi @chipweinberger Thanks so much for that.

However, this now leads to the next problem....

The Pixel 7a has the correct content in the ScanResult:
[image]

Yet the Pixel 6a doesn't have any data in ScanResult (hence why it was crashing previously i guess):
[image]

This device works fine for the old version i was using before...

Any thoughts on what i can do to investigate further please ?

chipweinberger, 2023-11-14:
open a new issue :)

ringonotts, 2023-11-20:
Good call @chipweinberger! While creating the bug report I had to get the example code working. To do this I had to upgrade my version of Android studio. This has fixed the issue.

ringonotts, 2023-11-20:
@chipweinberger is there a way I can buy you a virtual coffee or something as a token gesture ? Thanks again, Ringo

chipweinberger, 2023-11-20:
appreciate the gesture! no need! thanks