chipweinberger/flutter_blue_plus #665: [Bug]:

da1nerd, 2023-11-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.27.0

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

iPadOS 16.6.1

### Bluetooth Module

Connecting to a ZEBRA ZD621 printer

### What is your problem?

After restarting my app, the first time I try to scan for devices, the scan doesn't find anything.
If I immediately attempt to scan again, I get results.

I was using the original Flutter Blue library before this without any problems, so it seems like the iOS code needs to wait for the BL module to indicate it's ready before starting the scan.

### Logs

```shell
[log] (my code logs that it's about to scan for devices)

[FBP-iOS] handleMethodCall: flutterHotRestart
[FBP-iOS] initializing CBCentralManager
[FBP-iOS] initializing checkForMtuChangesTimer
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x282129a40> can only accept this command while in the powered on state
[FBP-iOS] disconnectAllDevices(flutterHotRestart)
[FBP-iOS] connectedPeripherals: 0
[FBP-iOS] handleMethodCall: startScan
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x282129a40> can only accept this command while in the powered on state
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
[FBP-iOS] handleMethodCall: stopScan

[log] (my code logs that it was unable to find any devices)
[log] (my code logs that it's about to scan for devices)

[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] handleMethodCall: stopScan
```

da1nerd, 2023-11-09:
My current working solution is to first scan with a short timeout (like 100ms), and wait for that to complete before performing my actual scan.

chipweinberger, 2023-11-09:
are you checking that the adapter is on?

use `adapterState`

chipweinberger, 2023-11-10:
you should be doing something like this

```
    // listen to adapterState
    FlutterBluePlus.adapterState.listen((state) {
      if (state != BluetoothAdapterState.on) {
        _error = "Bluetooth is not enabled";
      } else {
        _error = null;
        // start scanning
        FlutterBluePlus.startScan(withKeywords: ["jamcorder"]).catchError((e) {
          _error = "Scan Error: $e";
          setState(() {});
        });
      }
      setState(() {});
    });

    // listen to results
    _scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      _scanResults = results;
      setState(() {});
    }, onError: (e) {
      _error = "Scan Error: $e";
      setState(() {});
    });
```

closed 2023-11-10 by chipweinberger

da1nerd, 2023-11-17:
thanks. That did the trick.